gnuradio/gnuradio #4112: fft_filter: default move assignment operator doesn't work together with const fields

marcusmueller, 2021-01-17:
```
In file included from ../gr-filter/lib/pfb_interpolator_ccf_impl.cc:15:
In file included from ../gr-filter/lib/pfb_interpolator_ccf_impl.h:18:
In file included from ../gr-filter/lib/../include/gnuradio/filter/polyphase_filterbank.h:17:
../gr-filter/lib/../include/gnuradio/filter/fft_filter.h:97:21: warning: explicitly defaulted move assignment operator is implicitly deleted [-Wdefaulted-function-deleted]
    fft_filter_fff& operator=(fft_filter_fff&&) = default;
                    ^
../gr-filter/lib/../include/gnuradio/filter/fft_filter.h:63:15: note: move assignment operator of 'fft_filter_fff' is implicitly deleted because field 'd_decimation' is of const-qualified type 'const int'
    const int d_decimation;
              ^
../gr-filter/lib/../include/gnuradio/filter/fft_filter.h:210:21: warning: explicitly defaulted move assignment operator is implicitly deleted [-Wdefaulted-function-deleted]
    fft_filter_ccc& operator=(fft_filter_ccc&&) = default;
                    ^
../gr-filter/lib/../include/gnuradio/filter/fft_filter.h:176:15: note: move assignment operator of 'fft_filter_ccc' is implicitly deleted because field 'd_decimation' is of const-qualified type 'const int'
    const int d_decimation;
              ^
../gr-filter/lib/../include/gnuradio/filter/fft_filter.h:323:21: warning: explicitly defaulted move assignment operator is implicitly deleted [-Wdefaulted-function-deleted]
    fft_filter_ccf& operator=(fft_filter_ccf&&) = default;
                    ^
../gr-filter/lib/../include/gnuradio/filter/fft_filter.h:289:15: note: move assignment operator of 'fft_filter_ccf' is implicitly deleted because field 'd_decimation' is of const-qualified type 'const int'
    const int d_decimation;
              ^
```

I'd honestly love @ThomasHabets' opinion on this; I think not having a move assignment operator isn't terrible here, but I don't fully understand the reasoning of clang here.

drmpeg, 2021-01-18:
Here's the previous discussion. https://github.com/gnuradio/gnuradio/pull/3928

```const``` was removed in that case.

ThomasHabets, 2021-01-18:
Yeah, I still think it's safer to remove const, in order to enable moves.

The inherent inability of C++ to combine const members with moves was not sufficiently apparent to me at the time. :-(

closed 2021-01-21 by drmpeg