gnuradio/gnuradio #4113: uhd_fft crash with uhd-4.0 + gr3.9 + b210

Hoernchen, 2021-01-18:
osmosdr_fft in my yet-to-be-pushed brach works ;)
```
lldb-10 python3 -- /opt/gr39/install/bin/uhd_fft -f 100000000 -s 10000000
(lldb) target create "python3"
Current executable set to 'python3' (x86_64).
(lldb) settings set -- target.run-args  "/opt/gr39/install/bin/uhd_fft" "-f" "100000000" "-s" "10000000"
(lldb) r
Process 717195 launched: '/usr/bin/python3' (x86_64)
Process 717195 stopped and restarted: thread 1 received signal: SIGCHLD
[INFO] [UHD] linux; Clang version 11.1.0 ; Boost_107100; UHD_4.0.0.0-0-e520e3ff
Process 717195 stopped
* thread #1, name = 'python3', stop reason = signal SIGABRT
    frame #0: 0x00007ffff7deb18b libc.so.6`__GI_raise(sig=2) at raise.c:51:1
(lldb) bt
* thread #1, name = 'python3', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff7deb18b libc.so.6`__GI_raise(sig=2) at raise.c:51:1
    frame #1: 0x00007ffff7dca859 libc.so.6`__GI_abort at abort.c:79:7
    frame #2: 0x00007fffeb4d7212 libunwind.so.8`___lldb_unnamed_symbol5$$libunwind.so.8 + 10
    frame #3: 0x00007fffd5234b66 libuhd.so.4.0.0`void boost::exception_detail::throw_exception_<boost::property_tree::ptree_bad_path>(boost::property_tree::ptree_bad_path const&, char const*, char const*, int) + 214
    frame #4: 0x00007fffd522fa53 libuhd.so.4.0.0`boost::property_tree::basic_ptree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::get_child(boost::property_tree::string_path<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::property_tree::id_translator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) + 323
    frame #5: 0x00007fffd522e939 libuhd.so.4.0.0`uhd::config_parser::options(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) + 169
    frame #6: 0x00007fffd52578a9 libuhd.so.4.0.0`(anonymous namespace)::get_args(uhd::device_addr_t const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) + 617
    frame #7: 0x00007fffd5257577 libuhd.so.4.0.0`uhd::prefs::get_usrp_args(uhd::device_addr_t const&) + 263
    frame #8: 0x00007fffd5302c48 libuhd.so.4.0.0`uhd::device::make(uhd::device_addr_t const&, uhd::device::device_filter_t, unsigned long) + 6472
    frame #9: 0x00007fffd5036796 libuhd.so.4.0.0`uhd::usrp::multi_usrp::make(uhd::device_addr_t const&) + 358
    frame #10: 0x00007fffe407ba5d libgnuradio-uhd.so.3.9.0`gr::uhd::usrp_block_impl::usrp_block_impl(this=0x0000000001920f18, vtt=<unavailable>, device_addr=0x00007fffffffcac0, stream_args=0x00007fffffffc938, ts_tag_name=<unavailable>) at usrp_block_impl.cc:127:12
    frame #11: 0x00007fffe4097f28 libgnuradio-uhd.so.3.9.0`gr::uhd::usrp_source_impl::usrp_source_impl(this=0x0000000001920f10, device_addr=0x00007fffffffcac0, stream_args=0x00007fffffffc938, issue_stream_cmd_on_start=<unavailable>) at usrp_source_impl.cc:36:7
    frame #12: 0x00007fffe40979bd libgnuradio-uhd.so.3.9.0`gr::uhd::usrp_source::make(device_addr=0x00007fffffffcac0, stream_args=<unavailable>, issue_stream_cmd_on_start=true) at usrp_source_impl.cc:28:34
    frame #13: 0x00007fffe411137d uhd_python.cpython-38-x86_64-linux-gnu.so`gr::uhd::usrp_source::make(device_addr_str=<unavailable>, stream_args=<unavailable>, issue_stream_cmd_on_start=<unavailable>) at usrp_source.h:65:16
    frame #14: 0x00007fffe41132ed uhd_python.cpython-38-x86_64-linux-gnu.so`void pybind11::cpp_function::initialize<void pybind11::detail::initimpl::factory<std::shared_ptr<gr::uhd::usrp_source> (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type (*)(), std::shared_ptr<gr::uhd::usrp_source> (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type ()>::execute<pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >, pybind11::arg, pybind11::arg, pybind11::arg_v, char const*>(pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >&, pybind11::arg const&, pybind11::arg const&, pybind11::arg_v const&, char const* const&) &&::'lambda'(pybind11::detail::value_and_holder&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), void, pybind11::detail::value_and_holder&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool, pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::detail::is_new_style_constructor, pybind11::arg, pybind11::arg, pybind11::arg_v, char const*>(pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >&&, pybind11::arg (*)(void pybind11::detail::initimpl::factory<std::shared_ptr<gr::uhd::usrp_source> (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type (*)(), std::shared_ptr<gr::uhd::usrp_source> (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type ()>::execute<pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >, pybind11::arg, pybind11::arg, pybind11::arg_v, char const*>(pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >&, pybind11::arg const&, pybind11::arg const&, pybind11::arg_v const&, char const* const&) &&::'lambda'(pybind11::detail::value_and_holder&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool)...), void const&...)::'lambda'(pybind11::detail::function_call&)::__invoke(pybind11::detail::function_call&) [inlined] void pybind11::detail::initimpl::factory<std::shared_ptr<gr::uhd::usrp_source> (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type (*)(), std::shared_ptr<gr::uhd::usrp_source> (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type ()>::execute<pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >, pybind11::arg, pybind11::arg, pybind11::arg_v, char const*>(this=<unavailable>, v_h=0x00007fffffffccf0, args="", args=<unavailable>, args=<unavailable>) &&::'lambda'(pybind11::detail::value_and_holder&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool)::operator()(pybind11::detail::value_and_holder&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool) const at init.h:237:35
    frame #15: 0x00007fffe41132e1 uhd_python.cpython-38-x86_64-linux-gnu.so`void pybind11::cpp_function::initialize<void pybind11::detail::initimpl::factory<std::shared_ptr<gr::uhd::usrp_source> (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type (*)(), std::shared_ptr<gr::uhd::usrp_source> (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, uhd::stream_args_t const&, bool), pybind11::detail::void_type ()>::execute<pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >, pybind11::arg, pybind11::arg, pybind11::arg_v, char const*>(pybind11::class_<gr::uhd::usrp_source, gr::uhd::usrp_block, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::uhd::usrp_source> >&, pybind11::arg const&, pybind11::arg const&, [... cut ...]
    frame #44: 0x000000000067de4e python3`PyRun_SimpleFileExFlags at pythonrun.c:428:13
    frame #45: 0x00000000006b6032 python3`Py_RunMain at main.c:381:15
    frame #46: 0x00000000006b63bd python3`Py_BytesMain(argc=<unavailable>, argv=<unavailable>) at main.c:739:12
    frame #47: 0x00007ffff7dcc0b3 libc.so.6`__libc_start_main(main=(python3`main at python.c:15:1), argc=6, argv=0x00007fffffffde28, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffde18) at libc-start.c:308:16
    frame #48: 0x00000000005fa4de python3`_start + 46
```

mormj, 2021-01-21:
Appears as a segmentation fault when launching uhd_fft

Seeing the same thing here and currently bisecting.  Doesn't seem to be related to the UHD version - have replicated with UHD 3.15 as well.  Going back far enough on master branch (went back to ccda11205b09ceaa59dcd0d547a845ce8de4b373), the issue doesn't appear

mormj, 2021-01-21:
6d6c9a254102374d3fdbc4c39eac0b67e07168b7 is the offending commit

noc0lour, 2021-01-21:
Hmmm, seems `UHD` tries to catch an error here, but instead GNU Radio catches it. Don't know who's fault it is. 
@mbr0wn might have more insight?

https://github.com/EttusResearch/uhd/blob/master/host/lib/utils/config_parser.cpp#L45
```
std::vector<std::string> config_parser::options(const std::string& section)
{
    try {
        return _options(_pt.get_child(section));
    } catch (const boost::property_tree::ptree_bad_path&) {
        return std::vector<std::string>{};
    }
}
```

mormj, 2021-01-21:
Looks like a workaround for now is to uninstall `libunwind-dev` to make all this code go away

mbr0wn, 2021-01-24:
No clue at first sight.

mbr0wn, 2021-02-02:
Looks like this fails with any combo of UHD + Qt. UHD-only works, Qt-only works too.

mbr0wn, 2021-02-02:
So... here's an idea: What if something wasn't throwing the right exception type.

mbr0wn, 2021-02-02:
No that's not it. The backtrace already should have told me that.