gnuradio/gnuradio #4119: error on windows 10 doesn't open any file in gnu radio companion

Zupayruna96, 2021-01-19:
Error message output is: (gnuradio-companion.py:3804): GLib-GIO-ERROR **: 01:10:50.458: Settings schema 'org.gtk.Settings.FileChooser' is not installed

gnieboer, 2021-01-19:
This is likely a windows-specific error, which installer are you using?

gnieboer, 2021-01-19:
Disregard, duplicated in most recent installer!

gnieboer, 2021-01-19:
The issue is multiple schemas were merged into share/glib-2.0/schemas/, with each pre-compiled, but when combined only one compiled file is left.  So the schemas need to recompiled after combining.

gnieboer, 2021-01-19:
Please downloaded an updated installer (same link) and remove/reinstall.  Apologies, in my limited test setup the flowgraphs I use to test the install are already open so I didn't try to open dialog.

closed 2021-01-19 by gnieboer