gnuradio/gnuradio #4141: Errors for Flow diagrams using type 'int'

bobbryce, 2021-01-23:
Using master 3.9.0, and getting basic errors on flow diagrams that worked in v3.8.2.

Ref diagram, this is a simple signal generator into a frequency sink and waterfall, but two of the Blocks show errors (related to type 'int') that stop the program running.

1. Signal Source: Param - Waveform(waveform):
	Expression gr_waveform_t.GR_SIN_WAVE is invalid for type 'int'.
2. Waterfall: Param - Window Type(wintype):
	Expression win_type.WIN_BLACKMAN_hARRIS is invalid for type 'int'.

This is a fresh install of Linux Mint 20.1, and a fresh install of GRC from the master 
(sudo add-apt-repository ppa:gnuradio/gnuradio-master)

Any thoughts?  It may be related to #3454?

[image]

nickoe, 2021-02-07:
@bobbryce What happens if you save the grc and open it again or open the block and close it? IIRC there was an issue wher that is required to update the grc for those window types.

bobbryce, 2021-02-07:
Hi nickoe - thanks for trying to help.
I've had to revert from v3.9.0 for several reasons - most important of which is that I want to use the Adalm Pluto hardware and have eventually got it confirmed by the GRC team that v3.9.0 does not yet work with the gr-iio library.

So - reverted back to installing from binary and this meant v3.8.1.

Next issue I had is that if I intend using the Pluto hardware with both GRC AND with Python, there is an issue with a conflict in iio namespace between the Analog Devices iio library code and the GRC iio library code. I seem to have (yesterday!) got past that issue by renaming the AD iio.py file and then adjusting a couple of files within the pyadi-iio library.  So, I now have the Pluto responding to both GRC flow diagrams and to Python scripts using pyadi-iio.

Thanks again for your interest / support.
Bob

daniestevez, 2021-02-07:
I can confirm that this problem happens with my GNU Radio 3.9.0 install. Maybe it's something wrong with my install. I built from source using `-DCMAKE_INSTALL_PREFIX` to install into my home directory. The problem also happens if I create a new flowgraph in `gnuradio-companion` 3.9.

nickoe, 2021-02-07:
@daniestevez I have not got around to try your branch, but I was thinking it was to this and related changes. https://github.com/gnuradio/gnuradio/pull/3998

willcode, 2021-02-27:
@daniestevez Does reselecting the signal/window type in the params dialog clear the error for you? That one we've seen. Or is this something else?

daniestevez, 2021-03-04:
I've managed to find the cause of my problem. I have GNU Radio 3.8 installed in `/usr/` and GNU Radio 3.9 installed in a folder in my home dir. Not only I had forgot to export `GRC_BLOCKS_PATH` to point `gnuradio-companion` to pull the blocks from the GNU Radio 3.9 install, but also `gnuradio-companion` was pulling the 3.8 blocks from `/usr/share` because of the `global_blocks_path` in `~/.gnuradio/config.conf` (maybe related to https://github.com/gnuradio/pybombs/issues/505).

After exporting `GRC_BLOCKS_PATH`  and commenting out `global_blocks_path` in `~/.gnuradio/config.conf`, the signal source block works correctly for me.

willcode, 2021-03-04:
Ok to close?

daniestevez, 2021-03-04:
Fine with me, since @bobbryce hasn't commented again since opening and the rough instructions fix this are in my previous comment.

willcode, 2021-03-04:
@bobbryce Since you've gone back to 3.8, and people are successfully using the Signal Source, we'll close the issue.

There is an "upgrade" step for some enum parameters like this when moving from 3.8 to 3.9. If anyone runs across a problem like this, try re-selecting the value (waveform in this case) from the dropdown.

closed 2021-03-04 by willcode

bobbryce, 2021-03-07:
Jeff - OK with me too. Thanks.

pierre-olivier832, 2022-02-20:
I have the same error with grc. I have the windows installer. What can i do to fix it?