gnuradio/gnuradio #4154: Error installing gr-iqbal with PyBOMBS

fdgm0001, 2021-01-26:
Hi all, I had recently installed GNU Radio 3.8 with PyBOMBS. Then, I have tried to install gr-iqbal but I have obtained the following result:

```
[INFO] Prefix Python version is: 3.8.5
[INFO] PyBOMBS Version 2.3.4
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-iqbal
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gr-iqbal
[WARNING] Build dir already exists: /home/fdgm0001/pybombs/bladeRF/src/gr-iqbal/build
Configuring: (100%) [===================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
CMake Error at CMakeLists.txt:32 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.9".

  The following configuration files were considered but not accepted:

    /home/fdgm0001/pybombs/bladeRF/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.1.0

-- Configuring incomplete, errors occurred!
See also "/home/fdgm0001/pybombs/bladeRF/src/gr-iqbal/build/CMakeFiles/CMakeOutput.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gr-iqbal:
Configuration failed
[ERROR] Error installing package gr-iqbal. Aborting.
```
Attending to the result, I understand that the GNU Radio version should be 3.9. However, I don't know how to install this version with PyBOMS. It is possible to solve the problem without install GNU Radio again?

RafaelRuizFeliu, 2021-01-29:
Hi fdgm0001. Just replace the master files cloned in ~/pyBombs/bladeRF/src/gr-iqval by the files in branch 3.8 (https://github.com/osmocom/gr-iqbal/tree/gr3.8) and try again.
I fix the problem by this way.
Regards

dkozel, 2021-01-30:
I've just opened an issue on the gr-recipes repo. This is where PyBOMBS gets the instructions on how to install each OOT module. Right now the iqbal recipe doesn't do any branch selection.

Thanks for pointing it out. I'm going to close this issue and any fix would be tracked on the other repo.

closed 2021-01-30 by dkozel

dails08, 2021-06-03:
This is where it's being tracked, in case anyone else cares to follow it there: https://github.com/gnuradio/gr-recipes/issues/246