gnuradio/gnuradio #4161: gnuradio-config-info on current master (3.10) reports 3.8 version number

mormj, 2021-01-27:
```
josh@josh-m15:/share/gnuradio/gr39/bin$ gnuradio-config-info -v
v3.8.0.0-938-g5e8286ee
```
at first i thought the prefix was pointing to the wrong thing, but this checks out, and my 3.9 OOTs run

```
josh@josh-m15:/share/gnuradio/gr39/bin$ ldd gnuradio-config-info 
	linux-vdso.so.1 (0x00007fff65cf7000)
	libgnuradio-runtime.so.3.10.0git => /share/gnuradio/gr39/lib/libgnuradio-runtime.so.3.10.0git (0x00007f0b852b3000)
etc
```

dave-pi, 2021-01-27:
I think It's the git describe in GrVersion.cmake. Basically walks back until it finds a tag and adds the number of commits since that tag (938) and the latest commit abbreviated hash (5e8286ee). v3.8.0.0 was probably created in the master branch.

drmpeg, 2021-01-28:
Master branch needs to be tagged. https://github.com/gnuradio/gnuradio/issues/4059

Also, version in CMakeLists.txt on maint-3.9 needs to be changed to ```git``` (so that you get the commit number since last tag and commit hash). Just like I did on maint-3.8 https://github.com/gnuradio/gnuradio/commit/f2ff682120c226a6faa5768df615ed8ea570f9e1

jacobagilbert, 2021-01-31:
I stumbled into this too. For anyone else this is causing trouble for you can just run:

`git tag -a 3.10api 0901fcb5459c5169017c3073630d7aca158bd626 -m "change API to 3.10"`

dl1ksv, 2021-02-13:
Just a general question: What is gnuradio-config-info expected to provide ?
At the moment --builddate and --version provide information that are valid at the time cmake was run.  The cmake run builds
build/gnuradio-runtime/lib/constants.cc
from gnuradio-runtime/lib/constants.cc.in.
So if you pull the latest commits, and run only make and make install, you get neither a different builddate nor a different version.

Is this really what we need ?
Shouldn't the version info contain the commit hash the build was really made from ?
And which information gives the latest tag git describe finds ?
Why not ( only ) use the version info the cmake file provides + the commit hash  gnuradio is build from ?

mbr0wn, 2021-02-15:
@drmpeg sez:

> I would suggest commit 0901fcb be tagged with v3.10.0.0git

...and I agree.

mbr0wn, 2021-02-18:
https://github.com/gnuradio/gnuradio/releases/tag/v3.10.0.0git

closed 2021-02-18 by mbr0wn