gnuradio/gnuradio #4162: How does Constellation Modulator block work

seuliang, 2021-01-29:
the wiki page says  Constellation Modulator block  is a Hierarchical block which i can't open it.
so I'd like to know how this block works.
[image]
for example, how can i get the same function that Constellation Modulator block in picture above with basic blocks in follow picture.
[image]

dkozel, 2021-01-30:
Hi @seuliang 

The source code of the Constellation Modulator block is here:
https://github.com/gnuradio/gnuradio/blob/4849af9ad83bc16d21e2c63ab95bab1755289c2f/gr-digital/python/digital/generic_mod_demod.py#L67

The documentation page describes the blocks inside the Constellation Modulator. I hope that will answer your questions.
https://www.gnuradio.org/doc/doxygen/page_digital.html

If you have other questions about how specific blocks work or how to set them up I would like to invite you to join the chatroom where it is easier to ask and answer questions.
https://chat.gnuradio.org

closed 2021-01-30 by dkozel

seuliang, 2021-01-30:
thank you @dkozel 
this is exactly what I want.

seuliang, 2021-02-18:
hi, @dkozel 
I have read the source code of generic_mod_demod.py and used blocks that in class generic_mod(gr.hier_block2). 
The blocks like blow.
[image]
However, I can't get right output. the real part of the block's output is 
[image]

the right real part output of qpsk is
[image]

Is there anything I missed?