gnuradio/gnuradio #4169: Custom Block Froze in GRC

XDAChen, 2021-01-30:
Hi fellow gnuradio users, I am here because I tried to use a custom python block but the block just froze my grc when I drag it. Here is the story:
I want to use a custom RFID module developed by Nikos Kargas: [https://github.com/nkargas/Gen2-UHF-RFID-Reader](url). From the comments and the academic papers I had read it seems like that module shall be working if installed correctly.

To successfully went through the installation processed, I used ubuntu 14, and I PPA install gnu-radio 3.7
then I install log4cpp
 purge cmake 2, and install cmake 3 
 download the code, unzip, went into the folder
 mkdir build
 cd build/
 sudo cmake ../
a few bug came up, and I had to apt install doxygen and apt install gnuradio.dev
sudo make install
sudo ldconfig

and it looks like it went through
[image]

So then I created a config.conf to specify my XML file path at usr/local/share/gnuradio/grc/blocks. When I opened my GRC I was able to find the rfid_blocks.

However when I try to drag it to make a flow graph, the screen froze and let me with this:
[image]

I suspected there were something wrong when compiling, so I ran a ctest in ~gr-rfid/build, but the result is not informative at all:

[image]

anyone having similar trouble or any idea?
thanks in advance.

dkozel, 2021-01-30:
Hi @XDAChen 

gr-rfid is written by Nikos Kargas as you mention. It requires GNU Radio 3.7.2 and I can see that you've installed that version, so that should work. That version of GNU Radio was released 30 Nov 2013 and a huge amount has changed since then. I suggest asking your on their repository since they may have some knowledge of what issues the code has. 

From the screenshot you posted GNU Radio is listing several errors that are in the XML file. Looking quickly at that file it is clear that it was never finished. It has the defaults from the template, missing important information. This has to be fixed before that block can work in GNU Radio companion.
https://github.com/nkargas/Gen2-UHF-RFID-Reader/blob/master/gr-rfid/grc/rfid_reader.xml

The readme has a section on how to run the application. It says there is a python program you should run directly.
https://github.com/nkargas/Gen2-UHF-RFID-Reader#how-to-run

We aren't going to be able to help more here. This is a setup issue with this out of tree module specifically and is for a very old version of GNU Radio that we just don't have the ability to continue supporting. If you have questions about using GNU Radio (hopefully a newer version) or RF in general I hope you'll join the chatroom.
https://chat.gnuradio.org

closed 2021-01-30 by dkozel

XDAChen, 2021-01-30:
Thank you @dkozel ,
I see, reinstalled the whole thing and successfully ran a test on it. 

Just in case anyone in the future would struggle with this:
Kargas RFID reader with gnu radio is a very handy module for anyone who wants to customize an rfid block with USRP. But this piece of code is super old. Maybe in the future I can update the code for new version of both OS and GNU radio.

For now, use Ubuntu 16 with gnuradio 3.7 will do the job. 
Ubuntu 14 is too old, there will just be troubles here and there
have not tested 18, but gnu radio 3.7 is not compatible with 20.
16+3.7 will work well by following the instrcution.