gnuradio/gnuradio #4172: Can't install 3.8

ahmedshahein, 2021-01-30:
I have updated from 3.7 to 3.9, than I am reverting back to 3.8 as advised. For 3.7 and 3.9 I used sudo apt-get install gnuradio. However, for 3.8 I am installing it from the source.
I am getting the following error when running the cmake command:

Scanning dependencies of target trellis_swig0
[ 90%] Building CXX object gr-trellis/swig/CMakeFiles/trellis_swig0.dir/CMakeFiles/trellis_swig0.dir/trellis_swig0PYTHON_wrap.cxx.o
/home/shahein/gnuradio/build/gr-trellis/swig/CMakeFiles/trellis_swig0.dir/trellis_swig0PYTHON_wrap.cxx: In function ‘PyObject* _wrap_constellation_base(PyObject*, PyObject*)’:
/home/shahein/gnuradio/build/gr-trellis/swig/CMakeFiles/trellis_swig0.dir/trellis_swig0PYTHON_wrap.cxx:6247:29: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<std::shared_ptr<gr::digital::constellation> >’ and ‘gr::digital::constellation_sptr {aka boost::shared_ptr<gr::digital::constellation>}’)
       result = (arg1)->base();
                             ^
/home/shahein/gnuradio/build/gr-trellis/swig/CMakeFiles/trellis_swig0.dir/trellis_swig0PYTHON_wrap.cxx:32:21: note: candidate: SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = std::shared_ptr<gr::digital::constellation>]
   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);

curtcorum, 2021-02-01:
@ahmedshahein 
It is hard to tell from the information you provide.
Also this looks like output from 'make' command.

`<pre>... wavelet_swig
... pygen_gr_wavelet_python_wavelet_301a9
... gnuradio-zeromq
... zeromq_swig_swig_doc
... _zeromq_swig_doc_tag
... zeromq_swig
... zeromq_swig_swig_compilation
... pygen_gr_zeromq_swig_98ce4
... pygen_gr_zeromq_python_zeromq_aba78
<font color="#8AE234"><b>curt@green</b></font>:<font color="#729FCF"><b>~/src/gnuradio/build</b></font>$ make trellis_swig
trellis_swig0                    trellis_swig0_swig_doc/          trellis_swig1_swig_doc           trellis_swig2_swig_compilation/  trellis_swig3_swig_compilation
trellis_swig0/                   trellis_swig1                    trellis_swig1_swig_doc/          trellis_swig2_swig_doc           trellis_swig3_swig_compilation/
trellis_swig0_swig_compilation   trellis_swig1/                   trellis_swig2                    trellis_swig2_swig_doc/          trellis_swig3_swig_doc
trellis_swig0_swig_compilation/  trellis_swig1_swig_compilation   trellis_swig2/                   trellis_swig3                    trellis_swig3_swig_doc/
trellis_swig0_swig_doc           trellis_swig1_swig_compilation/  trellis_swig2_swig_compilation   trellis_swig3/                   
<font color="#8AE234"><b>curt@green</b></font>:<font color="#729FCF"><b>~/src/gnuradio/build</b></font>$ make trellis_swig0
[  0%] Built target _pmt_swig_doc_tag
[  0%] Built target pmt_swig_swig_doc
[  0%] Built target pmt_swig_swig_compilation
[  7%] Built target volk_obj
[  7%] Built target volk
[ 10%] Built target gnuradio-pmt
[ 23%] Built target gnuradio-runtime
[ 23%] Built target pmt_swig
[ 26%] Built target _runtime_swig_doc_tag
[ 26%] Built target runtime_swig_swig_doc
[ 26%] Built target runtime_swig_swig_compilation
[ 26%] Built target runtime_swig
[ 26%] Built target _trellis_swig0_doc_tag
[ 26%] Built target trellis_swig0_swig_doc
[ 26%] Built target trellis_swig0_swig_compilation
[ 60%] Built target gnuradio-blocks
[ 63%] Built target gnuradio-fft
[ 71%] Built target gnuradio-filter
[ 76%] Built target gnuradio-analog
[ 92%] Built target gnuradio-digital
[ 97%] Built target gnuradio-trellis
[100%] Built target trellis_swig0
<font color="#8AE234"><b>curt@green</b></font>:<font color="#729FCF"><b>~/src/gnuradio/build</b></font>$ cmake ..
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring testing-support support...
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
CMake Warning (dev) in CMakeLists.txt:
  A logical block opening on the line

    /home/curt/src/gnuradio/CMakeLists.txt:299 (if)

  closes on the line

    /home/curt/src/gnuradio/CMakeLists.txt:303 (endif)

  with mis-matching arguments.
This warning is for project developers.  Use -Wno-dev to suppress it.
[... 351 lines cut ...]
--   * * oss
--   * * jack
--   * * portaudio
--   * gr-channels
--   * gr-qtgui
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr_modtool
--   * gr-vocoder
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * sphinx
--   * gr-video-sdl
-- 
-- Using install prefix: /usr/local
-- Building for version: v3.8.2.0-59-gd2129cff / 3.8.2git
-- Configuring done
-- Generating done
-- Build files have been written to: /home/curt/src/gnuradio/build
</pre>`

mbr0wn, 2021-02-02:
Please don't post support questions on this issue tracker.

closed 2021-02-02 by mbr0wn

tkircher, 2022-01-26:
I'm having the same issue. It's not a support question, it's a bug in gnuradio that this doesn't build correctly.