gnuradio/gnuradio #4173: Source build on Mac OSX 10.14.6: ld: library not found for -lgsl

dtgriscom, 2021-01-30:
I'm trying to build gnuradio from source on OS X 10.14.6, but even though I have GSL installed I get loader errors "library not found for -lgsl". I've included lots of info below; I'm happy to do tests or provide more info if you like.

The details: I installed gnuradio using brew, but was then having trouble building [gqrx](https://gqrx.dk/) because gnuradio's Wav File Sink block hadn't been built. So, after some advice from the IRC I uninstalled gnuradio, installed libsndfile using brew, and started building gnuradio from source (maint-3.9 branch b51c07b33). For consistency, I ran cmake from a script:

````
#!/bin/bash

rm -rf build
mkdir build
cd build

CC=/usr/bin/llvm-gcc CXX=/usr/bin/llvm-g++ cmake \
-DPYTHON_EXECUTABLE=/usr/local/opt/python@3.9/bin/python3 \
-DPYTHON_INCLUDE_DIR=/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/Headers \
-DPYTHON_LIBRARY=/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/Python \
-DGR_PYTHON_DIR=/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages \
.. | tee cmake.log
````
cmake ran fine, but then make ended with error:
````
[ 41%] Linking CXX shared library libgnuradio-fec.dylib
ld: library not found for -lgsl
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [gr-fec/lib/libgnuradio-fec.3.9.0.0.dylib] Error 1
make[1]: *** [gr-fec/lib/CMakeFiles/gnuradio-fec.dir/all] Error 2
make: *** [all] Error 2
````
So, gnuradio-fec isn't building. Using a Google'd trick:
````
make SHELL="/bin/bash -x" gnuradio-fec
````
I found the commands that died:
````
cd /Users/griscom/Documents/Tools/gnuradio/build/gr-fec/lib
/usr/local/Cellar/cmake/3.15.4/bin/cmake -E cmake_link_script CMakeFiles/gnuradio-fec.dir/link.txt --verbose=
````
Sure enough, running these by hand triggered the same issue. Contents of CMakeFiles/gnuradio-fec.dir/link.txt (word-wrapped, some *.cc.o files removed):
````
/usr/bin/llvm-g++  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/-O3 -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk
/-dynamiclib -Wl,-headerpad_max_install_names -Xlinker -undefined -Xlinker
/dynamic_lookup -compatibility_version 3.9.0 -current_version 3.9.0 -o
/libgnuradio-fec.3.9.0.0.dylib -install_name @rpath/libgnuradio-fec.3.9.0.dylib
/CMakeFiles/gnuradio-fec.dir/generic_decoder.cc.o
/CMakeFiles/gnuradio-fec.dir/generic_encoder.cc.o

<snipped lots of *.cc.o files>

/reed-solomon/CMakeFiles/gr_fec_rs.dir/encode_rs_ccsds.c.o
/reed-solomon/CMakeFiles/gr_fec_rs.dir/init_rs.c.o -Wl,-rpath,/usr/local/lib
/../../gr-blocks/lib/libgnuradio-blocks.3.9.0.0.dylib
/../../gnuradio-runtime/lib/libgnuradio-runtime.3.9.0.0.dylib
/../../gnuradio-runtime/lib/pmt/libgnuradio-pmt.3.9.0.0.dylib
//usr/local/lib/libboost_program_options-mt.dylib
//usr/local/lib/libboost_filesystem-mt.dylib
//usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_regex-mt.dylib
//usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/liblog4cpp.dylib
//usr/local/Cellar/gmp/6.2.1/lib/libgmpxx.dylib
//usr/local/Cellar/gmp/6.2.1/lib/libgmp.dylib /usr/local/lib/libvolk.2.4.dylib
//usr/local/Cellar/libsndfile/1.0.31/lib/libsndfile.dylib -lgsl -lgslcblas -lm 
````
So, although gsl is linked in (`-lgsl`), there's no path to where gsl is located. `gsl-config --preset` output:
````
/usr/local/Cellar/gsl/2.6
````
For fun, I added the following after the `find_package(GSL)` in `gr-fec/CMakeLists.txt`:
````
message(NOTICE "GSL_FOUND ${GSL_FOUND}")
message(NOTICE "GSL_LIBRARY_DIRS ${GSL_LIBRARY_DIRS}")
message(NOTICE "GSL_INCLUDE_DIRS ${GSL_INCLUDE_DIRS}")
message(NOTICE "GSL_DEFINITIONS ${GSL_DEFINITIONS}")
message(NOTICE "GSL_LIBRARIES ${GSL_LIBRARIES}")
message(NOTICE "GSL_LDFLAGS ${GSL_LDFLAGS}")
message(NOTICE "GSL_CONFIG_EXECUTABLE ${GSL_CONFIG_EXECUTABLE}")
message(NOTICE "GSL_CFLAGS ${GSL_CFLAGS}")
````
Output:
````
GSL_FOUND TRUE
GSL_LIBRARY_DIRS /usr/local/Cellar/gsl/2.6/lib
GSL_INCLUDE_DIRS /usr/local/Cellar/gsl/2.6/include
GSL_DEFINITIONS 
GSL_LIBRARIES gsl;gslcblas;m
GSL_LDFLAGS -L/usr/local/Cellar/gsl/2.6/lib;-lgsl;-lgslcblas;-lm
GSL_CONFIG_EXECUTABLE 
GSL_CFLAGS -I/usr/local/Cellar/gsl/2.6/include
````
So, GSL is indeed being found, but somehow its location isn't being added to the linkage command. I suspect that the fault is somewhere in `cmake/Modules/FindGSL.cmake`, but I'm not enough of a cmake maven to figure it out.

Full cmake output:

[cmake.log](https://github.com/gnuradio/gnuradio/files/5898572/cmake.log)

Full make output:

[make.log](https://github.com/gnuradio/gnuradio/files/5898573/make.log)

mkierczak, 2021-01-31:
I managed to build using `brew install --build-from-source gnuradio`. It behaves erratic though and freezes when I search for components with the following error message: 
```
2021-01-31 20:36:54.225 Python[53213:865823] unlockFocus called too many times. Called on <GdkQuartzView: 0x7fcb4246e060>.
```

dtgriscom, 2021-01-31:
AFAIK, my "brew install gnuradio" worked just fine, except that it didn't install the "Wav File Sink" block, which gqrx required.

dtgriscom, 2021-02-12:
Is there any other info I can provide that would make it easier to help me?

NeuroForLunch, 2021-05-04:
After spending many hours on this, I figured it out. To find gsl place these in .zshrc, .zprofile, or execute from command prompt.

```
export PATH="/usr/local/opt/gsl/lib:$PATH"
export LD_LIBRARY_PATH="/usr/local/opt/gsl/lib;-lgsl;-lgslcblas;-lm:$LD_LIBRARY_PATH"
export LDFLAGS="-L/usr/local/opt/gsl/lib:$LDFLAGS"
export LDFLAGS="-L/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib":"$LDFLAGS"
```

NeuroForLunch, 2021-06-24:
`brew install gnuradio` works correctly now

michaelld, 2021-09-25:
`port installed gnuradio` works as well. I'm guessing the issue was fixed in GR's and/or GSL's CMake scripts. I'm closing this issue; if this is still a problem please reopen and add relevant info as to what's going on.

closed 2021-09-25 by michaelld