gnuradio/gnuradio #4174: GRC freezes when searching for components (OsX, BigSur)

mkierczak, 2021-01-31:
I have compiled from sources using Homebrew (OsX, BigSur):
`brew install --build-from-source gnuradio`. 

In general, it works, I can sometimes run a simple workflow, but GRC behaves erratic and freezes when I search for components, specifically, when I try to delete current search string. Sometimes "reload blocks" helps, sometimes not. Always with the following error message:
```
2021-01-31 20:36:54.225 Python[53213:865823] unlockFocus called too many times. Called on <GdkQuartzView: 0x7fcb4246e060>. 
```

sam-k, 2021-02-09:
Same issue here. GRC is unusable at this point on macOS Big Sur; nothing is responsive.

jackdally, 2021-02-25:
I am also running into the same issue. It freezes the GUI completely.

NeuroForLunch, 2021-05-02:
I found the answer in the source code - the GUI is limited to some maximum pixel size so if you are attempting to do full screen on a macbook it will freeze. On a macbook retina 15" the GUI has to be no more than 3/4 of the screen (maybe even smaller).

dkozel, 2021-06-19:
@NeuroForLunch Where is this limitation in the code. Clearly this would be excellent to fix.

NeuroForLunch, 2021-06-19:
I am unsure of the exact line of code that was causing the issue but I have narrowed it down to VariableEditor.py, Constants.py, Config.py, flowgraph.py, and block.py. Some of these set a default value but possibly limit the maximum size by only allowing some multiple which is still less than the resolution on a 15" mac which is 2880x1600.

NeuroForLunch, 2021-06-24:
Here is a video example of the GUI behavior. You'll see that at a certain point of shrinking the window it just clicks in and starts working normally. 

https://www.youtube.com/watch?v=eVMLWy6u_hM (HD is still processing so check back tomorrow)

NeuroForLunch, 2021-06-24:
The fix is in: #4814

ryanvolz, 2021-09-24:
Has this been fixed in some other way, since #4814 is closed, or does it still occur? It would be great to fix this before the next release so I don't have to end up carrying patches in conda.

michaelld, 2021-09-25:
Thinking NO, and also that your fix from #4814 should do the trick. Have you made a PR for that yet?

ryanvolz, 2021-09-25:
I got subsequent reports from the user who was testing the patch from #4814 that it actually didn't fix #5085, neither with a hardcoded scale factor as in the PR or with the call to gtk to get the scale factor. Moreover, that gtk function was returning a scale factor of 1 when we might have expected it to be 2 in that case. I'm really hampered by not being a Mac user myself or having any way to test it. ☹️

That's not to say that it doesn't solve *this* bug, if it is possibly different from #5085. So if someone thinks that #4814 or my modification of it is still useful, then I can easily put in a PR.

geraman21, 2021-09-29:
I have tried running GRC with QT_AUTO_SCREEN_SCALE_FACTOR=1   env variable, but the bug is still there. When trying to increase the size past 960 resolution pixels in width or height with connected blocks on the flow-graph it freezes
and rather that resizing flowgraph starts to stretch. Shrinking back below 960 res px pops it back to normal.

ryanvolz, 2021-09-29:
@geraman21 Trying to follow the discussion here and on matrix, and it's not clear to me: did you try the patch written out in https://github.com/gnuradio/gnuradio/pull/4814#issuecomment-925190019? And you can confirm that doesn't help? How about hard-coding in the value of 2 as with the original #4814?

dkozel, 2021-09-29:
@ryanvolz geraman21 doesn't (yet) have a way of compiling GNU Radio from source and trying the different patches.

ryanvolz, 2021-09-29:
> @ryanvolz geraman21 doesn't (yet) have a way of compiling GNU Radio from source and trying the different patches.

Got it. For these patches though, it is sufficient to modify the Python module in-place wherever it is installed, e.g. `$PREFIX/lib/python3.9/site-packages/gnuradio/grc/gui/DrawingArea.py`.

michaelld, 2021-09-29:
@ryanvolz do you know how to trigger this issue? I'm running GRC 3.8.3.1 + Python 3.7 -x11 +Quartz on ARM64 (all built from source) with full sized display and it looks fine and doesn't freeze.

ryanvolz, 2021-09-29:
> @ryanvolz do you know how to trigger this issue? I'm running GRC 3.8.3.1 + Python 3.7 -x11 +Quartz on ARM64 (all built from source) with full sized display and it looks fine and doesn't freeze.

Not a clue. Maybe playing with setting `GDK_SCALE` in the environment before running GRC can trigger things. I've done that on my Linux machine and it just makes the GUI big or small, no crashes, but I don't have a high DPI screen.

michaelld, 2021-09-29:
hmmm ... OK let me try my laptop & see what happens. I'm using VNC / RDP to the M1, which might impact the display size and screen DPI.

geraman21, 2021-09-29:
@ryanvolz No I didnt try the change they had in the commit. I need to build gnu radio from source if I want to do that, and up until now I was using Brew version. Or do you mean that I can change the value in the installed brew GR ?

michaelld, 2021-09-29:
@geraman21 he's meaning you can edit the installed brew GR

michaelld, 2021-09-29:
I'm running GRC +Quartz -X11, and I do not experience this issue on any of my Macs. Some are older & one is current MacBook Pro 16" and another is M1 Mini. Let me try going for GRC +X11 to see if that changes things.

geraman21, 2021-09-29:
Solution proposed here fixed my issue on MBP 16' 2019: 
https://github.com/gnuradio/gnuradio/pull/4814#issuecomment-925190019

PR #5111 (2021-09-30): grc: Include scale factor when computing drawing area size
Fixes #4174.

Re-mix of #4814.
commit 59d04f4a (2021-09-30): grc: Include scale factor when computing drawing area size.

Fixes #4174.
closed 2021-09-30 by mormj