gnuradio/gnuradio #4177: stream_mode_t API (from stream_cmd_t) not exposed in gnuradio uhd module

than97, 2021-02-02:
In transitioning from Python 2.7/ GNURadio 3.7.9 to Python 3/ GNURadio 3.10, the stream_mode_t API (from stream_cmd_t) is no longer exposed in gnuradio uhd module.  It does not appear as though the gnuradio uhd module fully exposes the public API for the stream command. This results in breaks between 3.7 and 3.10. The desired outcome is that the gnuradio uhd block correctly exploses all the public API's, as it previously did, using swig.

Command Input: 
from gnuradio import uhd
uhd.stream_cmd.__dict__

Python 2.7/ GNURadio 3.7.9 Output: 
dict_proxy({'STREAM_MODE_NUM_SAMPS_AND_MORE': 109, '__module__': 'gnuradio.uhd.uhd_swig', 'stream_mode': <property object at 0x7f287b5b2a70>, 'thisown': <property object at 0x7f287b5b2a10>, 'STREAM_MODE_START_CONTINUOUS': 97, 'STREAM_MODE_STOP_CONTINUOUS': 111, '__repr__': <function _swig_repr at 0x7f287afcd2d0>, 'num_samps': <property object at 0x7f287b5b2290>, 'STREAM_MODE_NUM_SAMPS_AND_DONE': 100, '__dict__': <attribute '__dict__' of 'stream_cmd_t' objects>, 'time_spec': <property object at 0x7f287b5b2710>, 'stream_now': <property object at 0x7f287b5b2470>, '__weakref__': <attribute '__weakref__' of 'stream_cmd_t' objects>, '__doc__': 'Proxy of C++ uhd::stream_cmd_t class.', '__init__': <function __init__ at 0x7f287aaaf050>, '__swig_destroy__': <built-in function delete_stream_cmd_t>})

Python 3/ GNURadio 3.10: 
mappingproxy({'__init__': <instancemethod __init__ at 0x7f396f4c90d0>, '__doc__': None, '__module__': 'gnuradio.uhd.uhd_python', 'num_samps': <property object at 0x7f396f70d1d0>, 'time_spec': <property object at 0x7f396f70d2c0>, 'stream_now': <property object at 0x7f396f70d3b0>})

dkozel, 2021-02-03:
I may have tagged this as a bug too fast. The stream_cmd_t struct is bound in pybind11. I'm not in front of a computer with GNU Radio at the moment so can't check, but it looks like it should be working.

https://github.com/gnuradio/gnuradio/blob/master/gr-uhd/python/uhd/bindings/uhd_types_python.cc#L93-L100

vmw, 2021-02-05:
I've briefly looked over that file, and it appears that it's not fully defined. In particular, the API shows that stream_mode_t is the enum consisting of the various stream modes ([as shown in the API](https://files.ettus.com/manual/structuhd_1_1stream__cmd__t.html#a4df1f2e22148b7e09ace0eca0dfbf904)).

Moreover, if I look for the any of the enums, like, "STREAM_MODE_START_CONTINUOUS", it doesn't appear within the python bindings. 

I suspect that if we were to fully define stream_mode_t, along with the provided enums, everything should work.

vmw, 2021-02-10:
Hi,

We've also just realized that the public types are also not defined for;

enum uhd::tune_request_t::policy_t
enum | policy_t { POLICY_NONE = int('N'),  POLICY_AUTO = int('A'),  POLICY_MANUAL = int('M')  }

enum uhd::stream_cmd_t::stream_mode_t
enum | stream_mode_t { STREAM_MODE_START_CONTINUOUS = int('a'),  STREAM_MODE_STOP_CONTINUOUS = int('o'),  STREAM_MODE_NUM_SAMPS_AND_DONE = int('d'),  STREAM_MODE_NUM_SAMPS_AND_MORE = int('m')  }

This causes a regression, relative to previous versions of gnuradio, because these bindings aren't defined.

I would try fixing this myself, but I haven't previously done this, and I'm not sure the best way forward; if you can show me one of them, then I can take a stab at fixing the remaining ones.

dkozel, 2021-02-12:
Thanks @vmw that clarifies things. These should be bound in the `uhd_types_python.cc` file. Here's the documentation on enums.
https://pybind11.readthedocs.io/en/stable/classes.html#enumerations-and-internal-types

@mormj is the resident expert on pybind if you aren't able to get it working.

85ayush, 2021-02-23:
I would love to work on this issue, I am new to open source contribution, can you please guide me?

than97, 2021-02-26:
Our build on Archlinux is run through a Jenkins pipeline using a Dockerfile script. I entered the container through a bash terminal and was unable to locate uhd_types_python.cc. How is this file generated? It may be possible that a specific dependency responsible for creating this file is not currently being installed in our build. I can forward a copy of the Dockerfile and PKGBUILD used if that would help to troubleshoot.

vmw, 2021-03-01:
I've taken an initial stab at this, and believe we've successfully instantiated the define, but I'm getting a bit of a cryptic error when trying to pass the enum, as per the pybind documentation.

```
/home/notroot/gnuradio/src/gnuradio/gr-uhd/python/uhd/bindings/uhd_types_python.cc: In function ‘void bind_uhd_types(pybind11::module&)’:
/home/notroot/gnuradio/src/gnuradio/gr-uhd/python/uhd/bindings/uhd_types_python.cc:105:43: error: expected primary-expression before ‘(’ token
  105 |     py::enum_<stream_cmd_t::stream_mode_t>(stream_cmd_t, "stream_mode_t")
      |                                           ^
/home/notroot/gnuradio/src/gnuradio/gr-uhd/python/uhd/bindings/uhd_types_python.cc:105:56: error: expected primary-expression before ‘,’ token
  105 |     py::enum_<stream_cmd_t::stream_mode_t>(stream_cmd_t, "stream_mode_t")
      |                                                        ^
make[2]: *** [gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/build.make:82: gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/uhd_types_python.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7599: gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/all] Error 2

```
Some preliminary googling suggests the issue may be related to how the function is defined, and may require overriding pybind, but I haven't quite figured out exactly what I'm doing wrong. I suspect that perhaps I should be referencing the object instead of the type, but I'm a bit stuck.

Below, please see the patch file that shows the work so far; it results in the errors above, and I would definitely appreciate some insight into why it's not working.

------------

```

From 19351d561616a5dd3348188ac731a4d751d61ef4 Mon Sep 17 00:00:00 2001
From: Victor W <nobody@nobody.com>
Date: Mon, 1 Mar 2021 18:23:02 +0000
Subject: [PATCH] UHD: Add additional UHD python bindings.

---
 gr-uhd/python/uhd/bindings/uhd_types_python.cc | 12 ++++++++++++
 1 file changed, 12 insertions(+)

diff --git a/gr-uhd/python/uhd/bindings/uhd_types_python.cc b/gr-uhd/python/uhd/bindings/uhd_types_python.cc
index 13f92c748..78142ea1a 100644
--- a/gr-uhd/python/uhd/bindings/uhd_types_python.cc
+++ b/gr-uhd/python/uhd/bindings/uhd_types_python.cc
@@ -91,13 +91,25 @@ void bind_uhd_types(py::module& m)
         .def(py::self - double());
 
     // ::uhd::stream_cmd_t
     using stream_cmd_t = ::uhd::stream_cmd_t;
     py::class_<stream_cmd_t>(m, "stream_cmd_t")
         .def(py::init<stream_cmd_t::stream_mode_t>())
         // Properties
+        .def_readwrite("stream_mode", &stream_cmd_t::stream_mode)
         .def_readwrite("num_samps", &stream_cmd_t::num_samps)
         .def_readwrite("time_spec", &stream_cmd_t::time_spec)
         .def_readwrite("stream_now", &stream_cmd_t::stream_now);
+    
+    py::enum_<stream_cmd_t::stream_mode_t>(stream_cmd_t, "stream_mode_t")
+        .value("97", stream_cmd_t::stream_mode_t::STREAM_MODE_START_CONTINUOUS)
+        .value("111", stream_cmd_t::stream_mode_t::STREAM_MODE_STOP_CONTINUOUS)
+        .value("100", stream_cmd_t::stream_mode_t::STREAM_MODE_NUM_SAMPS_AND_DONE)
+        .value("109", stream_cmd_t::stream_mode_t::STREAM_MODE_NUM_SAMPS_AND_MORE)
+        .export_values();
+ 
     // ::uhd::meta_range_t
     using meta_range_t = ::uhd::meta_range_t;

```

mormj, 2021-03-01:
A couple of things would keep that from working:
1) you need the enum to bind to a module, so `m` should be the first argument, not the class name
2) The quoted names of the enums should pa to the STREAM_MODE_... enum names

Something like this:
```
    // ::uhd::stream_cmd_t
    using stream_cmd_t = ::uhd::stream_cmd_t;
    py::class_<stream_cmd_t>(m, "stream_cmd_t")
        .def(py::init<stream_cmd_t::stream_mode_t>())
        // Properties
        .def_readwrite("num_samps", &stream_cmd_t::num_samps)
        .def_readwrite("time_spec", &stream_cmd_t::time_spec)
        .def_readwrite("stream_now", &stream_cmd_t::stream_now)
        .def_readwrite("stream_mode", &stream_cmd_t::stream_mode);

    py::enum_<stream_cmd_t::stream_mode_t>(m, "stream_mode_t")
        .value("STREAM_MODE_START_CONTINUOUS", stream_cmd_t::stream_mode_t::STREAM_MODE_START_CONTINUOUS)
        .value("STREAM_MODE_STOP_CONTINUOUS", stream_cmd_t::stream_mode_t::STREAM_MODE_STOP_CONTINUOUS)
        .value("STREAM_MODE_NUM_SAMPS_AND_DONE", stream_cmd_t::stream_mode_t::STREAM_MODE_NUM_SAMPS_AND_DONE)
        .value("STREAM_MODE_NUM_SAMPS_AND_MORE", stream_cmd_t::stream_mode_t::STREAM_MODE_NUM_SAMPS_AND_MORE)
        .export_values();
```

vmw, 2021-03-01:
@mormj 

Thanks for the prompt reply; it helped quite a bit. I've just made a pull request against master for this;

mbr0wn, 2021-03-04:
https://github.com/gnuradio/gnuradio/pull/4293 has been merged, closing.

closed 2021-03-04 by mbr0wn