gnuradio/gnuradio #4178: Can't access the blocks which has accidentally moved beyond left or top boundary

surpscare, 2021-02-03:
The new interface has the ability to create flow graphs of unlimited size by adaptive canvas expanding. However, the canvas can't expand to the left or up, and there is no "overshoot protection" for blocks being moved beyond the left or top boundary. This leads to losing access to the block groups which have been moved too far because the screen can not scroll to them being visible. I want to delete and re-create them, but I can't select them from GUI. Is there any way to edit the GRC file correctly without breaking it? Also, boundary protection needs to be added.
[image]

dkozel, 2021-02-03:
That's a bit funny. Thank you for the report. This seems like something that can be fixed pretty easily with a check for negative coordinates.

To manually edit the file search for all `coordinate: [7, 6]` parameters and make any negative values 0 or positive.

I don't have a running copy of GNU Radio right at the moment, but can you use Ctrl-a to select all the blocks and drag them?

dkozel, 2021-02-13:
Looks like we can sanitize the coordinates here.
https://github.com/gnuradio/gnuradio/blob/master/grc/gui/canvas/block.py#L61

85ayush, 2021-02-23:
I would love to work on this issue, I am new to open source contribution, can you please guide me?

jasonuher, 2021-02-24:
@85ayush the minimum viable solution would be to edit the linked section of code as described by @dkozel. You could add bounds checking on the coordinates and limiting their range prior to returning.  That is "if less than 0, set to 0".  This would have the effect that all infringing blocks would be clumped at the edges.  It's not ideal, but it's better than having to manually edit the *ML to find your blocks.

PR #4334 (2021-04-06): gui : fix block can move outside top and left boundary

This fixes #4178 
Tested on GR 3.8.2 
If ever a block have been saved with negative coordinate, coordinates are sanitized when opening the flowgraph
This can be backported to maint-3.8 and maint 3.9 
As compared to Derek proposal, the coordinate sanitizing is made inside `scale(coor, reverse=False):` method which is only called in https://github.com/gnuradio/gnuradio/blob/master/grc/gui/canvas/block.py#L61
closed 2021-04-06 by mormj