gnuradio/gnuradio #4179: FFT block is broken

surpscare, 2021-02-03:
Reproducible with even the simplest flowgraph given below. FFT block always returns the error:
```
self.fft_vxx_0 = fft.fft_vcc(2048, False, , True, 4)
                                              ^
SyntaxError: invalid syntax

>>> Done (return code 1)
```
FFT block had to work under version 2.7x, does not work in 3.8+
[image]

dkozel, 2021-02-03:
You can supply a rectangular window using `numpy.ones(fft_size)`.

If the fft implementation allows it could be nice to explicitly have an option for no windowing and to skip those multiplies. It should default to a rectangular window if none is supplied in the block, or at least the GRC block should show an error if that param is empty.

willcode, 2021-02-03:
`[]` is allowed, and will bypass the multiply.

dkozel, 2021-02-12:
I don't think we can do better in this situation. Adding a default parameter to the python and/or C++ FFT function won't work because GRC uses positional arguments rather than named ones.

We have a rectangular window in the library and if an empty array is provided the multiplies will even be skipped. I've added a note to the wiki documentation about the latter.
https://wiki.gnuradio.org/index.php/FFT

I'm going to close this, but please let us know if you think there's a better option.

closed 2021-02-12 by dkozel