gnuradio/gnuradio #4184: Embedded Python blocks can not use gr.io_signature()

willcode, 2021-02-04:
GRC assumes that embedded Python blocks use the list-style input/output signature. The user sees:

  Can't create an instance of your block: object of type 'io_signature_sptr' has no len()

when attempting to use gr.io_signature() for in_sig or out_sig.

willcode, 2021-02-04:
Since there is no yml file associated with an embedded Python block, fixing this would require GRC to parse the the io_signature AND provide num_inputs/num_outpus where needed in the params dialog.

Along the same lines, it would be nice if data type could be changed from the dialog. Unfortunately, there is nowhere to specify what data types the block can accept, so no clue how this would be done.

marcusmueller, 2024-01-28:
Removing the bug tag here. It's OK that embedded blocks aren't fully-featured. The list-style iosig def is definitely a good start, and the rest can be added through enhancements