gnuradio/gnuradio #4185: Generation of sine_table.h with CMake doesn't work on MacOS

dkozel, 2021-02-05:
Reported in a comment on a commit by @ulrichloose

https://github.com/gnuradio/gnuradio/commit/c9b01b8409b5fcfa12154898bfedc45fed7f5354

@michaelld or @ryanvolz do you see this in your builds?

ryanvolz, 2021-02-05:
I've started a build to test it, but I expect it will fail on macOS as the comment indicated and it will certainly fail on Windows since /dev/stdin won't exist. I'm trying to think of a universal way to do this with CMake.

ryanvolz, 2021-02-05:
> I've started a build to test it, but I expect it will fail on macOS as the comment indicated and it will certainly fail on Windows since /dev/stdin won't exist. I'm trying to think of a universal way to do this with CMake.

I can confirm that the build fails on macOS and Windows, but fortunately this has already been fixed: b3f2edadb719c7d639fb48e48368121ec6eee911.

closed 2021-02-05 by ryanvolz