gnuradio/gnuradio #4191: Message Strobe/Debug Error?

bmagistro, 2021-02-08:
Was looking to use the `message strobe` block to send a periodic message for testing.  While setting up and validating things, had the output connected to a `message debug` block, specifically the `print_pdu`.   When trying to run, it throws the following error: `thread[thread-per-block[0]: <block message_debug(5)>]: pmt_car: wrong_type : TEST`.

Attached is a minimal flow graph that can replicate this issue.
[strobe.grc.txt](https://github.com/gnuradio/gnuradio/files/5940659/strobe.grc.txt)

GR 3.8.1 via pybombs
Ubuntu 18.04

willcode, 2021-02-08:
Message PMT should be of the form `pmt.to_pmt(['meta','vector'])`.

willcode, 2021-02-08:
Here's one that actually works: `pmt.cons(pmt.PMT_NIL, pmt.init_u8vector(3,[0x01,0x02,0x03]))`.

bmagistro, 2021-02-09:
Thanks for the insight, while I can manually transform a string to a vector easy enough, the defaults for the blocks seem to say a string should work.  Your comments also make me wonder if the defaults for the `message strobe` and `message strobe random` blocks are wrong then?

Will keep looking at it later.

dkozel, 2021-02-12:
The Message Strobe default value correctly displays if you connect it to the Message Debug print port.

Some nice improvements to the Message Debug blog were added to GNU Radio 3.9 in October.
https://github.com/gnuradio/gnuradio/commit/9248b29768c7810f029a1188726b2ad05141f382

This includes a check on the `print_pdu` port which discards messages that aren't pairs (or are null). It does discard silently. I think logging something would be useful. The older behavior was a bit obscure, but did print a technically useful message.

I think that there is no error in the latest code. Possibly back-porting the discard behavior and adding a print statement "Non-PDU message received" or similar would be a useful enhancement. What do you two think?

jacobagilbert, 2021-02-12:
I think adding a "non PDU message received on port print_pdu" or something is reasonable. This should also be updated to use the [3.9+ is_pdu() check](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/pmt/pmt.cc#L821) that was added. The current check is more restrictive than what is in this block as:

```
 A PDU (protocol data unit) in GNU Radio has a special PMT type, it is a pair of a dictionary (on CAR) and a uniform vector type.
```

https://wiki.gnuradio.org/index.php/Message_Passing#PDUs

bmagistro, 2021-02-13:
So things are making a little more senses now.  I had been under the impression that all messages are PDUs.  It probably didn't help that I was also looking at the output of `socket pdu` in `message debug`.  From the additional discussion, this does not seem to be the case and instead not all messages are PDUs but all PDUs are messages?

If the above is true, when should you use a non-PDU message and when should you use a PDU message?  Would simplifying it as input blocks (tempted to say sinks, but don't think it can be generalized that much) should be a non-PDU input and commands + processing block inputs (again might be over generalizing) should be a PDU?  I probably need to reread the wiki page again, since I felt like the terms `message` and `PDU` may have been used a little interchangeably at times so I probably missed some details.

I had been basing the input to the block I am working on off of `PDU to tagged stream` but it seems like if I want to support a generic "message input", the input should be modeled closer to that of `set_msg` on the `message strobe` block or the input to `ZMQ PUSH msg sink` where it doesn't need to be a PDU.

jacobagilbert, 2021-02-13:
@bmagistro A message refers to any PMT object passed via the asynchronous messaging API. PDUs are a subset of messages which specifically refer to messages that are PMT pairs that consist of a PMT dictionary and PMT uniform vector.

PDUs are convenient for passing around data and accompanying metadata to provide necessary context to the data. This metadata can be aggregated as the PDUs are processed. If a message passed the `is_pdu()` check it will be safe to process the message like a PDU; for example extract the metadata/data with car()/cdr().

Your immediate issue should be resolved on master, maint-3.9 and maint-3.8 so I am going to close this -- thanks for opening this.

PR #4219 (2021-02-13): gr-blocks: Sanitize the Print PDU input of the message_debug block
No message received on this port should cause the block to crash, currently a non-pair message will.

Fixes #4191

This is not a backport as this is not an issue in 3.9/3.10
closed 2021-02-13 by jacobagilbert