gnuradio/gnuradio #4196: Embedded Python Block - Cannot make variable input/output length

Djgreene730, 2021-02-09:
Background: I'm using GNURadio 8.2.0 on Ubuntu 18.04

I'm trying to create an Embedded Python Block in my GRC flowgraph. I want a variable to control the vector size of the input and output (`vlen` in my example). I whipped up the following code, but **GRC always uses the default of `vlen` as the vector size** of the input and output (1 in my example). I've tried a couple of combinations, but no luck. Any thoughts?

```python
class blk(gr.sync_block):
    def __init__(self, vlen=1):
        gr.sync_block.__init__(
            self,
            name='Simple Example',
            in_sig=[(np.float32, vlen), (np.float32, 1)],
            out_sig=[(np.float32, vlen)]
        )
```

Djgreene730, 2021-02-10:
I see now that the epy `__init__(self, vlen=1)` I've written is called with the defaults; however, GRC does not update the port signatures after the variables have been changed (i.e. when a user updates a value in the block properties window).

I need to have the port signature change when GRC sees that the variables are updated. I'll try to look for this, but I'd appreciate some guidance on where this can be found. Thanks in advance!

willcode, 2021-02-12:
Related to https://github.com/gnuradio/gnuradio/issues/4184.

Djgreene730, 2021-03-03:
I'm able to use gr.io_signature(); I don't think this isn't related to #4184.

I found a workaround. TLDR: Use inspect to find a top-level variable and set the port size using that.

Here's a simple example - A flowgraph with an embedded python block where the input vector size is adjusted with a top-level variable "vlen". The surrounding "Stream to Vector" and "Throttle" blocks are controlled using the variable "vlen_other". A class method in the embedded python block uses inspect to search for a variable block in the top flowgraph with the name "vlen" and applies the value as the input port length. There are three situations when it could find the variable: (1) compilation of the python block, (2) making alterations to the flow graph, or (3) when the flowgraph is being generated immediately before execution; I've handled all three.

Example Python Block:
```python
"""
Demo: Find a top-level variable name and use it's value to set my vlen.

Greene, 2021
"""

import numpy as np
from gnuradio import gr

# Includes for finding top variables
from gnuradio import grc
import inspect

def get_top_variable(variable_name="", default=None):
    '''
    Returns the value of a variable from the flow graph.
    '''

    # Let's go!
    try:
        ## Run Condition: GNURadio is starting the flowgraph
        top = inspect.currentframe().f_back.f_back.f_locals

        # Check if top has the variable name we're looking for
        if top.__contains__(variable_name):

            # Let the user know we found the variable
            print(f"[Block Debug] While starting, I found top variable \'{variable_name}\': type={type(top[variable_name])}, value={top[variable_name]}")

            # Return the value of the variable name
            return top[variable_name]

        ## Run Condition: Saving the flowgraph
        top = inspect.currentframe().f_back.f_back.f_back.f_back.f_back.f_locals

        # Make sure top has 'self'
        if top.__contains__("self") and \
                (isinstance(top['self'], grc.gui.canvas.flowgraph.FlowGraph)) and \
                (hasattr(top['self'], 'blocks')):

            # Get a list of all blocks
            block_names = [block.name for block in top['self'].blocks]

            # Find the index to the variable we need
            block_index = block_names.index(variable_name)

            # Store the result
            result = top['self'].blocks[block_index].params['value'].value

            # Let the user know we found the variable
            print(f"[Block Debug] While modifying the flowgraph, I found top variable \'{variable_name}\': type={type(result)}, value={result}")

            # Return the value of the variable we found
            return result

    # Something went wrong
    except Exception as e:
        print("[Block Debug]: Threw an exception", e)
    except:
        print("[Block Debug]: Threw an unknown exception.")

    # Return the default result
    return default

class blk(gr.sync_block):  # other base classes are basic_block, decim_block, interp_block
    """
    Top Variable Block Demo

    Add a Variable block to the top flow-graph called "vlen" and set an int length.
    """

    def __init__(self, example_param=1.0):  # only default arguments here

        # Get packet length
        vlen = get_top_variable("vlen", default=1)
        print(f"[Block Debug] Setting input length using \'vlen\': {vlen}, type={type(vlen)}")

        """arguments to this function show up as parameters in GRC"""
        gr.sync_block.__init__(
            self,
            name='Top Variable Block Demo',   # will show up in GRC
            in_sig=[(np.complex64, vlen if isinstance(vlen, int) else int(vlen))],
            out_sig=[np.complex64]
        )
        # if an attribute with the same name as a parameter is found,
        # a callback is registered (properties work, too).
        self.example_param = example_param

    def work(self, input_items, output_items):
        """example: sum all inputs and multiply with constant"""
        output_items[0] = np.sum(input_items[0], axis=1) * self.example_param
        return len(output_items[0])
```

When the variable block "vlen" is not present, it uses a default value of 1.
[image]

Once I add the variable block "vlen" with value 10, the input length of the python block updates to 10. (Notice the connection error; the output size of the Throttle block is 10 and the input size of the python block is 1)
[image]

And when I change "vlen_other" to match "vlen", everything jives.
[image]

I can also update the value of "vlen" and it immedeately changes the input size of the example python block.

I'm not closing this because it's still a bug that should be resolved, but I hope this helps someone else.

willcode, 2021-03-04:
Thanks for the sample code. That is a noble amount of frame backtracking and variable searching.

I can't remember - did you already try passing vlen in as a parameter to the embedded block?

Djgreene730, 2021-03-04:
> I can't remember - did you already try passing vlen in as a parameter to the embedded block?

Yes, but the parser only uses the default values in the init function and does not update the input/output port sizes when an entry is made or altered. That's the real bug.

willcode, 2021-03-04:
Making sure I'm seeing the same thing you are. Passing vlen as a parameter works correctly when you run the flowgraph. But when assigned via a variable, you do not get visual feedback in GRC that the I/O sizes will be wrong. Instead, you have to wait until runtime to see the error and get something like

`RuntimeError: itemsize mismatch: throttle0:0 using 8, Embedded Python Block0:0 using 80`

Djgreene730, 2021-03-04:
> Passing vlen as a parameter works correctly when you run the flowgraph.

Not quite. If the default assignment (in the `__init__` method) for vlen is not what I need, it won't work. If it's what I need (i.e. the default matches the input and output lengths of the connected block) then it works.

Here's another simple example. I've create a block where the input size is set by the parameter `vlen` and I've set it to a default of 1.

```python
class blk(gr.sync_block):  # other base classes are basic_block, decim_block, interp_block
    """Embedded Python Block example - a simple multiply const"""

    def __init__(self, vlen=1, example_param=1.0):  # only default arguments here
        """arguments to this function show up as parameters in GRC"""
        self.vlen = vlen
        gr.sync_block.__init__(
            self,
            name='Embedded Python Block',   # will show up in GRC
            in_sig=[(np.complex64, vlen)],
            out_sig=[(np.complex64, vlen)]
        )
        # if an attribute with the same name as a parameter is found,
        # a callback is registered (properties work, too).
        self.example_param = example_param

    def work(self, input_items, output_items):
        """example: multiply with constant"""
        for input_num in range(self.vlen):
            output_items[0][:] = input_items[0] * self.example_param
        return len(output_items[0])
```

So I don't mix up things in this example, I'm using a top-level variable block called `length` to set the output length of a vector source and the input length of a null sink. For the python block, I've set vlen to length.
[image]
[image]

This setup does not work, even if I manually put 10 into the vlen field of the block - it only works if I set 10 as the default of vlen in `__init__` or use my inspect approach.

Now, if I change the default for `vlen` to 10 in the `__init__` method, it all works fine...
```python
...
def __init__(self, vlen=10, example_param=1.0):  # only default arguments here
...
```
[image]

But it completely ignores any manual changes using the Properties box! It only sees the mismatch once I run the flowgraph.
[image]
[image]

Here's the GRC file:
[simple_example.zip](https://github.com/gnuradio/gnuradio/files/6080815/simple_example.zip)

willcode, 2021-03-04:
I had added `self.vlen = vlen`, like `self.example_param = example_param`. Does that change anything?

Djgreene730, 2021-03-04:
It doesn't for me.

In your example, `vlen` requires a default value in the `__init__` constructor. When GNUradio checks the I/O signature after the python block has been modified, it always uses the default value. I see this even if I set `vlen` to a good value in the block property editor in the flowgraph.

willcode, 2021-03-04:
@skoslowski do you or anyone else more familiar with GRC have any ideas here?

skoslowski, 2021-03-04:
I am fairly certain this is currently not supported (I originally hacked this together).
The way it works is not by parsing source code, but by creating an actual block instance and inspect that. 
This is always done with default arguments (which is why they are required). I realize now, that this is not explicitly stated in the default docstring.

Anyway, may I kindly suggest to stop putting efforts in work-arounds and instead add this the [block IO extraction code](https://github.com/gnuradio/gnuradio/blob/master/grc/core/utils/epy_block_io.py). One would need to get the current parameter values, get them evaluated and pass them to the extractor.

Djgreene730, 2021-03-04:
I'll take a look at the block IO extraction code, Sebastian. Thanks, I
think this is what I was looking for.

tclarke, 2021-04-10:
https://github.com/gnuradio/gnuradio/pull/4506

Fixed in pull #4506

21salvador86, 2021-12-05:
what about for c++? gr-keyfob parser_packet.cc? how to read input or output data with python?

xeno00, 2022-07-12:
Any more work being done to support this functionality? Still having the same issue here in 3.10.3.

> Not quite. If the default assignment (in the `__init__` method) for vlen is not what I need, it won't work. If it's what I need (i.e. the default matches the input and output lengths of the [...]

AlienXandr, 2023-11-19:
Background: GNURadio 3.10.8.0 on Ubuntu 23.10

### Variant 1 - Create *.py and *.block.yml files
I found a simple solution for me.
1. We need to save the Python code into a separate file "name.py".
2. For the "name.py" file, we need to create a file "name.block.yml" with a description of the inputs, outputs and category of this block.
3. Copy these two files to the "home/user/.grc_gnuradio" folder (we must create a folder ".grc_gnuradio" if it does not exist)
4. In GNU Radio, click the "Reload Blocks" button and find our block in the "myCategory" section.

Now in more detail.

Add "Embedded Python Block" to the diagram.

[image]

We open its properties and open its code in the editor.
[image]

Next, we save Python Code into a separate file, clean the comments and change the python code.
[image]

After creating the "my_block.block.yml" file and describe the inputs, outputs, imports, etc.
[image]
**NOTE 1**: Category can be any, for example, "[MyPyBlocks]" or "[MyPyBlocks]/Filters" ('[Level 1 Category]/Level 2 Category' [link](https://wiki.gnuradio.org/index.php/YAML_GRC#Category_(optional)))

**NOTE 2**: Do not delete the comment in the import "# grc-generated hier_block"! Using this comment, GNURadio will automatically add the folder ".grc_gnuradio" to the list for searching for Python modules. Otherwise there will be a mistake:
"Traceback (most recent call last):
   File "~/test/main.py", line 25, in \<module\>
     from my_block import blk
ModuleNotFoundError: No module named 'my_block' "

Copy these two files to the "home/user/.grc_gnuradio" folder and in GNU Radio, click the "Reload Blocks" button and find our block in the "[MyPyBlocks]/Filters" section.
Next, we create a simple scheme. We also create a common variable "vlen" for all blocks.
**TEST 1**: vlen = 1024
[image]

**TEST 2**: vlen = 64
[image]

### Variant 2 - Change the common variable in main.py
Create a similar scheme with Embedded Python Block. 
In the Python code, add the length equal to the common variable .
`def __init__(self, vlen=64):`
[image]

After starting GNURadio compiles the scheme in main.py.
[image]

Now we change the common variable in main.py and launch main.py only through the console.
[image]
[image]

**NOTE**: Variant 2 is suitable only for a finished scheme. With each change in the scheme in main.grc GNURadio will compile the new main.py.

Examples of files you will find here
[example.zip](https://github.com/gnuradio/gnuradio/files/13403053/example.zip)

marcusmueller, 2024-01-28:
Workaround exist; it's pretty engrained in how GRC deals with embedded python blocks that evaluations aren't possible to calculate a parameter-dependent io size: I'm taking the freedom to declare this as a feature we can't support for now.

closed 2024-01-28 by marcusmueller