gnuradio/gnuradio #4203: ZMQ Message Sinks fail across computers using master branch

duggabe, 2021-02-10:
Using ZMQ Message Sinks and Sources to pass messages from one computer to another fails using the master branch (v3.8.0.0-953-g6de23f43 built on Feb 3).

These tests work on maint-3.8 and maint-3.9, and they work if using the same computer. They only fail from one computer to another using the master branch.

Obviously one needs to change the IP addresses for the user's situation.
```
rPi_switch_test.grc   https://pastebin.com/grHgkRGX
rPi_switch_2.grc      https://pastebin.com/Cp33FEM5
```

[image]
---
[image]
---
Note: these pictures were taken during tests on different computers. Of course they wouldn't work together as is.

willcode, 2021-02-10:
I set up a remote console-only version of rPi_switch_2 and ran rPi_switch_test locally, both on master. Can not duplicate the problem.

duggabe, 2021-02-10:
These notes are from discussion in the Matrix #development:gnuradio.org room:
There were no error messages. Also I tested with PUSH/PULL with same results. Also ran rPi_switch_test on 3.8 machine and rPi_switch_2 on master. In that case, the message was received but no echo came back.
So it looks like it's just the PUB or PUSH (i.e. the outgoing) blocks which don't work.
One additional note: the original rPi code was straight Python, not created by a flowgraph. it works with 3.8 and 3.9

I will load a new master branch in a few days and test again. I wiped out the old one when I built the maint-3.9.

duggabe, 2021-02-11:
Built version: v3.8.0.0-965-g404d09cf / 3.10.0git from the master branch.
Repeated the tests above. Everything works correctly.
Apparently some stray bug snuck in before - one of those 'I don't know what happened' issues.
I am closing this.

closed 2021-02-11 by duggabe