gnuradio/gnuradio #4209: gr.hier_block2 does not inherit basic_block properties

mormj, 2021-02-11:
```
>>> dir(gr.hier_block2)
['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', 'connect', 'disconnect', 'message_port_register_hier_in', 'message_port_register_hier_out', 'msg_connect', 'msg_disconnect']
```

So there is currently no way in python to `set_msg_handler`

mormj, 2021-02-12:
It has something to do with how the hier block's pybind interface is wrapped into python - but doesn't have the same gateway block properties that a regular python block has.  This needs some thought but the end result is you can't have message ports in hier blocks currently.

bistromath, 2021-02-13:
clarification: you can't have message input ports on hier blocks. Registering message output ports and posting messages to those ports seems to work fine.

duggabe, 2021-02-24:
I'm not sure what the conditions are, but I am running an embedded python block with a message input port. See https://github.com/duggabe/gr-control/blob/main/xmt_rcv_switch_epy_block_0.py
On Feb 19, I built GNU Radio Companion v3.10.0.0git-155-g8e6f3482 from source.

mormj, 2021-02-24:
It is only a bug for Hier Blocks as far as i know.  

That is good confirmation that message input ports are working for the embedded python blocks.

marcusmueller, 2024-01-28:
persists.

marcusmueller, 2024-02-21:
@krono-i2 no, see the second entry: this is due a shortcoming of the bindings. PRs welcome!