gnuradio/gnuradio #4214: Python block signature inconsistency

willcode, 2021-02-12:
The variety of ways to specify in/out signatures in Python is confusing, and possibly broken. Collecting thoughts here.

Python `hier_block2` takes `io_signature`, wrapped by pybind11.
Python `basic_block` takes `py_io_signature` or a list of types, pure python.
The `block_gateway` passes `io_signature`.
GRC embedded Python blocks use "list of types".

`io_signature` allows only one type for all inputs
'io_signature.make[2,3,4]` allow multiple types
`py_io_signature` allows an arbitrary number types of types (though 1-4 are likely sufficient)

`py_io_signature` is not used explicitly by in-tree Python blocks, but is the internal representation, translated from the commonly used "list of types" representation. Unknown whether OOTs use `py_io_signature`. Most probably use "list of types".

C++ blocks use `io_signature`, which carries through Python wrappers.

AkshayKumarTripathi, 2021-04-01:
Hello, I want to work on this, can you assign me this?

willcode, 2021-04-01:
Go ahead, there's no need for an assignment. I don't think anyone else is looking into this at the moment. Send updates or submit a PR as you make progress so you can get feedback as you go.

AkshayKumarTripathi, 2021-04-09:
> Go ahead, there's no need for an assignment. I don't think anyone else is looking into this at the moment. Send updates or submit a PR as you make progress so you can get feedback as you go.

SIr can you help me, how can I begin working on the issue and where do I find the code for this.

willcode, 2021-04-09:
Start with `gnuradio-runtime/python/gnuradio/gr/gateway.py` to understand how the Python end works. For the C++ end, look for files under `gnuradio-runtime` containing `gateway` or `io_signature`. Also, make sure you understand how/which C++ functionality is wrapped via PyBind11.