gnuradio/gnuradio #4223: Add translation support to GNU Radio

dkozel, 2021-02-13:
Currently GNU Radio is available only in English. GNU Radio Companion and the GUI blocks should support other languages. `gettext` appears to be a stable method of adding string translation to both C++ and Python programs.

This issue is to collect ideas, concerns, and interest in adding translation support and to track progress towards implementing it.

KiCad is doing a good job of translating it's GUI and uses the website Weblate to make it simple and easy for people to review and contribute translations. This doesn't prevent using any of the other po translation tools or workflows so there's no lock-in. They seem to offer support to FOSS projects.

`gettext` is included by default with Python. For C++ libintl.h is part of glibc.

Before making this a dependency we should figure out the workflow for coding and translating, code layout, CI and build impacts.

dkozel, 2021-02-15:
Michelle Thompson implemented internationalization of GRC at the ATA Hackfest in May 2019 and had German and Japanese sample translations. She used gettext for that work.
https://forum.amsat-dl.org/index.php?thread/285-internationalization-for-gnu-radio-companion-german-language-support/

Talking with (a very few) people it does not seem to be common for code arguments and values to be translated. Log messages, help text, and UI text all can and should be translated.

Translating the UI strings that are in block YAML files is not as straightforward as the Python or C++ code. I have a few ideas, but none are entirely straightforward and do add extra work. Having a GUI block description editor would help in several ways and make the extra boilerplate manageable. Possible GSoC project?

Doxygen documentation in the code can be translated, but may quickly become overwhelming in the source code as there isn't separation between the root language and translation strings as there is for gettext.
https://www.doxygen.nl/manual/commands.html#cmdtilde

Translating the website and wiki are desirable, but out of scope for this issue.

ChristopheSeguinot, 2021-03-02:
I've got very little experience in internationalization , here are my feeling about Translation for GNURadio. 

My experience is related to PHP plugin wrote for CMS such as Joomla for which I used transifex.com platform so that volunteer users could translate. Using CMS for constructing website, we obviously need translation. 

However, these plugins where 99% PHP/Javascript which allow using `gettext`. 

- I agree with Derek, Log messages, help text, and UI text all can and should be translated. 
- Documentation  including blocks documentation must probably not. It would drastically increase the amount of text to translate . Moreover, long sentences taken out their context are sometime impossible to translate. 
- I don't see how to do with .yml files (`gettext` seems of no help here)
- If ever we add translation support to GNURadio, we must find a unique platform for all translations

I've seen for various CMS that keeping translation up to date is a long process. Most of the time a complete translation is available for a few languages and only remains partial or obsolete for others even after one or two years. 

As a french user , I feel comfortable with the actual GNURadio and don't need a french translation. When I Don't understand a block behavior, it's due to a lack of scientific background, not to a translation problem. I would probably not be keen for translating some tutorial or documentation, It's more important to extend existing tutorial then translating existing one.. It's time consuming to maintain a documentation in several languages. 

As a teacher, I believe my students should be able to work with software and complex equipments without a translation. 

Regards, Christophe