gnuradio/gnuradio #4237: error: 'ALL_LOS' is not a member of 'uhd::usrp::multi_usrp' ?

ReverseControl, 2021-02-15:
I am trying to install gnuradio 3.9 in a docker container:

I get this error:

```
[ 88%] Built target pygen_gr_uhd_python_uhd_73240
[ 88%] Built target uhd_docstrings
[ 88%] Building CXX object gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/usrp_block_python.cc.o
/gnuradio/gr-uhd/python/uhd/bindings/usrp_block_python.cc: In function 'void bind_usrp_block(pybind11::module&)':
/gnuradio/gr-uhd/python/uhd/bindings/usrp_block_python.cc:26:58: error: 'ALL_LOS' is not a member of 'uhd::usrp::multi_usrp'
     m.attr("ALL_LOS") = py::str(::uhd::usrp::multi_usrp::ALL_LOS);
                                                          ^~~~~~~
gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/build.make:110: recipe for target 'gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/usrp_block_python.cc.o' failed
make[2]: *** [gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/usrp_block_python.cc.o] Error 1
CMakeFiles/Makefile2:5658: recipe for target 'gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/all' failed
make[1]: *** [gr-uhd/python/uhd/bindings/CMakeFiles/uhd_python.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

Any suggestions as to how to fix this? It is not a dependency issue, it doesn't seem like.

dkozel, 2021-02-15:
Hi @ReverseControl Yes, that was a bug that was fixed 10 days ago by this commit. It wasn't backported to 3.9 yet though!
https://github.com/gnuradio/gnuradio/commit/6df5b3a6e5397eb0b08dfabfcd78d96aea46f447

mbr0wn, 2021-02-15:
Yeah it is: 4fcb0c29da0c508230b9efe8a07ef46d4d0ff36a

dkozel, 2021-02-15:
Thanks for pointing that out Martin. I forgot that of course the hash would change during the backporting.

@ReverseControl Can you ensure that commit 4fcb0c2 exists in your source and try rebuilding? Or paste the output of `git describe`. I'm guessing you might be running the 3.9.0.0 release which is missing this fix.

ReverseControl, 2021-02-15:
@dkozel 

```
root@12a1674f5a13:/gnuradio/build# git describe
v3.9.0.0-49-gfbf34f222
```

which branch should I run? I was running the maint-3.9 branch.

ReverseControl, 2021-02-15:
That commit is in the branch I am building from(4fcb0c2); 

```
root@12a1674f5a13:/gnuradio/build# git show  4fcb0c2
commit 4fcb0c29da0c508230b9efe8a07ef46d4d0ff36a
Author: karel <5636152+karel@users.noreply.github.com>
Date:   Fri Feb 5 15:40:03 2021 +0200

    gr-uhd: ALL_LOS binding
    
    Signed-off-by: karel <5636152+karel@users.noreply.github.com>

diff --git a/gr-uhd/python/uhd/bindings/usrp_block_python.cc b/gr-uhd/python/uhd/bindings/usrp_block_python.cc
index 09d164195..b93d96815 100644
--- a/gr-uhd/python/uhd/bindings/usrp_block_python.cc
+++ b/gr-uhd/python/uhd/bindings/usrp_block_python.cc
@@ -23,6 +23,7 @@ void bind_usrp_block(py::module& m)
     using usrp_block = ::gr::uhd::usrp_block;
 
     m.attr("ALL_MBOARDS") = py::int_(::uhd::usrp::multi_usrp::ALL_MBOARDS);
+    m.attr("ALL_LOS") = py::str(::uhd::usrp::multi_usrp::ALL_LOS);
 
     py::class_<usrp_block,
                gr::sync_block,

```

dkozel, 2021-02-15:
Ok, it is unexpected that you're seeing this compile error then. Can you double check that you have an entirely clean build and see this error, in that case we'll have to do some digging.

ReverseControl, 2021-02-15:
I decided to use the docker container from the CI pipeline(https://github.com/gnuradio/gnuradio-docker/blob/master/ci/ci-ubuntu-20.04-3.9/Dockerfile) to install gnuradio 3.9. It has been building for some time and currently is at 41%. I will let you know if it finds that error again.

ReverseControl, 2021-02-16:
So, it built just fine with the image created from that docker file.

Question: How do I open gnuradio-companion from docker? I get this:

```
> xhost +local:root
> sudo docker run -ti --net=host --privileged  --env="DISPLAY" gnuradio3.9_installed bash
root@user:/# gnuradio-companion 
Gtk-Message: 00:43:26.570: Failed to load module "canberra-gtk-module"

(gnuradio-companion:9): Gdk-ERROR **: 00:43:26.589: The program 'gnuradio-companion' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadAccess (attempt to access private resource denied)'.
  (Details: serial 180 error_code 10 request_code 130 (MIT-SHM) minor_code 1)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)

```

ReverseControl, 2021-02-16:
I got it to run without errors on the terminal but a popup comes up:

```
docker run -ti --net=host --env="DISPLAY" --env QT_X11_NO_MITSHM=1  --privileged --ipc=host   gnuradio3.9 bash
```

Here's the Dockerfile:

```
FROM ubuntu:20.04
MAINTAINER Martin Braun <martin@gnuradio.org>

ENV security_updates_as_of 2020-12-18

# Prepare distribution
RUN apt-get update -q \
    && apt-get -y upgrade

# CPP deps
RUN DEBIAN_FRONTEND=noninteractive \
       apt-get install -qy \
         libasound2 \
         libboost-date-time1.71.0 \
         libboost-filesystem1.71.0 \
         libboost-program-options1.71.0 \
         libboost-thread1.71.0 \
         libcomedi0 \
         libfftw3-bin \
         libgmp10 \
         libgsl23 \
         libgtk-3-0 \
         libjack-jackd2-0 \
         liblog4cpp5v5 \
         libpangocairo-1.0-0 \
         libportaudio2 \
         libqwt-qt5-6 \
         libsndfile1-dev \
         libsdl-image1.2 \
         libthrift-dev \
         libuhd3.15.0 \
         libusb-1.0-0 \
         libzmq5 \
         libpango-1.0-0 \
         gir1.2-gtk-3.0 \
         gir1.2-pango-1.0 \
         pkg-config \
         thrift-compiler \
         --no-install-recommends \
    && apt-get clean

# Py3 deps
RUN DEBIAN_FRONTEND=noninteractive \
       apt-get install -qy \
         pybind11-dev \
         python3-click \
         python3-click-plugins \
         python3-mako \
         python3-dev \
         python3-gi \
         python3-gi-cairo \
         python3-lxml \
         python3-numpy \
         python3-opengl \
         python3-pyqt5 \
         python3-yaml \
         python3-zmq \
         python3-six \
         --no-install-recommends \
    && apt-get clean

# Build deps
RUN mv /sbin/sysctl /sbin/sysctl.orig \
    && ln -sf /bin/true /sbin/sysctl \
    && DEBIAN_FRONTEND=noninteractive apt-get install -y \
       --no-install-recommends \
       build-essential \
       ccache \
       cmake \
       libboost-date-time-dev \
       libboost-dev \
       libboost-filesystem-dev \
       libboost-program-options-dev \
       libboost-regex-dev \
       libboost-system-dev \
       libboost-test-dev \
       libboost-thread-dev \
       libcomedi-dev \
       libcppunit-dev \
       libfftw3-dev \
       libgmp-dev \
       libgsl0-dev \
       liblog4cpp5-dev \
       libqwt-qt5-dev \
       qtbase5-dev \
       libsdl1.2-dev \
       libuhd-dev \
       libusb-1.0-0-dev \
       libzmq3-dev \
       portaudio19-dev \
       pyqt5-dev-tools \
       doxygen \
       doxygen-latex \
       swig \
    && rm -f /sbin/sysctl \
    && ln -s /usr/bin/ccache /usr/lib/ccache/cc \
    && ln -s /usr/bin/ccache /usr/lib/ccache/c++ \
    && mv /sbin/sysctl.orig /sbin/sysctl

# Testing deps
RUN DEBIAN_FRONTEND=noninteractive \
       apt-get install -qy \
       xvfb \
       lcov \
       python3-scipy \
       --no-install-recommends \
    && apt-get clean

# Install other dependencies (e.g. VOLK)
RUN apt-get -y install -q \
        git \
        ca-certificates \
        --no-install-recommends
RUN apt-get clean
RUN apt-get autoclean

# Install VOLK
RUN mkdir -p /src/build
RUN git clone --recursive https://github.com/gnuradio/volk.git /src/volk --branch v2.4.1
RUN cd /src/build && cmake -DCMAKE_BUILD_TYPE=Release ../volk/ && make && make install && cd / && rm -rf /src/

#Install GNURADIO 3.9
WORKDIR /root/
RUN apt update && apt install -y libcanberra-gtk*
RUN git clone https://github.com/gnuradio/gnuradio.git && \
    cd gnuradio && \ 
    mkdir build && \
    cd build    && \
    cmake ..    && \
    make        && \
    make install
   
ENV PATH="/root/gnuradio/bin:$PATH"
ENV PYTHONPATH="/root/gnuradio/lib/python3/site-packages:/root/gnuradio/lib/python3/dist-packages:/root/gnuradio/lib/python3.6/site-packages:/root/gnuradio/lib/python3.6/dist-packages:/root/gnuradio/lib64/python3/site-packages:/root/gnuradio/lib64/python2/dist-packages:/root/gnuradio/lib64/python3.6/site-packages:/root/gnuradio/lib64/python3.6/dist-packages:$PYTHONPATH"
ENV LD_LIBRARY_PATH="/root/gnuradio/lib:/root/gnuradio/lib64/:$LD_LIBRARY_PATH"
ENV LIBRARY_PATH="/root/gnuradio/lib:/root/gnuradio/lib64/:$LIBRARY_PATH"
ENV PKG_CONFIG_PATH="/root/gnuradio/lib/pkgconfig:/root/gnuradio/lib64/pkgconfig:$PKG_CONFIG_PATH"
ENV PYBOMBS_PREFIX="/root/gnuradio"
```

I get the following popup:

[image]

How do I go about fixing this? Adding those ENV at the end used to solve this issue for me in earlier gnuradio versions. Help!

dkozel, 2021-02-16:
I'd have guessed that your supplied PYTHONPATH variables were correct. Check where the make install put the python files and make sure that directory is in your PYTHONPATH. I'm not a docker user so I'm afraid I can't be of much more help with any details that are specific to Docker.

I'd recommend joining the GNU Radio chatroom, someone there might have more experience.
https://chat.gnuradio.org or #gnuradio on the Freenode IRC server.

willcode, 2021-02-27:
@ReverseControl Is it ok to close this issue, since the original problem was solved?

closed 2021-02-27 by ReverseControl