gnuradio/gnuradio #4239: Signal Source block pair message handling broken

bgottula, 2021-02-15:
When sending a `pair` message to the the Signal Source block's `cmd` port, I get the following exception (on maint-3.8):

```
thread[thread-per-block[0]: <block sig_source(5)>]: pmt_car: wrong_type : offset
```

The message was constructed as follows (using a Message Strobe block in GRC), which should be allowed if I understand the source code and documentation correctly:

```
pmt.cons(pmt.intern("offset"), pmt.from_double(25.0))
```

Digging into the source, I found the following lines which seem problematic:
```
    if (pmt::is_dict(msg)) {
        list_of_items = pmt::dict_items(msg);
    } else if (pmt::is_pair(msg)) {
        list_of_items = pmt::list1(msg);
    } else {
        GR_LOG_WARN(this->d_logger, "malformed message: is not dict nor pair");
        return;
    }
```
Ref: https://github.com/gnuradio/gnuradio/blob/master/gr-analog/lib/sig_source_impl.cc#L93-L96

The docstring for `pmt::is_dict()` says "Return true if \p obj is a dictionary (warning: also returns true for a pair)", and similarly the docstring for `pmt::is_pair()` says "Return true if \p obj is a pair, else false (warning: also returns true for a dict)". This means the first condition in the `if` statement will evaluate to `true` if `msg` is a pair _or_ a dict, and I don't think the `else if` part is reachable under any circumstances, which is clearly not what the author of this if statement was expecting.

I'm new to message passing in GNU Radio so I'm not sure what the right course of action is to fix this. Is there a reliable way to distinguish between pairs and dicts? Or is that even the right thing to be doing here?

willcode, 2021-02-20:
In this case,
```
>>> x=pmt.cons(pmt.intern("offset"), pmt.from_double(25.0))
>>> x.is_pair()
True
>>> x.is_dict()
False
```
In the QA code for this block, here is the message format:
```
        src._post(
            pmt.to_pmt('freq'),
            pmt.to_pmt({
                "freq": frequency,
                "ampl": amplitude,
                "offset": offset
            }))
```
where not all of the params need to be present. See if that helps.

willcode, 2021-02-27:
@bgottula Have you had a chance to try this again using the `to_pmt` method above?

jacobagilbert, 2021-02-28:
@willcode The `pdu_dict` class that allows differentiation between pairs and dicts as you mention was not added until 3.9. 

This is a systemic issue with blocks that accept configuration messages as input are not well sanitized and unexpected but valid messages (hey...any PMT is a valid message!) can take blocks/flowgraphs down.

@bgottula You can work around this by providing the message in a format the block expects, which in this case is a dictionary as opposed to a pair, though you in general your observation is a bug in the sense that what you are doing is totally reasonable and the result is not -- so thank you.

`pmt.dict_add(pmt.make_dict(), pmt.intern("offset"), pmt.from_double(25.0))`

bgottula, 2021-03-02:
@jacobagilbert thanks, yeah the dict work around is good enough for me. I reported the problem for pair in case it was a bug worth fixing on maint-3.8. If not I don't mind if this is closed.

One implication of this problem is this example doesn't work as shown in 3.8, which is how I stumbled on it: https://wiki.gnuradio.org/index.php/Signal_Source#Example_Flowgraph

willcode, 2021-03-03:
Added a PR to backport the changes to `is_dict()` mentioned above. We need to research whether it can be applied safely to maint-3.8.

Moving discussion to the PR, closing this issue.

closed 2021-03-03 by willcode