gnuradio/gnuradio #4241: Usage of std::to_string introduces locale-dependency where it shouldn't

marcusmueller, 2021-02-15:
I just realized: while merging the commit that got rid of the last couple dozen std::cout / printf logging calls, I had to fix a lot of stuff "on the fly".

While airborne, I used std::to_string on numeric values in quite a few places.

That was a terrible idea: to_string is localized, and you don't want your comma-separated sample values logged anywhere to suddenly contain commas as decimal separators (and a variety of other ugly things that can go wrong).

This is a bug :(

mbr0wn, 2021-02-19:
If we want to make it translatable, isn't a locale'd version of to_string() the right thing?

mbr0wn, 2021-02-19:
FYI, I've used to_string() too.

marcusmueller, 2021-03-24:
> If we want to make it translatable, isn't a locale'd version of to_string() the right thing?

Uff, good question. I was assuming GRC, and our CLI tools should be localized, but logging? Would you want to have a mailserver that logs in German, so that you can't search the internet for error messages?

marcusmueller, 2024-01-28:
This is a significant problem in gr-soapy: here ranges and other floats are converted using the localied to_string, which can lead to broken address strings

But I think it's solved in other places (i.e., the rest is pretty safe integer conversions).

marcusmueller, 2024-03-11:
tagging as medium, because this also affects parameters passed to libiio which passes them to the kernel which expects specific decimal separators