gnuradio/gnuradio #4249: GUI Hint issue in GR 3.8

ChristopheSeguinot, 2021-02-17:
Hi all 
I encounter a GUI Hint issue in my flowgraph as soon as GUI Hint are not properly set. I surprisingly did not found any issue report for this annoying bug. 
The problem is affecting GR 3.8, not 3.9. It is illustrated in the attached flowgraph (activate view/show block comments)
Set two QT GUI widget overlapping (GUI Hint=0,0,1,1 for example):

- GRC don't notice the widget are at the same position or overlapping
- GR generate an error when compiling flowgraph
'
Generating: '/home/...../test_grc/GUI_issue.py'
Generate Error: (TypeError("unsupported operand type(s) for %: 'NoneType' and 'str'"), '<%\n    win = \'self._%s_win\'%id\n    range = \'self._%s_range\'%id\n%>\\\n${range} = Range(${start}, ${stop}, ${step}, ${value}, ${min_len})\n${win} = RangeWidget(${range}, self.set_${id}, ${(label if (len(label) - 2 > 0) else repr(id))}, "${widget}", ${rangeType})\n${gui_hint() % win}')
Failure
'

It seems 

- the error is due to GR not detecting widget overlap, 
- and that this error would not be encountered is the overlap had been detected

[test_GUI_issue.grc.txt](https://github.com/gnuradio/gnuradio/files/5996489/test_GUI_issue.grc.txt)

duggabe, 2021-02-17:
This is related to https://github.com/gnuradio/gnuradio/issues/3371

ChristopheSeguinot, 2021-02-17:
Yes it's related to #3371 (I mentioned it inside the flowgraph block comments)
I discovered 2 additional information: 

1- collision_detection() crash
- My GRC is repeatedly crashing, without closing the flowgraph, and I'm steel able to run my flowgraph if I close the "send report" window
- however the error indicate that the collision_detection crashed with the message in the "send report" : 
- "gnuradio-companion crashed with exception in collision_detection(): Block 'volume' is also using parent 'main', cell (0, 0)."

2- Using these 2 correct GUI Hints should work However it also generate the error reported in the first message: 

- GUI Hint1 = 0,0,**3**,1 (for freq)
- GUI Hint2 =1,0,1,1 (for volume)
- GR 3.8 generate error and don't compile flowgraph, and also crash
- with the same message with "cell (1, 0)"

testing with this works :
- GUI Hint1 = 0,0,**1**,1 (for freq)
- GUI Hint2 =1,0,1,1 (for volume)

Could someone guide me to debug this? 
Note 
I'm using Ubuntu 20.04, 
GNU Radio Companion 
Python 3.8.5

ChristopheSeguinot, 2021-02-17:
Hi 
The solution has been implemented in GR 3.9  in issue #2843
As asked by Marcus (see https://github.com/gnuradio/gnuradio/pull/2843#issuecomment-559648193), this must be backported to GR 3.8. I've just tested, it works. 

I will make a PR for this

P.S. this PR will also solve issue  #3371  (don't ask me why!)

closed 2021-02-18 by ChristopheSeguinot