gnuradio/gnuradio #4267: maint-3.8 test fail RPi4

MountainLogic, 2021-02-20:
Not sure if this matters much for 3.8:

```
99% tests passed, 3 tests failed out of 366

Total Test time (real) = 774.70 sec

The following tests FAILED:
	 71 - qa_volk_32fc_32f_multiply_32fc (Failed)
	 96 - qa_volk_32fc_x2_multiply_32fc (Failed)
	150 - qa_pmt (Failed)
Errors while running CTest
make: *** [Makefile:86: test] Error 8
```

drmpeg, 2021-02-20:
Are you running a 32-bit OS? If yes, then qa_pmt will fail. See this fix. https://github.com/gnuradio/gnuradio/pull/3355

Unfortunately, that was never backported to 3.8.

As for the VOLK failures, it's more complex. 3.8 uses a pretty old version of VOLK. If you're building from source, you can use a newer version of VOLK by just checking it out with git.
```
cd volk
git checkout v2.4.1
git submodule update --init
```
If you're on a 32-bit OS, you can get more performance by adding some flags to the cmake stage.
```
cmake -DCMAKE_CXX_FLAGS:STRING="-march=native -mfloat-abi=hard -mfpu=neon" -DCMAKE_C_FLAGS:STRING="-march=native -mfloat-abi=hard -mfpu=neon" -DCMAKE_ASM_FLAGS="-march=native -mfloat-abi=hard -mfpu=neon -g" ../
```

MountainLogic, 2021-02-20:
I found [this GnuRadio distro flavor of RPiOS](https://github.com/luigifcruz/pisdr-image).  It is 32 bit and seems to just work so I'm going to just use it.   Luigi Cruz deserves a great deal of credit and support for his efforts.

closed 2021-02-20 by MountainLogic