gnuradio/gnuradio #4273: Update documentation of Correlation Estimator "Threshold"

mormj, 2021-02-21:
https://lists.gnu.org/archive/html/discuss-gnuradio/2021-02/msg00116.html

summary:
What is shown as "pfa" in the code is actually a THRESHOLD
What is shown as "threshold" in the code is actually 1-PFA.
The user actually inputs 1-PFA (erroneously named "threshold").

Ipsit1234, 2021-02-22:
@mormj  I want to take up this issue. Can you assign this to me?

mormj, 2021-02-22:
@Ipsit1234 - thank you for volunteering to address this.  There are 2 places where this should be consistently documented:
1) Submit a pull request with changes in the `gr-digital/include/gnuradio/digital/corr_est_cc.h` - the Doxygen description of the block is what gets turned into the online API documentation as well as the Python docstrings
2) Once the text is agreed on the PR (since that has a method of review), then the wiki for the block should be updated as well:
https://wiki.gnuradio.org/index.php/Correlation_Estimator

anastas, 2021-02-22:
Changing the documentation is maybe the minimal that has to be done.
thank you mormj for taking this up!

Please keep in mind the following as you make changes:

The code essentially does:

|r[k]|^2 + |r[k-1]|^2 > 4 E thr

where r[k] is the correlator output, E is the estimated energy per sample and thr is the evaluated thr, which in the code is evaluated as thr=-log (pfe).
However, this is not to be trusted either, because the above is true ONLY when the distribution is a chi-sq with 2 degrees of freedom. This would have been the case if the code did this: |r[k]|^2 > 4 E thr.
However, because two consecutive samples are added together, we have a chi-squared with 4 degrees of freedom where 
the correct formula for the FA probability is:
Pfa = (1+ thr/E) exp(-thr/E)

I guess at the minimum, one can alert the user in the documentation that the parameter they enter (which is actually 1-pfe)
is not to be taken at face value....

A more appropriate change (that would change the API though) is that the user enters a "threshold" which is then used directly in the algorithm without translating it through a pfe...

BTW, I can also provide an updated code (without any API changes) that at least renames the variables to what they really are.
I do not have time to prepare a full PR at this point but I can attach the code and someone can incorporate it in their PR.

Please let me know if this is OK.

Achilleas

anastas, 2021-02-22:
Here are the files that I have modified:

modified:   gr-digital/grc/digital_corr_est_cc.xml
modified:   gr-digital/include/gnuradio/digital/corr_est_cc.h
modified:   gr-digital/lib/corr_est_cc_impl.cc
modified:   gr-digital/lib/corr_est_cc_impl.h

Again, please note these modifications involve only renaming 
internal variables to be roughly consistent with their true meaning.
They also modify the methods  "threshold()" and "set_threshold()" to
"oneminuspfa" and "set_oneminuspfa()" accordingly.

Apologies, but at this point i do not have the bandwidth to take care of this PR formally and completely and thanks for anyone who is willing to take it up to completion.
[ets_corr_cc.zip](https://github.com/gnuradio/gnuradio/files/6023002/ets_corr_cc.zip)

anastas, 2021-02-26:
The above comments are valid for 3.7.
I looked at 3.8 and the situation is slightly different:
there there are two types of threshold calculation (dynamic and absolute).

The above comments are valid for the case of "dynamic" threshold calculation.
This needs to be reflected in the documentation.

Having the same parameter having two different meanings depending on the dynamic/absolute way of calculating the threshold is not very good strategy!
In the "absolute" it is actually a threshold, while in the "dynamic" is some kind of 1-false alarm probability.
The default is "absolute" so this at least will not confuse many users (as it did in the 3.7 version).

duggabe, 2021-03-02:
I have added an example flowgraph and output to the wiki: https://wiki.gnuradio.org/index.php/Correlation_Estimator

mbr0wn, 2021-03-16:
Fixed: https://github.com/gnuradio/gnuradio/pull/4274

closed 2021-03-16 by mbr0wn