gnuradio/gnuradio #4278: Codec2 isn't detected properly (at least on NixOS)

doronbehar, 2021-02-23:
On NixOS (linked issue: https://github.com/NixOS/nixpkgs/issues/82263#issuecomment-779400020 ) we are having problems with how GR (3.8 and 3.9) detects Codec2.

With both GR3.8 and GR3.9, cmake seems to find codec2 by it's own:

```
Configuring gr-vocoder support...
  Dependency Boost_FOUND = TRUE
  Dependency ENABLE_GNURADIO_RUNTIME = ON
  Dependency ENABLE_GR_FFT = ON
  Dependency ENABLE_GR_BLOCKS = ON
  Dependency ENABLE_GR_FILTER = ON
  Dependency ENABLE_GR_ANALOG = ON
  Enabling gr-vocoder support.
  Override with -DENABLE_GR_VOCODER=ON/OFF
Could NOT find Codec2 (missing: LIBCODEC2_INCLUDE_DIRS)
Could NOT find GSM (missing: LIBGSM_INCLUDE_DIRS)
```

Where codec2 has the following files in it's installation directory:

```
/nix/store/...-codec2/
├── bin
│   ├── c2dec
│   ├── c2enc
│   ├── drs232
│   ├── drs232_ldpc
│   ├── fdmdv_demod
│   ├── fdmdv_get_test_bits
│   ├── fdmdv_mod
│   ├── fdmdv_put_test_bits
│   ├── fm_demod
│   ├── fsk_mod
│   └── insert_errors
├── include
│   └── codec2
│       ├── codec2_cohpsk.h
│       ├── codec2_fdmdv.h
│       ├── codec2_fifo.h
│       ├── codec2_fm.h
│       ├── codec2.h
│       ├── codec2_ofdm.h
│       ├── comp.h
│       ├── comp_prim.h
│       ├── filter.h
│       ├── freedv_api.h
│       ├── freedv_api_internal.h
│       ├── fsk.h
│       ├── golay23.h
│       ├── horus_api.h
│       ├── kiss_fft.h
│       ├── modem_stats.h
│       ├── varicode.h
│       └── version.h
└── lib
    ├── cmake
    │   └── codec2
    │       ├── codec2-config.cmake
    │       └── codec2-config-release.cmake
    ├── libcodec2.so -> libcodec2.so.0.9
    ├── libcodec2.so.0.9
    └── pkgconfig
        └── codec2.pc

7 directories, 34 files
```

I.e it has the freedv api files, and indeed with 3.8 if I specify some Cmake files:

```
      "-DLIBCODEC2_LIBRARIES=${codec2}/lib/libcodec2.so"
      "-DLIBCODEC2_INCLUDE_DIRS=${codec2}/include"
      "-DLIBCODEC2_HAS_FREEDV_API=ON"
      "-DLIBGSM_LIBRARIES=${gsm}/lib/libgsm.so"
      "-DLIBGSM_INCLUDE_DIRS=${gsm}/include/gsm"
```

I don't see the lines:

```
Could NOT find Codec2 (missing: LIBCODEC2_INCLUDE_DIRS)
Could NOT find GSM (missing: LIBGSM_INCLUDE_DIRS)
```

And GR 3.8 builds fine and reverse dependencies of GR 3.8 that need these features are compiled correctly and work fine - [qradiolink](https://github.com/qradiolink/qradiolink) for instance (see: https://github.com/qradiolink/qradiolink/issues/67 ).

But, now with GR 3.9 Specifying or not specifying the cmake flags results in the same behavior, but in either case cmake fails after configuring everything else and it's errors are:

```
gnuradio> CMake Error at /nix/store/5d7dgha0nl6hixhszb9x5pfczpblch1a-python3.8-pybind11-2.6.1/share/cmake/pybind11/pybind11Tools.cmake:151 (add_library):
gnuradio>   Target "vocoder_python" links to target "CODEC2::CODEC2" but the target was
gnuradio>   not found.  Perhaps a find_package() call is missing for an IMPORTED
gnuradio>   target, or an ALIAS target is missing?
gnuradio> Call Stack (most recent call first):
gnuradio>   cmake/Modules/GrPybind.cmake:120 (pybind11_add_module)
gnuradio>   gr-vocoder/python/vocoder/bindings/CMakeLists.txt:44 (GR_PYBIND_MAKE_CHECK_HASH)
gnuradio> CMake Error at /nix/store/5d7dgha0nl6hixhszb9x5pfczpblch1a-python3.8-pybind11-2.6.1/share/cmake/pybind11/pybind11Tools.cmake:151 (add_library):
gnuradio>   Target "vocoder_python" links to target "GSM::GSM" but the target was not
gnuradio>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
gnuradio>   an ALIAS target is missing?
gnuradio> Call Stack (most recent call first):
gnuradio>   cmake/Modules/GrPybind.cmake:120 (pybind11_add_module)
gnuradio>   gr-vocoder/python/vocoder/bindings/CMakeLists.txt:44 (GR_PYBIND_MAKE_CHECK_HASH)
gnuradio> CMake Error at gr-vocoder/lib/CMakeLists.txt:11 (add_library):
gnuradio>   Target "gnuradio-vocoder" links to target "CODEC2::CODEC2" but the target
gnuradio>   was not found.  Perhaps a find_package() call is missing for an IMPORTED
gnuradio>   target, or an ALIAS target is missing?
gnuradio> CMake Error at gr-vocoder/lib/CMakeLists.txt:11 (add_library):
gnuradio>   Target "gnuradio-vocoder" links to target "GSM::GSM" but the target was not
gnuradio>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
gnuradio>   an ALIAS target is missing?
gnuradio> -- Generating done
```

It seems like the something with the switch to pybind from swig broke something.

ryanvolz, 2021-04-24:
Does https://github.com/gnuradio/gnuradio/pull/4476/files help with this? Because I suspect it might. That has also been backported to maint-3.9, although after the 3.9.1 release.

doronbehar, 2021-04-24:
> Does [#4476 (files)](https://github.com/gnuradio/gnuradio/pull/4476/files) help with this? Because I suspect it might. That has also been backported to maint-3.9, although after the 3.9.1 release.

Yes it does! But we still need to set for GR 3.9:

```
      "-DLIBGSM_FOUND=TRUE"
      "-DLIBGSM_LIBRARIES=${gsm}/lib/libgsm.so"
      "-DLIBGSM_INCLUDE_DIRS=${gsm}/include/gsm"
```

Otherwise:

```
gnuradio> CMake Error at /nix/store/yikabla1qm5n1gmgak9hzqy7sbagl6cq-python3.8-pybind11-2.6.1/share/cma
e/pybind11/pybind11Tools.cmake:151 (add_library):
gnuradio>   Target "vocoder_python" links to target "GSM::GSM" but the target was not
gnuradio>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
gnuradio>   an ALIAS target is missing?
gnuradio> Call Stack (most recent call first):
gnuradio>   cmake/Modules/GrPybind.cmake:120 (pybind11_add_module)
gnuradio>   gr-vocoder/python/vocoder/bindings/CMakeLists.txt:44 (GR_PYBIND_MAKE_CHECK_HASH)
gnuradio> CMake Error at gr-vocoder/lib/CMakeLists.txt:11 (add_library):
gnuradio>   Target "gnuradio-vocoder" links to target "GSM::GSM" but the target was not
gnuradio>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
gnuradio>   an ALIAS target is missing?
gnuradio> -- Generating done
```

PR #4618 (2021-06-01): Fix finding Thrift w/ MSVC, add target for Python bindings, CMake find module cleanup
Default CMake build for thrift with MSVC calls the library "thriftmd" or "thriftmt" depending on if the build is done with /MD or /MT flags.

Without a target, the Python bindings are not built (i.e. the custom command is not run) until they are needed, during install. By adding a target, they will be generated during the build step. (Helps with conda packaging, which doesn't run the "install" target and instead rusn the individual CMake install scripts on their own in order to split GR into sub-packages.)

I [... cut]
closed 2021-06-01 by mormj