gnuradio/gnuradio #4281: gr_modtool rm does not remove ref in python_bindings.cc

xynium, 2021-02-24:
Hi,
It seems that gr_modtool rm does not remove ref of the deleted block in python_bindings.cc.
73 FM4PN

solomonbstoner, 2021-05-07:
Has this already been done? When I delete a block, its reference appears to be deleted. I compared the `python_bindings.cc` before and after running `gr_modtool rm block2 `.

```
diff --git a/python/bindings/python_bindings.cc b/python/bindings/python_bindings.cc
index 51beaad..1630c19 100644
--- a/python/bindings/python_bindings.cc
+++ b/python/bindings/python_bindings.cc
@@ -21,7 +21,6 @@ namespace py = pybind11;
 // Please do not delete
 /**************************************/
 // BINDING_FUNCTION_PROTOTYPES(
-    void bind_block2(py::module& m);
 // ) END BINDING_FUNCTION_PROTOTYPES
 
 
@@ -50,6 +49,5 @@ PYBIND11_MODULE(to_delete_python, m)
     // Please do not delete
     /**************************************/
     // BINDING_FUNCTION_CALLS(
-    bind_block2(m);
     // ) END BINDING_FUNCTION_CALLS
 }
\ No newline at end of file
```

solomonbstoner, 2021-05-15:
I was unable to reproduce the behavior. Neither was I able to find a change in the code between `v3.9.0.0` and the current `master` that may cause the behavior.

marcusmueller, 2021-05-15:
Worked on the livestream together with Solomon on this, please specify your version and/or make a reproducible example if this problem persists!

PR #4686 (2021-06-16): modtool: Fix `TypeError`
Fixes #4682 and fixes #4281, a bug triggered in the `rm` and `bind` commands when the
block name (or regex) is entered as stdin rather than an argument.
closed 2021-05-15 by marcusmueller