gnuradio/gnuradio #4283: Socket PDU drops data sending sending 2500+ byte PDU's

BitBangingBytes, 2021-02-26:
Using Socket PDU to send data via TCP from one computer to another. 400 byte packets seem to come across no problem, 2596 byte packets only transfer about 1400 bytes 80% of the time and the remaining bytes are all zero's.

GNURadio 3.8.2

willcode, 2021-02-26:
@BitBangingBytes Do you have a simple test case you can add? What type of network connection (localhost, direct ethernet, via router, etc.)?

BitBangingBytes, 2021-02-28:
I am sending various size messages from the PDU Utilities -> PDU Align block to the Socket PDU block. They are anywhere from 40 bytes to ~2600 bytes. The server is a wired connection and the client is wireless.

Changing the client to wired into the same network switch as the server seems to make the problem go away entirely. I tried changing the MTU to 1500 figuring it would split the data and perhaps fix the issue but it appears that truncates what is sent as opposed to splitting up larger packets.

If the intent of the block is to work only with wired connections we can close this issue. The ZMQ Message blocks seem to work flawlessly for my application so I switched to them.

willcode, 2021-02-28:
Could you also try a smaller MTU, say 1472 or even smaller? Not all links have a MTU of 1500. If it works with something smaller, then I'd say it's working as intended, even if it's a manual effort to find the MTU of your network path.

willcode, 2021-03-14:
@BitBangingBytes any updates experimenting with link MTU?

BitBangingBytes, 2021-03-16:
I tried a number of different MTU’s but nothing that corrected the issue. If there are some specific tests I could run that would be helpful I am happy to do so.

rear1019, 2021-03-17:
Note that there are multiple MTU values involved: The Socket PDU block, the network interface and the IP socket have MTU values.

> Using Socket PDU to send data via TCP from one computer to another. 400 byte packets seem to come across no problem, 2596 byte packets only transfer about 1400 bytes 80% of the time and the remaining bytes are all zero's.

> I tried changing the MTU to 1500 figuring it would split the data and perhaps fix the issue but it appears that truncates what is sent as opposed to splitting up larger packets.

The Socket PDU server splits outgoing data as per the MTU setting of the block. Additional segmentation may be performed by the TCP stack. You should receive all of the data ¹, however, the data might be split into multiple blocks. The Socket PDU client doesn’t reverse the segmentation of the server. This may be considered a bug depending on how the MTU setting of the Socket PDU is interpreted.

¹ Assuming no packets are lost if UDP is used or that TCP doesn’t give up transmission.