gnuradio/gnuradio #4317: Low pass, High pass convenience blocks not worked properly with QT GUI Range

patel999jay, 2021-03-04:
I was testing GNURadio Filter Introduction Example and found out that ` GRC convenience blocks`  for the filter tap generation don't have working callbacks as @bistromath  suggested in my post in the discussion forum. 

**Initial Testing:**

[image]

@patchvonbraun : suggested me to that it should work with my existing setup as  from the YAML it looks like they should work, but unfortunately didn't work(refer below screenshot).

[image]

But when i bypass those `convenience blocks` and directly replace `lp_taps` in the low-pass filter with `firdes.low_pass(1, samp_rate, cutoff_high, transition, firdes.WIN_HANN)`  - The actual filter generation callbacks work . I confirmed @bistromath suggestions is correct that

> for instance, changing the parameters with a QT GUI Range slider will work just fine. It's just the convenience blocks which don't work.

[image]

[filter_taps_modified.pdf](https://github.com/gnuradio/gnuradio/files/6081062/filter_taps_modified.pdf)

**Output QT GUI:**

[image]

ChristopheSeguinot, 2021-03-04:
The issue is present in GR>=3.8.2 (probably 3.8.0 too)
Looking at callbacks_all (generated inside top_block.pyfor the present flowgraph), I discovered that no callback is related to "xx filter taps blocks" (the RRC filter taps not present in the original flowgraph is also buggy)

in Yaml /gr-filters/grc/variable_low_pass_filter_taps.block.yml I don't see any callback !

I've just discovered what GNURadio callbacks are, and don't understand the difference between template and CPP_/usr/share/gnuradio/grc/blocks/variable_low_pass_filter_taps.block.yml so I Feel not able to fix this issue myself...

Regards, Christophe

willcode, 2021-06-02:
Callbacks added, so this should be resolved.

closed 2021-06-02 by willcode