gnuradio/gnuradio #4322: qa_filterbank flakiness

mbr0wn, 2021-03-04:
```
133/241 Test #133: qa_filterbank ................................***Failed    0.26 sec
F
======================================================================
FAIL: test_000 (__main__.test_filterbank_vcvcf)
Generates nfilts sets of random complex data.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/__w/gnuradio/gnuradio/gr-filter/python/filter/qa_filterbank.py", line 120, in test_000
    self.assertComplexTuplesAlmostEqual(comb_results2, outdata2, 6)
  File "/__w/gnuradio/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 70, in assertComplexTuplesAlmostEqual
    return all([
  File "/__w/gnuradio/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 71, in <listcomp>
    self.assertComplexAlmostEqual(x, y, places, msg)
  File "/__w/gnuradio/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 32, in assertComplexAlmostEqual
    raise self.failureException(
AssertionError: (-0.14535445023395024-0.020636200107483925j) != (-0.10142534971237183-0.014487081207334995j) within 6 places
```

https://github.com/gnuradio/gnuradio/pull/4298/checks?check_run_id=2013512503

ChristopheSeguinot, 2021-03-08:
This test succeed for GR 3.8.2 and 3.9 under Ubuntu 20.04
I don't have master compiled on my computer to test further.

mbr0wn, 2021-03-10:
For clarification, "flaky tests" describe tests that sometimes fail, but usually pass. This one rarely fails, but I'm trying track all failures here.

argilo, 2022-04-04:
This looks to be a duplicate of #2206.

I believe the root cause is that the Probe Signal Vector block is not synchronized, and so the work function can be writing to the `d_level` vector at the same time `level()` is reading from it:

https://github.com/gnuradio/gnuradio/blob/3349d483a1f7e5db0eea042815d237810373857b/gr-blocks/lib/probe_signal_v_impl.cc#L43-L54

https://github.com/gnuradio/gnuradio/blob/3349d483a1f7e5db0eea042815d237810373857b/gr-blocks/lib/probe_signal_v_impl.h#L31

PR #5744 (2022-04-05): blocks: Synchronize access to d_level to prevent race conditions
## Description
While looking into #4322, I found that the root cause of the issue is that the Probe Signal Vector block is not synchronized: if `level()` is called while the work function is writing to `d_level`, then invalid data will be returned. The same is true of the Probe Signal block. To fix these problems, I did the following:

1. In the Probe Signal block, change `d_level` from `T` to `std::atomic<T>`.
2. In the Probe Signal Vector block, use a mutex to synchronize access to `d_level`.

I also added regression tests whic [... cut]
closed 2022-04-05 by mormj