gnuradio/gnuradio #4323: Cannot open files

fsdonato, 2021-03-04:
I'm using gnuradio 3.7.13.5 and I also tested version 3.7.13.4
Windows Operating System
Gnuradio is unable to open saved files.
When I try to open a .grc file, the following message appears
"File C:\Users\Filipe\ Documents\ gnuradio\program1.grc Does not Exist!"
But the program does exist!
The program can only open what I am already doing.
Example: I created 5 blocks and saved. When opening the program again, the 5 blocks will appear in the program.
But if I take that saved program and change the folder and on gnuradio go to "file-open" the same error message appears. 
Any solution for this error?

ChristopheSeguinot, 2021-03-08:
I never used GNURadio under Windows but could this be due to having some blank character in your path? 
`File C:\Users\Filipe\ Documents\ gnuradio\program1.grc Does not Exist!`
has 2 spaces between Documents and between gnuradio

fsdonato, 2021-03-09:
The problem is in the Windows folders
So that gnuradio works correctly you need to avoid using spaces or accents in folder names.
Exemple: C:\Users\Filipe Donato\My Documents\Gnurádio Prógrâms\Prôgrâm1.grc
Correct is: C:\Users\Filipe-Donato\My-Documents\Gnuradio-Programs\Program1.grc

ei8fdb, 2021-03-22:
@fsdonato just so I understand - GRC doesn't currently support filepaths/filenames with spaces, or accents?

fsdonato, 2021-03-22:
> @fsdonato just so I understand - GRC doesn't currently support filepaths/filenames with spaces, or accents?

Yes
That was the way I managed to open the files 
Remove Accents

ei8fdb, 2021-03-22:
Thanks for clarifying that. 

I do wonder: how many users would expect to be able to type filenames/filepaths with spaces and/or accents?

mbr0wn, 2021-05-14:
Seems not uncommon, especially under Windows (for non-English speakers).

ei8fdb, 2021-05-15:
Is there a technical reason why GRC couldn't support .GRC filenames with special characters or spaces?

ChristopheSeguinot, 2021-05-15:
Test under Ubuntu 20.04 GRC 3.9.1: 
I'm able to open any flowgraph with accent and/or space in .grc filename as well as accent and/or space in path

dkozel, 2021-09-29:
Still a problem with 3.8 on Windows.

Also it seems that bare filenames (without the full directory path) do not work, though this could possibly be linked to the special characters depending on how GNU Radio was running.

marcusmueller, 2024-01-28:
No windows machine to test. Need help to reproduce!

gsuberland, 2024-01-29:
Tested with radioconda 2024.01.26 (64-bit) on Windows 10 Pro for Workstations x64 22H2 19045.3693 installed with the English (United Kingdom) locality and display language.

I cannot reproduce this. I have saved and re-opened the following files with success:

- `C:\Users\Graham\Documents\Test Folder\flowgraph.grc`
- `C:\Users\Graham\Documents\Test Folder\flögraph.grc` (note the ö in the filename)
- `C:\Users\Graham\Documents\Test Földer\flowgraph.grc` (note the ö in the folder name)
- `C:\Users\Graham\Documents\Test Földer\flögraph.grc` (note the ö in both the folder and file names)
- `C:\Users\Graham\Documents\Gnurádio Prógrâms\Prôgrâm1.grc` (as shown by the commenter above)

marcusmueller, 2024-01-29:
And with that, I'm happy to close this issue!
Thanks, @gsuberland :)

gsuberland, 2024-01-29:
Just for completeness, if someone runs into this in future (or anything similar) I would recommend the following steps for collecting enough information to debug the problem:

1. Open GNU Radio.
2. Make a new flow graph, close all others.
3. Go to File -> Open.
4. Browse to where the problematic file is (the one you can't open), but **do not actually open it yet**. It is important that you open the File -> Open dialog and browse to the right location _first_ because otherwise the logs you're about to generate will grow very large and contain lots of information about the files on your computer, which we don't need or want.
5. Download [Process Monitor](https://learn.microsoft.com/en-us/sysinternals/downloads/procmon) and extract it.
6. Run it as administrator. It'll pop up a window asking you to configure filters.
7. Configure the filters so they look like the image below (i.e. include process name "python.exe" and "gnuradio-companion.exe", and include event class "File System")
8. Go back to GNU Radio and open the file. Do nothing else in GNU Radio.
9. Go back to Process Monitor and click File -> Save. Choose "Events displayed using current filter", check "Also include profiling events", set the format to Native Process Monitor Format (PML), and save it.
10. Go to File -> Save again and do the same thing but in CSV format (so non-Windows folks here have some info to look at).
11. Zip the two saved log files and upload them to the GitHub issue.

**Filter config:**

[image]

(Note: don't worry about all the red ones, those are pre-filled by procmon by default - it's just the green ones you need to set)

This should provide the necessary information about what is happening under the hood.

For maintainers: you can get an overview of what operations were performed on what file paths, and what the results were, from the CSV. If you open the PML file in Process Monitor you will also be able to drill into the details of each event, including the process call stack at the time of the operation, which should help narrow things down.

marcusmueller, 2024-01-29:
@gsuberland  that's kind of the equivalent of a strace on Linux, right? Logs the system calls made?

gsuberland, 2024-01-29:
Essentially, yes, although non-invasive (works on any already-running process without any debug attach) and a fair bit more user friendly. It can record file, registry, network, and process/thread operations. I believe it's based on Event Tracing for Windows (ETW) under the hood.