gnuradio/gnuradio #4330: GRC: visually show language support for blocks

willcode, 2021-03-08:
The block yml files specify which languages can be generated (Python, C++). Make this information visible in
* block list
* documentation tab for blocks
* displayed block, in some compact way

ei8fdb, 2021-03-08:
@willcode you mean *just* the language _not_ the code?

To me this is a nice easy information architecture task to display this. Is there a canonical list of languages that can be generated? Is there a programmatic way to create this information for each block, or does it need to be added manually?

If someone can point me to a canonical list of languages, I can sketch some designs for how this would look in 1) block list (that's on the RHS of GRC right?) and 2) in each displayed block.

willcode, 2021-03-08:
Just the language. Right now, and for the foreseeable future, it's C++ and Python. Correct on the block list (RHS) and each displayed block.

In the block YAML, there is a flags field. The contents include, but are not limited to, language support. For example:
```
id: blocks_add_xx
label: Add
flags: [ python, cpp ]

parameters:
-   id: type
...
```

ei8fdb, 2021-03-09:
I thought this was going to be easy, but it's not!

@willcode a couple of more questions:

- what is the need to show this information?

**Displaying it in block list**
Would you expect to be able to search blocks by which language can be generated? E.g.:
- searching for python would return all blocks 1) with Python in the name, and 2) that can generate Python code?
- searching for cpp **OR** c++ would return all blocks 1) with cpp or c++ in the name, and 2) that can generate C++ code?

If so, I think it may be better to handle it that way as opposed to displaying Python, C++ in the block list. It looks a bit noisy. (See design link below)

The only other ways I can think of is with some visual queue - colour the block name text in the block list, not vert elegant. Or maybe a tag, again too much noise (I think).

:point_right: I'm curious if anyone has any other suggestions?

**Displaying it in documentation tab for blocks**

Where is this text generated? It does not seem to be the same text as is in the yml files. E.g. Sample and hold block documentation text is not the same as that in the sample and hold block in [gr-blocks/grc/sample](https://github.com/gnuradio/gnuradio/tree/master/gr-blocks/grc/blocks_sample_and_hold_xx.block.yml) Am I missing something?  

**Displaying it in the displayed block, in some compact way**
I've taken inspiration from the flowgraph options block which already mentions Output language. Essentially we're adding another line of text to each block. Would this information need to be seen all the time? Or only when required? If so maybe it can be "turned on" using a view menu option?

Have a look at the wireframe designs below. **They are rough designs so feedback and constructive criticism welcome.**

https://design.penpot.app/#/view/72d58650-8113-11eb-8919-57893f4da933/72d58651-8113-11eb-8919-57893f4da933?token=1aYXB3PtxwygsN6wHeJ5yw&index=0

willcode, 2021-03-09:
Short version: add a (P), (PC), or (C) to the end of each block name in the list. Keep thinking about on-canvas.

Long version:

"Need" might be putting is a little strongly, but it would be nice if a user could know whether a block was going to work for the chosen language before dragging it onto the canvas and checking for errors.

In the block list (this would be the higher priority), it could be something as simple as:
``` Add (P,C)```
unless someone wants to have fun with glyphs. Another possibility is to make the block list a grid that can (compactly) show a few attributes. Adding (P,C) seems like the easiest thing to prototype.

On-canvas (this would be a lower priority), some smaller graphical representation would be nice, along the lines of something in the corner of a box. We've already got a lot of lines of text in on-canvas blocks, so something smaller than that. We should avoid color coding where possible. If you developed the flowgraph, then you probably already know what blocks you used. Maybe someone gave you the flowgraph. Anyway, I think this is lower priority than showing something in the block list. BTW, this is still the list of supported languages, not "output language". The Options block controls the output language of all blocks at once.

One workflow I would use (if I used C++ blocks) would be to build everything with Python blocks (fast design/run/debug cycle) and then switch to C++ in the Options block to build a higher performance version.

The documentation tab for a block comes from doxygen markup in C++ headers, e.g, in `add_const_bb.h`:
```
/*!
 * \brief output = input + constant
 * \ingroup math_operators_blk
 */
```
or from docstrings in Python.