gnuradio/gnuradio #4348: qa_packet_headerparser_b: Flaky

mbr0wn, 2021-03-11:
I thought this would have been fixed by 02b909bbcfc27ffea582b06dd2f95e478e0cdee2, but that just changed the error:

```
214/244 Test #214: qa_packet_headerparser_b .....................***Failed    5.87 sec
..F.
======================================================================
FAIL: test_003_ofdm (__main__.qa_packet_headerparser_b)
Header 1: 193 bytes
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/__w/gnuradio/gnuradio/gr-digital/python/digital/qa_packet_headerparser_b.py", line 121, in test_003_ofdm
    self.waitFor(lambda: sink.num_messages() == 2)
  File "/__w/gnuradio/gnuradio/gnuradio-runtime/python/gnuradio/gr_unittest.py", line 147, in waitFor
    self.fail(fail_msg)
AssertionError: Timeout exceeded during call to waitFor()!

----------------------------------------------------------------------
Ran 4 tests in 5.621s
```

Latest failure: https://github.com/gnuradio/gnuradio/pull/4340/checks?check_run_id=2081291093

willcode, 2021-03-12:
In test 001, are there 3 headers in the data and one random tag? Then the test for num_messages might be 3 or you might not test it until it's 4. Then it would time out.

willcode, 2021-03-13:
Verified that it is too few messages coming through. In the cases seen, it was the last message that was dropped, not one in the middle. The happened randomly in different test cases. Additional logging in waitFor() shows that it is working properly.

willcode, 2021-03-14:
With additional input data, and the qa adjusted to ignore extra messages, the error does not happen. I do not think this is a problem with either the qa or the block. Best _guess_ is that this is a race between the DONE and the packet messages. If the DONE gets processed first, the block shuts down and further messages are ignored.

willcode, 2021-03-15:
Just got a qa_pdu failure on maint-3.9, one CI box, can't replicate. Same effect, I think.

willcode, 2021-03-15:
Possible that this happens more often with fewer cpus. CI uses 2?