gnuradio/gnuradio #4365: Upgrading from GRC 3.7 to 3.9: SSB receiver demo with massive noise within the stopband now

beteigeuze21, 2021-03-12:
I upgraded from GRC 3.7 to 3.9 today. With this version, an often used flowchart demonstrating SSB demodulation using the Weaver method does not run properly any longer. There is massive noise generated in the the stopband now. 

I'll attach the flowchart and the test data. Hopefully, you can fix the bug.
[tutorial4.zip](https://github.com/gnuradio/gnuradio/files/6130128/tutorial4.zip)

duggabe, 2021-03-12:
1. I built your flowgraph. It is very unlikely that your audio card can handle 96kHz. I removed the rational resampler and changed the audio sink to 32kHz. It works fine.
2. You might find the https://wiki.gnuradio.org/index.php/Simulation_example:_Single_Sideband_transceiver tutorial very instructive.
3. The github issues are for reporting software errors. For other problems (such as yours), see https://wiki.gnuradio.org/index.php/FAQ#The_Community:_Where_you_get_help.2C_advice_and_code

beteigeuze21, 2021-03-12:
Hi duggabe

thank you for your reply.
I am sorry that I did not express clearly what I experienced. The flowgraph is an older one I used without problems with GRC 3.7. After upgrade to GRC 3.9 it does not work anymore. Setting the audio sink to 32 kHz did not change the behaviour.

To demonstrate what the flowgraph produces in my environment, I'll attach some screenshots of its output. One screenshot was taken in the first seconds after start. It shows the 3 kHz passband without modulation as expected. The next two depict vast oscillations in the stopband that generate audio distortions. 

Reading your lines, I think you experience a different performance.

Cheers
beteigeuze

[image]
[image]
[image]

ChristopheSeguinot, 2021-03-18:
Whilst the 96 kHz audio_rate also seems unadapted to my point of view, I ran the original flowgraph and it works at the first step: audio received 5/5, audio spectrum is noiseless (as you first screenshot). 
(Ubuntu 20.04, GNURadio 3.8.2.0 (Python 3.8.5)
If you recently switched to GR 3.8, did you remove cache file (for Ubuntu it's here : ~.cache/grc_gnuradio/cache.json ?

beteigeuze21, 2021-03-18:
I did a full clean-up of the previous installation before installing v3.9. Removing the cache file now does not change the behaviour of the flowgraph.

Thank you very much for your report. I am now convinced that something went really wrong with my installation. :-) Maybe I have to remove all the Python stuff I installed manually for certain purposes in the past. Hopefully, I'll get GRC running again to continue my studies.

dkozel, 2021-04-06:
Hi @beteigeuze21. Were you able to get this working? Since two people have tried reproducing the issue and found that 3.9 seems to work correctly I'm going to close this issue, but *please* reopen if you are still having a problem or join the chat where folks can help in realtime. https://chat.gnuradio.org

closed 2021-04-06 by dkozel