gnuradio/gnuradio #4396: QT GUI: Time sink segfault

jacobagilbert, 2021-03-15:
The master branch QT GUI Time sink segfaults in complex PDU mode when when the number of elements in the PDU does not match the Number of Points set in the block config.

The Number of Points entry box is hidden when in either of the PDU modes are configured as the input type, and they should not be used to set behavior which should be controlled by the number of elements (and is for float type PDUs it appears).

Flowgraph: 

[image]

[pdu_test.grc.txt](https://github.com/gnuradio/gnuradio/files/6144952/pdu_test.grc.txt)

jacobagilbert, 2021-03-16:
I can confirm that after reverting 4b7006db (and 371ea713) the issue is gone, so its somewhere in there. 

I don't think this is severe enough to warrant rolling these back, @ThomasHabets if you have a few minutes to take a look at this I'd appreciate it. I'll post here if I make any progress.

ThomasHabets, 2021-03-16:
Yeah, my bad. PR coming in a few minutes.

ThomasHabets, 2021-03-16:
PR #4399 in flight.

Thanks for the minimal test case to reproduce!

PR #4399 (2021-03-16): qtgui: Fix segfaulting overflow
Problem introduced in 4b7006db76b570e4d916e263301333d2f4d2a2df.

It appears to be the only instance of this mistake in that commit.

Fixes: #4396
commit f4fc23b3 (2021-03-16): qtgui: Fix segfaulting overflow

Problem introduced in
4b7006db76b570e4d916e263301333d2f4d2a2df. `d_cbuffers` and `d_buffers`
are not the same size.

It appears to be the only instance of this mistake in that commit.

Fixes: #4396
closed 2021-03-16 by mormj