gnuradio/gnuradio #4403: primitive_connect(): incompatible function arguments

hosam37r, 2021-03-16:
I'm new to GNU radio. 
I'm trying to run the circuit attached. with gnuradio latest version and hackrf with latest firmware version on ubuntu  20
[image]
[image]

. 
I'm getting the following error 
`File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7f68a47c41f0>, <gnuradio.gr.gr_python.basic_block object at 0x7f68adce4ab0>, 0, <Swig Object of type 'gr::basic_block_sptr *' at 0x7f68a47ce210>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.
`

I tried uninstalling and installing gnuradio multiple times but no luck. Please help

ChristopheSeguinot, 2021-03-18:
Could you please provide: 

- your fm.grc file 
- the generated .py python file (if it is generated)

hosam37r, 2021-03-19:
> Could you please provide:
> 
>     * your fm.grc file
> 
>     * the generated .py python file (if it is generated)

Attached.
No python though 

Thank] you so much

`options:
  parameters:
    author: ''
    catch_exceptions: 'True'
    category: '[GRC Hier Blocks]'
    cmake_opt: ''
    comment: ''
    copyright: ''
    description: ''
    gen_cmake: 'On'
    gen_linking: dynamic
    generate_options: qt_gui
    hier_block_src_path: '.:'
    id: top_block
    max_nouts: '0'
    output_language: python
    placement: (0,0)
    qt_qss_theme: ''
    realtime_scheduling: ''
    run: 'True'
    run_command: '{python} -u {filename}'
    run_options: prompt
    sizing_mode: fixed
    thread_safe_setters: ''
    title: ''
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [8, 8]
    rotation: 0
    state: enabled

blocks:
- name: samp_rate
  id: variable
  parameters:
    comment: ''
    value: int(44100)
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [42, 169]
    rotation: 0
    state: enabled
- name: analog_wfm_tx_0
  id: analog_wfm_tx
  parameters:
    affinity: ''
    alias: ''
    audio_rate: samp_rate
    comment: ''
    fh: '-1.0'
    max_dev: 75e3
    maxoutbuf: '0'
    minoutbuf: '0'
    quad_rate: samp_rate
    tau: 75e-6
[... 347 lines cut ...]
    decim: '1'
    fbw: '0'
    interp: '100'
    maxoutbuf: '0'
    minoutbuf: '0'
    taps: '[]'
    type: ccc
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [750, 214]
    rotation: 0
    state: true

connections:
- [analog_wfm_tx_0, '0', rational_resampler_xxx_0, '0']
- [audio_source_0, '0', blocks_multiply_const_vxx_0, '0']
- [blocks_add_xx_0, '0', analog_wfm_tx_0, '0']
- [blocks_multiply_const_vxx_0, '0', blocks_add_xx_0, '1']
- [blocks_wavfile_source_0, '0', blocks_add_xx_0, '0']
- [rational_resampler_xxx_0, '0', osmosdr_sink_0, '0']

metadata:
  file_format: 1`

marcusmueller, 2021-03-19:
How did you install GNU Radio? Is there a chance you've got two conflicting GNU Radio installations?

ChristopheSeguinot, 2021-03-19:
- This fm.grc file is not correctly indented, it is not possible to open it with GNURadio! 
-  are you using GR 3.8 or 3.9

adigeak, 2021-03-28:
I'm facing same problem 
python file : https://paste.debian.net/1191307
grc file: https://paste.debian.net/1191308/

mbr0wn, 2021-05-14:
Closing due to lack of action. You can reopen, but please provide the requested info. Thanks!

closed 2021-05-14 by mbr0wn

ocean1124, 2022-06-22:
Please, has this problem been solved, can you tell me?