gnuradio/gnuradio #4420: Install on Macbook M1 Apple Silicon

Vinz87, 2021-03-20:
I tried `sudo port install gnuradio`, but I get the following error:
```
--->  Fetching archive for python37
--->  Attempting to fetch python37-3.7.10_1.darwin_20.arm64.tbz2 from http://fco.it.packages.macports.org/python37
--->  Attempting to fetch python37-3.7.10_1.darwin_20.arm64.tbz2 from https://lil.fr.packages.macports.org/python37
--->  Attempting to fetch python37-3.7.10_1.darwin_20.arm64.tbz2 from https://mse.uk.packages.macports.org/python37
--->  Fetching distfiles for python37
Error: Failed to fetch python37: python37 cannot yet be built for Apple Silicon
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_lang_python37/python37/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets to report a bug.
Error: Processing of port gnuradio failed
```

Has anyone had success in installing gnuradio on a Macbook M1? Is it a problem of Macports or is it known that gnuradio cannot be compiled yet on M1?
Thanks

Vinz87, 2021-03-20:
my understanding is also that Python 3.7 will never get compatibility for M1: [](https://bugs.python.org/issue41100#msg382939)

so, is Python 3.7 strictly required to install gnuradio? can I tell Macports to use Python 3.9 instead?

ryanvolz, 2021-03-22:
Python 3.7 is not strictly required for GNU Radio, so Python 3.9 could work. This is more of a packaging problem and getting GNU Radio and its dependencies compiled for the arm architecture as opposed to a blocker with GNU Radio itself. @michaelld is the best person to ask about Macports-specific questions.

Maybe the Homebrew package would work for you? Beyond that, you might be stuck compiling from source.

I know that Apple Silicon support is something that I'm looking at down the road for the conda package of GNU Radio, but the current sticking point there is getting Qt built and that's not something I have the experience to help push forward.

michaelld, 2021-03-22:
I think Py38 and Py39 are supported on macOS on ARM64 / M1. I believe you can install GNU Radio is you disable GRC; maybe documentation too. I think I got most of the dependencies working ... but not every single one ... it's been a bit since I had time to try to get further. By default GR uses Py38 ... so, maybe try `sudo port clean python37 gnuradio` and then try `sudo port install gnuradio` using the default and see how far it gets.

Vinz87, 2021-04-14:
> I think Py38 and Py39 are supported on macOS on ARM64 / M1. I believe you can install GNU Radio is you disable GRC; maybe documentation too. I think I got most of the dependencies working ... but [...]

Thanks for your answer, after a `sudo port selfupdate` and a `sudo port upgrade outdated` I tried what you suggested, but I get the following output:

```
Vincenzo@Vins-Macbook-Pro ~ % sudo port clean python37 gnuradio
--->  Cleaning python37
--->  Cleaning gnuradio
Vincenzo@Vins-Macbook-Pro ~ % sudo port install gnuradio       
--->  Computing dependencies for gnuradio
The following dependencies will be installed: 
 OpenBLAS
 assimp
 clang-11
 clang_select
 coreutils
 cython_select
 dbus-glib
 dbus-python37
 docbook-xml
 docbook-xml-4.1.2
 docbook-xml-4.2
 docbook-xml-4.3
 docbook-xml-4.4
 docbook-xml-4.5
 docbook-xml-5.0
 docbook-xsl-nons
 docutils_select
 double-conversion
 findutils
 fop
 getopt
 grep
 gsed
 gzip
 harfbuzz-icu
 libevent
 libmng
 libomp
 libsdl2
 libusb
 libxslt
 libyaml
 libzzip
 links
 llvm-11
 llvm_select
 meson
 minizip
 mypy_select
 nosetests_select
 orc
 pip_select
 potrace
 py37-Pillow
 py37-alabaster
 py37-babel
 py37-beaker
 py37-cairo
 py37-certifi
 py37-chardet
 py37-cheetah
 py37-click
 py37-click-plugins
 py37-cppy
 py37-cycler
 py37-cython
 py37-dateutil
 py37-docutils
 py37-gobject3
 py37-idna
 py37-imagesize
 py37-ipaddress
 py37-jinja2
 py37-kiwisolver
 py37-lxml
 py37-mako
 py37-markupsafe
 py37-matplotlib
 py37-mypy
 py37-mypy_extensions
 py37-nose
 py37-numpy
 py37-olefile
 py37-opengl
 py37-opengl-accelerate
 py37-packaging
 py37-parsing
 py37-pip
 py37-pybind11
 py37-pygments
 py37-pyqt5
 py37-pyqtgraph
 py37-requests
 py37-roman
 py37-scipy
 py37-setuptools
 py37-sip
 py37-six
 py37-snowballstemmer
 py37-sphinx
 py37-sphinxcontrib-applehelp
 py37-sphinxcontrib-devhelp
 py37-sphinxcontrib-htmlhelp
 py37-sphinxcontrib-jsmath
 py37-sphinxcontrib-qthelp
 py37-sphinxcontrib-serializinghtml
 py37-tkinter
 py37-tornado
 py37-typed-ast
 py37-typing_extensions
 py37-tz
 py37-urllib3
 py37-yaml
 py37-zmq
 py38-beaker
 py38-certifi
 py38-chardet
 py38-cython
 py38-docutils
 py38-idna
 py38-ipaddress
 py38-mako
 py38-markupsafe
 py38-nose
 py38-numpy
 py38-pybind11
 py38-requests
 py38-roman
 py38-setuptools
 py38-six
 py38-urllib3
 pygments_select
 python37
 qhull
 qt5
 qt5-qt3d
 qt5-qtbase
 qt5-qtconnectivity
 qt5-qtdeclarative
 qt5-qtgamepad
 qt5-qtgraphicaleffects
 qt5-qtimageformats
 qt5-qtlocation
 qt5-qtmacextras
 qt5-qtmultimedia
 qt5-qtnetworkauth
 qt5-qtquickcontrols
 qt5-qtquickcontrols2
 qt5-qtremoteobjects
 qt5-qtscript
 qt5-qtscxml
 qt5-qtsensors
 qt5-qtserialbus
 qt5-qtserialport
 qt5-qtspeech
 qt5-qtsvg
 qt5-qttools
 qt5-qttranslations
 qt5-qtwebchannel
 qt5-qtwebsockets
 qt5-qtxmlpatterns
 qt5-sqlite-plugin
 qwt-qt5
 sphinx_select
 swig
 swig-python
 swig3
 swig3-python
 tcl
 texlive-basic
 texlive-bin
 texlive-common
 texlive-latex
 tk
 uhd
 volk
 xar
 xmlcatmgr
 xmlto
 xorg-libXScrnSaver
 xorg-libXp
Continue? [Y/n]: Y
--->  Fetching archive for python37
--->  Attempting to fetch python37-3.7.10_1.darwin_20.arm64.tbz2 from http://fco.it.packages.macports.org/python37
--->  Attempting to fetch python37-3.7.10_1.darwin_20.arm64.tbz2 from https://lil.fr.packages.macports.org/python37
--->  Attempting to fetch python37-3.7.10_1.darwin_20.arm64.tbz2 from https://mse.uk.packages.macports.org/python37
--->  Fetching distfiles for python37
Error: Failed to fetch python37: python37 cannot yet be built for Apple Silicon
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_lang_python37/python37/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets to report a bug.
Error: Processing of port gnuradio failed
```

Let me know if there's something else I can try, I'm willing to help

J-Iszatt, 2021-04-26:
Hello guys,

As the OG, I am also struggling with a macport GRC install on Apple Silicon.
Following  the `sudo port clean python37` and `sudo port install gnuradio` by  _michaelld_, I am getting the exact same errors related to _Python37_...
Is there any way we can force the build to use _Python38 /39_ or something else to try?

For now, I am running my Rosetta2-based _Homebrew_ build ([follow](https://diewland.medium.com/how-to-install-python-3-7-on-macbook-m1-87c5b0fcb3b5) this guide, to setup "ibrew" and run `ibrew install gnuradio` afterwards).
As this is not very seamless (gr_modtool is not working at all), I would be happy to go the macports route!

Looking forward for a reply! Best regards,

JI

michaelld, 2021-04-29:
Here is the port info on "gnuradio"
```
% port info gnuradio
gnuradio @3.8.3.0 (science, comms)
Sub-ports:            gnuradio37, gnuradio-next, gnuradio-devel
Variants:             ctrlport, ctrlport_thrift, debug, [+]docs, [+]grc, jack, performance_counters,
                      portaudio, python36, python37, (+)python38, python39, [+]qtgui, (+)quartz, [+]sdl,
                      [+]swig, [+]uhd, universal, [+]wavelet, (-)x11, [+]zeromq

Description:          GNU Radio is Software Defined Radio (SDR): This port is kept up with the GNU Radio
                      release, currently 3.8.3.0, which is typically updated twice per year.
Homepage:             https://www.gnuradio.org/

Build Dependencies:   cmake, doxygen, graphviz, py38-sphinx, texlive-latex, swig-python, makeicns,
                      pkgconfig
Library Dependencies: boost171, fftw-3-single, mpir, volk, gmp, python38, py38-numpy, py38-cheetah,
                      codec2, libgsm, py38-click, py38-click-plugins, log4cpp, xmlto, py38-gobject3,
                      py38-mako, py38-lxml, py38-yaml, gtk3, cairo, pango, py38-pyqt5, qwt-qt5, uhd, gsl,
                      libsdl, cppzmq, qt5-qtbase
Runtime Dependencies: py38-matplotlib, py38-opengl, py38-scipy, adwaita-icon-theme, py38-pyqtgraph,
                      py38-zmq
Conflicts with:       gnuradio37, gnuradio-next
Platforms:            darwin, macosx
License:              GPL-3
Maintainers:          Email: michaelld@macports.org, GitHub: michaelld
                      Email: ra1nb0w@macports.org, GitHub: ra1nb0w
```

michaelld, 2021-04-29:
Thus, if you want to use something other than the default, just select the variant(s) you want ... for example:
```
sudo port install gnuradio +python39
```
will try to install the "gnuradio" port with use of Python 3.9 instead of the default of Python 3.8. It's that simple.

I had to return my DTK before I managed to get GNU Radio fully working on it. I have yet to obtain an M1 Mac to continue this work. It's on my "to do" list to do this, but to be honest here I have so little time to do this work at the moment that I don't know when I'd get to it. If others get to this, I welcome patches to get the GR M1 support in place!

robotastic, 2021-05-30:
Thanks @michaelld - selecting the python version seems to get the build farther along. Unfortunately it runs into trouble with the Volk install. I tried to remove it from the install with the following command, but it didn't seem to work:

````
sudo port install gnuradio +python39
````

Here is the snippet from the log where it crashes:
````
:info:build [ 84%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o
:info:build cd /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/build/lib && /usr/bi
n/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -I/opt/local/include/orc-0.4 -I/opt/local/var/macports/build/_opt_local_var_macports_
sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/build/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macpo
rts.org_macports_release_tarballs_ports_science_volk/volk/work/volk-2.4.1/include -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macpor
ts_release_tarballs_ports_science_volk/volk/work/volk-2.4.1/kernels -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tar
balls_ports_science_volk/volk/work/build/lib -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_vol
k/volk/work/volk-2.4.1/lib -I/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/volk-2
.4.1/cpu_features/include -pipe -Os -DNDEBUG -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk -Wall -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot 
/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk -mmacosx-version-min=11.0 -fPIC -std=gnu11 -funsafe-math-optimizations -funsafe-math-optimizations -o CMakeFiles/vo
lk_obj.dir/volk_machine_neonv8.c.o -c /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/wo
rk/build/lib/volk_machine_neonv8.c
:info:build In file included from /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/build/lib/volk_machine_neon_orc.c:151:
:info:build /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/volk-2.4.1/kernels/volk/volk_32u_reverse_32u.h:399:9: warning: value size does not match register size specified by the constraint and modifier [-Wasm-operand-widths]
:info:build         DO_RBIT;
:info:build         ^
:info:build /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/volk-2.4.1/kernels/volk/volk_32u_reverse_32u.h:383:32: note: expanded from macro 'DO_RBIT'
:info:build                : [result] "=r"(*out_ptr)  \
:info:build                                ^
:info:build /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/volk-2.4.1/kernels/volk/volk_32u_reverse_32u.h:399:9: note: use constraint modifier "w"
:info:build /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/volk-2.4.1/kernels/volk/volk_32u_reverse_32u.h:382:22: note: expanded from macro 'DO_RBIT'
:info:build     __VOLK_ASM("rbit %[result], %[value]" \
:info:build                      ^
:info:build /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_volk/volk/work/volk-2.4.1/kernels/volk/volk_32u_reverse_32u.h:399:9: warning: value size does not match register size specified by the constraint and modifier [-Wasm-operand-widths]
:info:build         DO_RBIT;
:info:build         ^
````

hexagonal-sun, 2021-06-19:
FYI I did manage to get gnuradio built on the M1 a while back but I [ran into issues](https://trac.macports.org/ticket/62180).

michaelld, 2021-06-23:
@hexagonal-sun I'm ordering an M1 Mac soon ... had to return my DTK so I'm without ARM64 at the moment. I hope to be getting things going again mid-July on this issue ... assuming it's still an issue!

NeuroForLunch, 2021-06-24:
Just got the fix in recently: GNURadio 3.9.2 installs on the M1 using brew install gnuradio.

ryanvolz, 2021-06-24:
> Just got the fix in recently: GNURadio 3.9.2 installs on the M1 using brew install gnuradio.

Nice! Are there any patches (anywhere in the software stack) that I would need to pull for the conda packages? I've had it [building for a little while](https://anaconda.org/search?q=platform%3Aosx-arm64+gnuradio) (minus Qt :frowning_face:) but have heard reports of segfaults.

NeuroForLunch, 2021-06-24:
Take a look at the install script here: https://github.com/Homebrew/homebrew-core/blob/be1cca27d39695f82a9b15a23b0bdba27fecf923/Formula/gnuradio.rb

michaelld, 2021-06-24:
Excellent! There's nothing special that I see in the Brew script in terms of patching GR. I will strive to get GR ports updated in MacPorts, for folks who use that method ... generic but only tested on Intel until I get the new M1.

michaelld, 2021-06-24:
Anybody know if GR actually works via these installs (Intel or M1)? Looks like Brew does some basic testing (C++ and Python), so I'd guess the basic data flow does work. No testing on the GUI / GRC that I see, so that might be an issue still.

hexagonal-sun, 2021-06-24:
I could not get gnuradio to install my M1.  There is no bottle of homebrew avilable.  It appears as though the issue is that `volk` won't build:

```
cd /tmp/volk-20210624-14654-1fbv8u3/lib && /opt/homebrew/bin/orcc --include math.h --implementation -o /tmp/volk-20210624-14654-1fbv8u3/lib/volk_32fc_x2_multiply_32fc_a_orc_impl.c /tmp/volk-20210624-14654-1fbv8u3/kernels/volk/asm/orc/volk_32fc_x2_multiply_32fc_a_orc_impl.orc
In file included from /tmp/volk-20210624-14654-1fbv8u3/cpu_features/src/cpuinfo_arm.c:15:
/tmp/volk-20210624-14654-1fbv8u3/cpu_features/include/cpuinfo_arm.h:118:2: error: "Including cpuinfo_arm.h from a non-arm target."
#error "Including cpuinfo_arm.h from a non-arm target."
```

NeuroForLunch, 2021-06-24:
It appears it didn't get bottled for ARM. I'll see if I can get it bottled.

wonderingStars, 2021-07-20:
has anyone solved this issue I'm stuck too I'm new to Mac and thought it must be something that I'm doing I reset the Mac and still the same issue  . can't get universal radio hacker working either on m1 .

NeuroForLunch, 2021-07-21:
This is not yet updated to work with arm: https://github.com/gnuradio/volk

michaelld, 2021-07-21:
Thinking the Volk issue is that `cpu_features` doesn't correctly support ARM64 / M1 yet ... yes? I can fix that in MacPorts easily by patching `cpu_features` with the PR that adds that support, after it is downloaded. Let me look into doing that.

michaelld, 2021-07-21:
I finally have an M1 Mac to do testing on. I had to return my DTK & then my W2 work got busy ...

wonderingStars, 2021-07-21:
hi michaelld thank you for the fast reply . also I would like to thank you for working on this problem . that would be awesome

michaelld, 2021-07-22:
For MacPorts folks: I just pushed a patch that works for me on my Mac Mini M1: https://github.com/macports/macports-ports/commit/e83a55ef196d4283be438c052295b2fc44f3df5b

michaelld, 2021-07-22:
Not all QA passes, but at least Volk builds and seems to mostly work

NeuroForLunch, 2021-07-22:
Once the pull request gets accepted on cpu_features and is available on gnuradio/volk I can add it to homebrew

hexagonal-sun, 2021-07-22:
@michaelld Out of interest, did you have issues with `gnuradio-companion` crashing?  I had the issue and [reported it](https://trac.macports.org/ticket/62180).

michaelld, 2021-07-22:
@hexagonal-sun I'll try the "gnuradio" port tomorrow to see if it installs, and if so does GR / GRC execute

Roarrk, 2021-07-23:
@hexagonal-sun I had the same issue with gui crashing, incl. some odd defects like right-side menus not updating when expanding groups. 

I gave up digging when I found out the brew installed gnuradio did not have the same issue after they updated to 3.9.2.0. Maybe some dependancy update?..in any case it worked until I got an M1 :)

vjarao, 2021-07-24:
Dear all. I just (7/24/2021) did a fresh install on MacBookPro M1. I had no errors in the installation process. However, I am also experiencing the same GUI crash issue, either from command line or directly via icon launcher. 
I then check my active python and there is where I think I might have a mix ....

vjarao@Victors-MBP ~ % port select --list python                  
Available versions for python:
	none
	python27-apple
	python37
	python38
	python38-apple
	python39 (active)
vjarao@Victors-MBP ~ % gnuradio-companion
dbus[22641]: Dynamic session lookup supported but failed: launchd did not provide a socket path, verify that org.freedesktop.dbus-session.plist is loaded!
No such schema “org.gnome.desktop.interface”
<<< Welcome to GNU Radio Companion 3.8.3.1 >>>

Block paths:
	/opt/local/share/gnuradio/grc/blocks
zsh: segmentation fault  gnuradio-companion

[image]

michaelld, 2021-07-24:
Well ... having no errors in the installation process is progress! I started installing the background dependencies for the `gnuradio` port Friday but that errored out somewhere & I wasn't able to get back to it. Will try more Monday.

vjarao, 2021-07-24:
Thank you very much Michael for helping this efforts.

michaelld, 2021-07-26:
I finally got the `gnuradio` port installed. I'm using `+python38 -x11 +no_x11 +quartz`. I get a segfault at GRC startup. GNU Radio itself runs just fine; it's something to do with the GRC interface. I believe there's a fix in GR already as part of 3.9 ... if so then hopefully it can be backported to 3.8.

vjarao, 2021-07-27:
Michael, mine keeps failing to install now. I am going to uninstall everything and perhaps even uninstall MacPorts and re-start from fresh. I will share my findings ..

--->  Fetching archive for cmake
Warning: Your DNS servers incorrectly claim to know the address of nonexistent hosts. This may cause checksum mismatches for some ports. See this page for more information: <https://trac.macports.org/wiki/MisbehavingServers>
--->  Attempting to fetch cmake-3.20.5_1+docs+python38.darwin_20.arm64.tbz2 from https://packages.macports.org/cmake
--->  Attempting to fetch cmake-3.20.5_1+docs+python38.darwin_20.arm64.tbz2 from https://ywg.ca.packages.macports.org/mirror/macports/packages/cmake
--->  Attempting to fetch cmake-3.20.5_1+docs+python38.darwin_20.arm64.tbz2 from https://mse.uk.packages.macports.org/cmake
--->  Building cmake
Error: Failed to build cmake: command execution failed
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_cmake/cmake/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there is a bug.
Error: Processing of port gnuradio failed

vjarao, 2021-07-27:
No joy.  Removed everything and re-installed Macports and build from there ....  sudo port install gnuradio +python38 -x11 +no_x11 +quartz

--->  Attempting to fetch cmake-3.20.5_1+docs+python38.darwin_20.arm64.tbz2 from https://mse.uk.packages.macports.org/cmake
--->  Fetching distfiles for cmake
--->  Attempting to fetch cmake-3.20.5.tar.bz2 from https://distfiles.macports.org/cmake
--->  Verifying checksums for cmake
--->  Extracting cmake
--->  Applying patches to cmake
--->  Configuring cmake
--->  Building cmake                                     
Error: Failed to build cmake: command execution failed   
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_cmake/cmake/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there is a bug.
Error: Processing of port gnuradio failed

michaelld, 2021-07-27:
@vjarao can you [open a MacPorts ticket](https://trac.macports.org/newticket) & tag me on it please? That's the best way to get stuff like this "seen". Make sure to attach the build log; I can't help you without that! I just did a [quick search](https://trac.macports.org/search?portsummarysearch=on&ticket=on&q=cmake) & do not find that issue noted within the last month or so. Also: I just tried `sudo port build cmake -x11 +no_x11 +quartz +python38` and it works fine for me. There are some strange Py39 dependencies, but the actual build uses just Sphinx from Py38 so that's as desired.

vjarao, 2021-07-28:
Thanks Michael. Well I re-install everything following your method and no errors on the installation process. However the GUI crashes.

I am going to open a ticket and tag you there. Thanks Michael.

vjarao@Victors-MacBook-Pro ~ % gnuradio-companion
dbus[1131]: Dynamic session lookup supported but failed: launchd did not provide a socket path, verify that org.freedesktop.dbus-session.plist is loaded!
No such schema “org.gnome.desktop.interface”
<<< Welcome to GNU Radio Companion 3.8.3.1 >>>

Block paths:
	/opt/local/share/gnuradio/grc/blocks
zsh: segmentation fault  gnuradio-companion
[gnru_crash_report.pdf](https://github.com/gnuradio/gnuradio/files/6889828/gnru_crash_report.pdf)

michaelld, 2021-07-28:
@vjarao OK; thanks for the updates. I see the exact same crash as you do. I'm going to try `+python39` as well as look at updating the `gnuradio-devel` port to the latest 3.9 commit & see if either of those work ... and, if so, then try to engineer a fix.

michaelld, 2021-07-28:
`gnuradio +python39` still fails in the same way ... on to `gnuradio-next` updates & then testing that ...

wonderingStars, 2021-08-10:
any luck fixing the problem

michaelld, 2021-08-10:
Between this and that I haven't had time. I hope to have time in the coming weeks

Roarrk, 2021-08-13:
@michaelld Has the aarch64/arm changes to Volk been implemented, or are you using a local repo with your own changes? 

Having some serious gnuradio withdrawals.

NeuroForLunch, 2021-08-13:
If it's possible to disable the volk dependency, I'm sure people would enjoy using that during this transition period.

yuyiwx, 2021-08-23:
I have the same issue, can't install on my m1 mac. Tried multiple methods, no success. Any updates?

fllay, 2021-08-24:
it seems that Conda install is ok but I cannot start _gnuradio-companion_.  it said `(gnuradio) cake@cakeM1 ~ % gnuradio-companion       
ImportError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(Typelib file for namespace 'HarfBuzz', version '0.0' not found)`

According to anaconda https://anaconda.org/conda-forge/gnuradio, it should officially support m1. For plain python, I can import gnuradio without any trouble and moreover , I can install package as well e.g. gr_osmocom. But no luck on gnuradio-companion. I have tested gtk by using gtk-demo and It seem to be ok.

fllay, 2021-08-24:
Can I  rebuilt   gobject-introspection by myself so that gnuradio-companion can be opened? If I understand it correctly, I can just rebuild  gobject-introspection into multiple output (whatever It means but it seems that this option is turn off when gobject-introspection  is built for Anaconda).

ryanvolz, 2021-08-24:
@fllay If you've never built a conda package before then it's probably going to be tricky, but what you'd need to do is rebuild the `harfbuzz` package. You can find the recipe [here](https://github.com/conda-forge/harfbuzz-feedstock/) and you should be able to use the `build-locally.sh` script to rebuild the package. I think that should do a native build which would have introspection enabled, but if it doesn't then you'd need to modify `recipe/build.sh` so that `-Dintrospection=enabled` is passed to meson. Once built, install the package into your environment.

The reason the solution is not that simple for the package that conda-forge provides is that the community relies on CI infrastructure to do the builds, and currently that requires cross-compilation for osx-arm64. We need to also modify the `gobject-introspection` package in some way (details unknown) to make enabling the typelib build for packages like `harfbuzz` not fail when cross-compiling, which is a bigger can of worms.

Anyway, thanks for the report! Again because of the cross-compilation, we can't really do much testing of the osx-arm64 conda packages, so we rely on users even more heavily to report issues. If you get past this one and there are more issues that seem related to the conda packaging in particular, you can report them over at https://github.com/conda-forge/gnuradio-feedstock/.

fllay, 2021-08-25:
I manage to build `harfbuzz` and now I can run `gnuradio-companion`. The UI open for a while and I get segmentation fault.

```
(grc388) cake@cakeM1 harfbuzz-feedstock % gnuradio-companion                   

(gnuradio-companion:16844): Gtk-WARNING **: 11:48:15.690: Locale not supported by C library.
	Using the fallback 'C' locale.
INFO:gnuradio.grc.core.cache:Outdated cache found, will be overwritten.
<<< Welcome to GNU Radio Companion 3.8.3.1 >>>

Block paths:
	/Users/cake/miniforge3/envs/grc388/share/gnuradio/grc/blocks
zsh: segmentation fault  gnuradio-companion
``` 

This is same error as 

> Thanks Michael. Well I re-install everything following your method and no errors on the installation process. However the GUI crashes.
>
> I am going to open a ticket and tag you there. Thanks [...]

I tried every combination e.g. gnuradio 3.8, 3.9 and python 3.8 and 3.9. All will end end the same.

vjarao, 2021-09-02:
Good evening ...@micahelld any luck?

Roarrk, 2021-09-09:
@michaelld Would you mind setting up a repo with the version you are working on (that has Volk running)? 

I would be happy to help with further testing and making pull requests, but doing duplicate work with what you got working so far seems silly.

michaelld, 2021-09-09:
@Roarrk I'm using Volk as provided by MacPorts, which uses a patch that tweaks `cpu_features` ... based off of the `cpu_features` PR of note herein. You can use that Volk to get started.

@vjarao I have been inundated with work that does not involve this. Hopefully others can make some forward progress while I'm occupied with my other work.

michaelld, 2021-09-23:
Success! Now to get the fixes into MacPorts. This is quite the rabbit hole to dive into! The hack/fix is to apply this patch to PyGObject3 ( https://github.com/Damenly/pygobject/commit/01013b1003bac67061e595cd857be9cf7b69d7fd ).

[image]

michaelld, 2021-09-23:
At least for MacPorts, the fixes proposed for GOI itself will be a challenge, since MP uses an older (pre-meson IIRC) version. At least we can do this hack/fix as a temporary measure to address the issue.

michaelld, 2021-09-25:
MacPorts PR is here: https://github.com/macports/macports-ports/pull/12377

michaelld, 2021-09-25:
The "meta" here is that the issue is not with GR, but rather with dependencies (LIBFFI, GObjectIntrospection, PyGObject3). It is very likely that the issue will not be fixed any time soon, given how complicated it is. Various package managers can choose to patch around the issue, which is what I'm trying to do in MacPorts, for 1 specific use-case (and I'll add the one that impacts GNU Radio). There is plenty of info here for anyone trying to get GR going on M1 / ARM64 to be able to hack/fix the dependencies. Given all of the above I'm going to go ahead and close this issue.

closed 2021-09-25 by michaelld

michaelld, 2021-09-29:
FYI I just merged in to MacPorts the fix/hack to PyGObject3. GNU Radio ports should not build and execute "out of the box" when using MacPorts on any M1 Mac.

fllay, 2021-09-29:
> /

Should or should not?

michaelld, 2021-09-29:
@fllay oops ... just "should" ... ignore the "not" ... my bad! Wishing the interface allowed for strikeout ...

vjarao, 2021-09-30:
Thank you so much @michaelld . I did sudo port -v update and then upgrade outdated. Since I already had a previous gnu-radio installation it was failing when updating py37-pyqt5 . The solution was to uninstall py37-pyqt5  and then run again update and upgrade. After that procedure gnu radio-companion started working !!!!

vjarao, 2021-11-05:
Good morning ... so I did upgrade my M1 Mac to the new OS (Monterrey). I successfully installed MacPorts. However, when I try to install gnu radio now I am stuck on py39-matplotlib. I have tried with python37 and python38 without any luck.  Bellow is the error I find. I have check on Macport tickets and cannot figure out the solution ... any help is greatly appreciated.

--->  Fetching archive for py39-matplotlib
Warning: Your DNS servers incorrectly claim to know the address of nonexistent hosts. This may cause checksum mismatches for some ports. See this page for more information: <https://trac.macports.org/wiki/MisbehavingServers>
--->  Attempting to fetch py39-matplotlib-3.4.3_0+cairo+webagg.darwin_21.arm64.tbz2 from https://packages.macports.org/py39-matplotlib
--->  Attempting to fetch py39-matplotlib-3.4.3_0+cairo+webagg.darwin_21.arm64.tbz2 from https://ywg.ca.packages.macports.org/mirror/macports/packages/py39-matplotlib
--->  Attempting to fetch py39-matplotlib-3.4.3_0+cairo+webagg.darwin_21.arm64.tbz2 from https://ema.uk.packages.macports.org/py39-matplotlib
--->  Building py39-matplotlib
Error: Failed to build py39-matplotlib: command execution failed
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_python_py-matplotlib/py39-matplotlib/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there is a bug.
Error: Processing of port gnuradio failed

vjarao, 2021-11-09:
Dear all, one step closer with Mac M1 OS 12.01 (Monterrey). I have successfully installed GNURADIO. No errors in the process. However when opening the application  I get:

dbus[2679]: Dynamic session lookup supported but failed: launchd did not provide a socket path, verify that org.freedesktop.dbus-session.plist is loaded!
No such schema “org.gnome.desktop.interface”
zsh: trace trap  gnuradio-companion

I suspect they are working on it because just few days ago I was not able to install gnu radio.

[15 more replies]