gnuradio/gnuradio #4421: GRC: "k/M/G" notation is accepted by GRC but not python

jacobagilbert, 2021-03-20:
This was noticed by a user on matrix, but if you enter `1M` for a parameter in a block, the tooltip will show `1000000.0` and there are no errors shown, when you close the Properties dialogue the canvas is happy. "Correctly" entered values such as 1e6 present on the canvas exactly as "1M".

[image]

However executing the python flowgraph obviously fails as "1M" is passed directly to the block:

```
Executing: /usr/bin/python3 -u /opt/notation_test.py

  File "/opt/notation_test.py", line 80
    self.blocks_threshold_ff_0 = blocks.threshold_ff(-1M, -1e6, 0)
                                                       ^
SyntaxError: invalid syntax

>>> Done (return code 1)
```

The obvious workaround is "just use correct python notation" but if python is going to balk GRC should too.

ei8fdb, 2021-03-22:
@jacobagilbert What would you expect to happen in this situation? I'm curious about other's thoughts also.

I would expect GRC to translate whatever notations **it** accepts to the "correct Python notation", and for the Python flowgraph to be created successfully.

Thoughts?

jacobagilbert, 2021-03-22:
I think what you describe would be most preferable, but it would also be acceptable for this notation to be rejected and the block (and FG) with errors.

ei8fdb, 2021-03-22:
The preferable behaviour here is:

- if a user was to type a value using k/M/G notation in a GRC block, it would display in k/M/G notation in the GRC UI
- if they then execute the flowgraph to create the Python code, the k/M/G notation value would be converted to correct python notation and inserted into the Pytho code

From Matrix discussions:

@dkozel mentions:

> GRC already has a converter for 1M -> 1e6 for input validation IIRC
> So it's figuring out how to get that to correctly run for numeric fields during code generation. I think it just isn't used during code generation
> 

@willcode mentions:

> 
> The code is in param.py, evaluate(). This is used to check that the value is acceptable, but it never gets converted to something usable by Python when the code is generated.
> 
> Right now, for instance, if sample_rate is set to 1M, a line of Python is generated that says
> self.samp_rate = samp_rate = 1M
> which is the problem.
>

marcusmueller, 2024-01-28:
time we fix that UI bug. Honestly, these prefix displays are less convenient than not having them.