gnuradio/gnuradio #4424: Cannot send command message pairs to USRPs

cclamb, 2021-03-20:
Hey guys, so updated to the latest maint-3.8 last week and pulled in a bug. I can no longer submit pairs to USRPs with commands. Not sure what exactly is happening, but I can no longer submit commands to the URSP as pairs. In my case, I wanted to change the frequency so I was sending a message like this:

      pmt::pmt_t command = pmt::cons
      ( // Make a pair                                                             
	      pmt::mp("freq"), // Key is 'freq' => sets the frequency                                                  
	      pmt::mp(new_frequency) // Set the frequency to the new frequency value                                                    
      );   

And sending it to the USRP. This now generates errors:

gr::log :ERROR: usrp_sink0 - Command message is neither dict nor pair: (freq . 2.81e+09)
1616283308 ERROR gr_log.usrp_sink0 : Command message is neither dict nor pair: (freq . 2.81e+09)
gr::log :ERROR: usrp_source0 - Command message is neither dict nor pair: (freq . 2.81e+09)
1616283308 ERROR gr_log.usrp_source0 : Command message is neither dict nor pair: (freq . 2.81e+09)

I've traced the error to this section inside of gnuradio/gr-uhd/lib/usrp_block_impl.cc:

        // Legacy code back compat: If we receive a tuple, we convert
        // it to a dict, and call the function again. Yep, this comes
        // at a slight performance hit. Sometime in the future, we can
        // hopefully remove this:
        if (pmt::is_tuple(msg)) {
            if (pmt::length(msg) != 2 && pmt::length(msg) != 3) {
                GR_LOG_ALERT(d_logger,
                             boost::format("Error while unpacking command PMT: %s") % msg);
                return;
            }
            pmt::pmt_t new_msg = pmt::make_dict();
            new_msg = pmt::dict_add(new_msg, pmt::tuple_ref(msg, 0), pmt::tuple_ref(msg, 1));
            if (pmt::length(msg) == 3) {
                new_msg = pmt::dict_add(new_msg, cmd_chan_key(), pmt::tuple_ref(msg, 2));
            }
            GR_LOG_WARN(d_debug_logger,
                        boost::format("Using legacy message format (tuples): %s") % msg);
            return msg_handler_command(new_msg);
        }
        // End of legacy backward compat code.
      
        // Turn pair into dict
        if (!pmt::is_dict(msg)) {
            GR_LOG_ERROR(d_logger,
                         boost::format("Command message is neither dict nor pair: %s") % msg);
            return;
        }

This is different from what is currently on master.

Thanks!

willcode, 2021-03-21:
@cclamb Thanks for the report. We did make a change under the hood, which looks like it had an unintended effect. The intent was to make `is_dict(x)` return `True` only when the argument really was a PMT dict. We'll take a look at it and get this fixed.

willcode, 2021-03-21:
@cclamb The intent is for you to send a dict, rather than a tuple. Can you try that and see if it works? The logic would be more or less what's in the usrp code you quoted.

cclamb, 2021-03-21:
Unfortunately, I can't really do that in this case; I work on a fairly large team and making this change would be problematic. It worked until roughly a week ago.

willcode, 2021-03-21:
OK. It is intended to keep working the way it did before, so we'll figure it out. Any other problems from the recent maint-3.8? We're making a release within a few days.

cclamb, 2021-03-21:
Nope this has been it!

cclamb, 2021-03-21:
And thanks for getting back to me so quickly!

cclamb, 2021-03-21:
In the meantime, can you give me a pointer on how to roll back to the last working version in pybombs? Thanks much!

willcode, 2021-03-21:
You can set `gitbranch: 3.8.2.0` in the `.pybombs/.config` file under the `gnuradio:` section. Pybombs always gives you the tip of a branch, unless you ask for something else.

cclamb, 2021-03-21:
Awesome thanks!

jacobagilbert, 2021-03-21:
Hi @cclamb - sorry about this, I should have caught it. I submitted #4425 which fixes this, if you'd like to test it and post there (or add yourself to 'reviewers' on the PR) that would be appreciated:

```
git fetch origin pull/4425/head:fix_uhd_pair_cmds_38
git checkout fix_uhd_pair_cmds_38
```

We will get this fixed shortly on the maint-3.8 branch.

willcode, 2021-03-21:
Merged onto maint-3.8 to make it easier to test. Works here with pair/tuple/dict messages.

cclamb, 2021-03-21:
Guys, Looks good! Verified that the frequency shift did occur, and pulled these from my logs:

gr::log :DEBUG: usrp_sink0 - Command message is pair, converting to dict: '(freq . 2.82e+09)': car(freq), cdr(2.82e+09)
1616341016 DEBUG gr_log.usrp_sink0 : Command message is pair, converting to dict: '(freq . 2.82e+09)': car(freq), cdr(2.82e+09)
gr::log :DEBUG: usrp_source0 - Command message is pair, converting to dict: '(freq . 2.82e+09)': car(freq), cdr(2.82e+09)
1616341016 DEBUG gr_log.usrp_source0 : Command message is pair, converting to dict: '(freq . 2.82e+09)': car(freq), cdr(2.82e+09)

looked at the code, seems these are emitted when the conversion occurs.

Thank you so much! I'm stunned at how quickly you were able to resolve and push!

closed 2021-03-21 by cclamb

jacobagilbert, 2021-03-21:
Glad it's working for you. If you ever refactor your code, command dictionaries are preferred. The API docs need to be updated to reflect this.

cclamb, 2021-03-21:
Will do, thanks!

PR #4425 (2021-03-21): gr-uhd: fix UHD command port for pairs
Backporting the is_dict() disambiguation for pairs caused pair type
commands to be rejected as they no longer pass is_dict(). This
restores pair command functionality.

Fixes #4424.