gnuradio/gnuradio #4431: New Hier Block (QT GUI) creates incomplete .py code

joncnist, 2021-03-22:
I created a hier block with a bunch of LEDs and put it in a flowgraph.  It fails to run with error: 
NameError: name 'Qt' is not defined
If I edit the .py code created by GRC and add a line:
`from PyQt5 import Qt`
then the hier block runs with no problems.

[led_array.py.block.zip](https://github.com/gnuradio/gnuradio/files/6184627/led_array.py.block.zip)

mormj, 2021-03-29:
I think you can/should add the `from PyQt5 import Qt' to the imports section under templates in the block yml.

joncnist, 2021-03-29:
The yml is autogenerated when the hier block is saved.  It has this line:
flags: need_qt_gui
Shouldn't that cause the proper import to be added?

mormj, 2021-03-29:
Ah I didn't catch that the hier block was created with GRC - yes that should be added then.

petrushy, 2021-10-03:
Can confirm this bug in GNU Radio version: 3.9.2.0 as well. Is there a workaround to be able to add this import statement so it is included in the autogenerated code?

mormj, 2021-10-04:
Can you add an import block in GRC with `from PyQt5 import Qt`

marcusmueller, 2024-01-28:
Seems fixed
[image]
https://gist.github.com/marcusmueller/372a4419168282f8f36a9e1b43427abd

closed 2024-01-28 by marcusmueller