gnuradio/gnuradio #4438: gnuradio 3.10: gnuradio-config-info misses enabled components

dl1ksv, 2021-03-23:
Doing a fresh build:

-- ######################################################
-- # Gnuradio enabled components                         
-- ######################################################
--   * testing-support
--   * python-support
--   * doxygen
--   * man-pages
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gnuradio-companion
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-audio
--   * * alsa
--   * * oss
--   * gr-channels
--   * gr-pdu
--   * gr-qtgui
--   * gr-trellis
--   * gr-utils
--   * gr_modtool
--   * gr_blocktool
--   * gr-vocoder
--   * gr-network

But:
$gnuradio-config-info --enabled-components
testing-support;python-support;doxygen;man-pages;gnuradio-runtime;gr-ctrlport

Older 3.9 versions report all enabled components

marcusmueller, 2021-03-23:
I guess our CMake scripting might be to blame, so carefully adding the cmake tag

drmpeg, 2021-03-24:
I have a fix for this, and it is CMake.

PR #4453 (2021-03-25): Repair gnuradio-config-info for enabled components
Fixes #4438
closed 2021-03-25 by mormj