gnuradio/gnuradio #4452: Was building as static removed?

robbiet480, 2021-03-24:
The build guide doc references `ENABLE_STATIC_LIBS` as a flag but giving it doesn't actually build like `libgnuradio-runtime.a` as [this article](http://www.trondeau.com/blog/2015/1/19/working-with-uhd-static-library) suggests it should.

robbiet480, 2021-03-26:
It appears at least part of the problem is that even with `ENABLE_STATIC_LIBS=ON`, `volk_profile` attempts to build against `libboost_filesystem.so` instead of `libboost_filesystem.a`, causing this:

```
/usr/bin/ld: attempted static link of dynamic object `/usr/local/lib/libboost_filesystem.so.1.75.0'
collect2: error: ld returned 1 exit status
make[2]: *** [volk/apps/CMakeFiles/volk_profile.dir/build.make:133: volk/apps/volk_profile] Error 1
make[1]: *** [CMakeFiles/Makefile2:1447: volk/apps/CMakeFiles/volk_profile.dir/all] Error 2
```

robbiet480, 2021-03-26:
Made some progress once I realized I needed to use `-DBUILD_SHARED_LIBS=OFF` which would trigger the static libs to be built. However, there still appears to be some key functionality missing, namely PMT, FFT and blocks. Running CMake I get this:

```
CMake Error: install(EXPORT "gnuradio-runtime-export" ...) includes target "gnuradio-runtime" which requires target "gnuradio-pmt" that is not in any export set.
CMake Error: install(EXPORT "gnuradio-filter-export" ...) includes target "gnuradio-filter" which requires target "gnuradio-fft" that is not in any export set.
CMake Error: install(EXPORT "gnuradio-filter-export" ...) includes target "gnuradio-filter" which requires target "gnuradio-blocks" that is not in any export set.
```

Commenting out the `if` for `BUILD_SHARED_LIBS=` in each of the three CMakeList.txt files for the targets lets them build properly. It seems this was added in ab2fb35677e38a384df3f9503d1f45f64bbc0374, replacing a explicit static build system. Can anyone provide context on why those three targets can't be built static normally?

robbiet480, 2021-03-26:
@noc0lour @marcusmueller As the last two that seemed to have worked on static builds, can you shed any light on how I can build GR and all the modules in the tree statically for use with `trunk-recorder`?

noc0lour, 2021-04-01:
If I remember correctly I removed the additional branch for building `static` binaries as CMake supports building dynamic and static binaries just via built-in flags. The only thing the separate branch achieved was building a static and dynamic library & binaries in the same build-directory. 

Aside from this the CMake should work, although I'm not surprised if there are broken things w.r.t to static builds as this is not tested regularly.

noc0lour, 2021-04-01:
Ah, the `GR_LIBRARY_FOO` call in gnuradio-runtime is not guarded with `if(BUILD_SHARED_LIBS)`. Try putting it around this call. IIRC everything done inside of the function is only relevant for shared libraries. If not that's another bug.

willcode, 2021-05-14:
I started on this, got a little further, gave up ...
```
diff --git a/cmake/Modules/GrBoost.cmake b/cmake/Modules/GrBoost.cmake
index a48bb232d..9df32c2d6 100644
--- a/cmake/Modules/GrBoost.cmake
+++ b/cmake/Modules/GrBoost.cmake
@@ -68,6 +68,10 @@ set(Boost_ADDITIONAL_VERSIONS
     "1.71.0" "1.71"
 )
 
+if (NOT BUILD_SHARED_LIBS)
+  set(Boost_USE_STATIC_LIBS ON)
+endif()
+
 find_package(Boost ${GR_BOOST_MIN_VERSION} REQUIRED ${BOOST_REQUIRED_COMPONENTS}
                                            OPTIONAL_COMPONENTS unit_test_framework)
 
diff --git a/gnuradio-runtime/lib/CMakeLists.txt b/gnuradio-runtime/lib/CMakeLists.txt
index 1256f6276..f887f67cf 100644
--- a/gnuradio-runtime/lib/CMakeLists.txt
+++ b/gnuradio-runtime/lib/CMakeLists.txt
@@ -129,6 +129,7 @@ target_link_libraries(gnuradio-runtime PUBLIC
   Boost::system
   Boost::regex
   Boost::thread
+  Boost::unit_test_framework
   Log4Cpp::log4cpp
   MPLib::mplib
   ${libunwind_LIBRARIES}
@@ -327,8 +328,9 @@ if(ENABLE_CTRLPORT_THRIFT)
     list(APPEND EXTRA_DEPS THRIFT)
   endif()
 endif()
-gr_library_foo(gnuradio-runtime ${EXTRA_DEPS})
-
+if (BUILD_SHARED_LIBS)
+  GR_LIBRARY_FOO(gnuradio-runtime ${EXTRA_DEPS})
+endif ()
 
 ########################################################################
 # Setup tests
diff --git a/gr-audio/lib/CMakeLists.txt b/gr-audio/lib/CMakeLists.txt
index 9921983f5..331b573ad 100644
--- a/gr-audio/lib/CMakeLists.txt
+++ b/gr-audio/lib/CMakeLists.txt
@@ -151,5 +151,7 @@ if(WIN32)
     endif()
 endif(WIN32)
 
-gr_library_foo(gnuradio-audio ${audio_deps})
+if (BUILD_SHARED_LIBS)
+  GR_LIBRARY_FOO(gnuradio-audio ${audio_deps})
+endif()
 install(FILES ${gr_audio_confs} DESTINATION ${GR_PREFSDIR})
diff --git a/gr-filter/lib/CMakeLists.txt b/gr-filter/lib/CMakeLists.txt
index a99ab975d..7cb48068f 100644
--- a/gr-filter/lib/CMakeLists.txt
+++ b/gr-filter/lib/CMakeLists.txt
@@ -80,7 +80,9 @@ if(MSVC)
     )
 endif(MSVC)
 
-GR_LIBRARY_FOO(gnuradio-filter)
+if (BUILD_SHARED_LIBS)
+  GR_LIBRARY_FOO(gnuradio-filter)
+endif ()
 
 ########################################################################
 # QA C++ Code for gr-filter
```

voibit, 2021-07-26:
> Ah, the `GR_LIBRARY_FOO` call in gnuradio-runtime is not guarded with `if(BUILD_SHARED_LIBS)`. Try putting it around this call. IIRC everything done inside of the function is only relevant for shared libraries. If not that's another bug.

Why is it only relevant for shared libs? I managed to build staticly removing all the guards, and setting the correct flags. That was against mingw. Removing the guards gets the shared archives built.

willcode, 2021-07-26:
@voibit Not sure. @noc0lour Any ideas here?

robbiet480, 2023-07-07:
Checking in on this two years later, any progress? Would love to get this working to take trunk-recorder Docker images from almost 2GB to like 100mb by only shipping what we need and statically compiled.

willcode, 2023-07-07:
No, I haven't seen any mention of static linking since the last attempt.

marcusmueller, 2024-01-28:
yeah, not doing a static build "quickly", will have to do that separately.

We should definitely sit down and discuss whether we can achieve this at all if it's not easy to achieve.

drewbug, 2024-12-24:
It's a little silly because it makes `BUILD_SHARED_LIBS` pointless, but this worked for me to build for Android statically:

```diff
--- a/gnuradio/gnuradio-runtime/lib/CMakeLists.txt
+++ b/gnuradio/gnuradio-runtime/lib/CMakeLists.txt
@@ -53 +53 @@
-add_library(gnuradio-runtime
+add_library(gnuradio-runtime STATIC
--- a/gnuradio/gnuradio-runtime/lib/pmt/CMakeLists.txt
+++ b/gnuradio/gnuradio-runtime/lib/pmt/CMakeLists.txt
@@ -20 +20 @@
-add_library(gnuradio-pmt
+add_library(gnuradio-pmt STATIC
--- a/gnuradio/gr-analog/lib/CMakeLists.txt
+++ b/gnuradio/gr-analog/lib/CMakeLists.txt
@@ -24 +24 @@
-add_library(gnuradio-analog
+add_library(gnuradio-analog STATIC
--- a/gnuradio/gr-audio/lib/CMakeLists.txt
+++ b/gnuradio/gr-audio/lib/CMakeLists.txt
@@ -25 +25 @@
-add_library(gnuradio-audio audio_registry.cc)
+add_library(gnuradio-audio STATIC audio_registry.cc)
--- a/gnuradio/gr-blocks/lib/CMakeLists.txt
+++ b/gnuradio/gr-blocks/lib/CMakeLists.txt
@@ -28 +28 @@
-add_library(gnuradio-blocks
+add_library(gnuradio-blocks STATIC
--- a/gnuradio/gr-channels/lib/CMakeLists.txt
+++ b/gnuradio/gr-channels/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-channels
+add_library(gnuradio-channels STATIC
--- a/gnuradio/gr-digital/lib/CMakeLists.txt
+++ b/gnuradio/gr-digital/lib/CMakeLists.txt
@@ -24 +24 @@
-add_library(gnuradio-digital
+add_library(gnuradio-digital STATIC
--- a/gnuradio/gr-dtv/lib/CMakeLists.txt
+++ b/gnuradio/gr-dtv/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-dtv
+add_library(gnuradio-dtv STATIC
--- a/gnuradio/gr-fec/lib/CMakeLists.txt
+++ b/gnuradio/gr-fec/lib/CMakeLists.txt
@@ -26 +26 @@
-add_library(gnuradio-fec
+add_library(gnuradio-fec STATIC
--- a/gnuradio/gr-fft/lib/CMakeLists.txt
+++ b/gnuradio/gr-fft/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-fft
+add_library(gnuradio-fft STATIC
--- a/gnuradio/gr-filter/lib/CMakeLists.txt
+++ b/gnuradio/gr-filter/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-filter
+add_library(gnuradio-filter STATIC
--- a/gnuradio/gr-qtgui/lib/CMakeLists.txt
+++ b/gnuradio/gr-qtgui/lib/CMakeLists.txt
@@ -25 +25 @@
-add_library(gnuradio-qtgui
+add_library(gnuradio-qtgui STATIC
--- a/gnuradio/gr-trellis/lib/CMakeLists.txt
+++ b/gnuradio/gr-trellis/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-trellis
+add_library(gnuradio-trellis STATIC
--- a/gnuradio/gr-uhd/lib/CMakeLists.txt
+++ b/gnuradio/gr-uhd/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-uhd
+add_library(gnuradio-uhd STATIC
--- a/gnuradio/gr-video-sdl/lib/CMakeLists.txt
+++ b/gnuradio/gr-video-sdl/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-video-sdl
+add_library(gnuradio-video-sdl STATIC
--- a/gnuradio/gr-vocoder/lib/CMakeLists.txt
+++ b/gnuradio/gr-vocoder/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-vocoder
+add_library(gnuradio-vocoder STATIC
--- a/gnuradio/gr-wavelet/lib/CMakeLists.txt
+++ b/gnuradio/gr-wavelet/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-wavelet
+add_library(gnuradio-wavelet STATIC
--- a/gnuradio/gr-zeromq/lib/CMakeLists.txt
+++ b/gnuradio/gr-zeromq/lib/CMakeLists.txt
@@ -23 +23 @@
-add_library(gnuradio-zeromq
+add_library(gnuradio-zeromq STATIC
--- a/gr-grand/lib/CMakeLists.txt
+++ b/gr-grand/lib/CMakeLists.txt
@@ -50 +50 @@
-add_library(gnuradio-grand SHARED ${grand_sources})
+add_library(gnuradio-grand STATIC ${grand_sources})
--- a/gr-ieee802-11/lib/CMakeLists.txt
+++ b/gr-ieee802-11/lib/CMakeLists.txt
@@ -70 +70 @@
-add_library(gnuradio-ieee802_11 SHARED ${ieee802_11_sources})
+add_library(gnuradio-ieee802_11 STATIC ${ieee802_11_sources})
--- a/gr-ieee802-15-4/lib/CMakeLists.txt
+++ b/gr-ieee802-15-4/lib/CMakeLists.txt
@@ -61 +61 @@
-add_library(gnuradio-ieee802_15_4 SHARED ${ieee802_15_4_sources})
+add_library(gnuradio-ieee802_15_4 STATIC ${ieee802_15_4_sources})
--- a/gr-osmosdr/lib/CMakeLists.txt
+++ b/gr-osmosdr/lib/CMakeLists.txt
@@ -43 +43 @@
-add_library(gnuradio-osmosdr SHARED)
+add_library(gnuradio-osmosdr STATIC)
--- a/gr-sched/lib/CMakeLists.txt
+++ b/gr-sched/lib/CMakeLists.txt
@@ -37 +37 @@
-add_library(gnuradio-sched SHARED ${sched_sources})
+add_library(gnuradio-sched STATIC ${sched_sources})
```