gnuradio/gnuradio #4454: fec: (en|de)code_rs_ccsds() functions are not exported

daniestevez, 2021-03-24:
In #3657, I pushed some changes to the Reed-Solomon functions in gr-fec in order to have the symbols for `(en|de)code_rs_char()`, `(en|de)code_rs_8()` and `(en|de)code_rs_ccsds()` exported in the gr-fec library. The goal was for OOT modules (in particular gr-satellites) to be able to re-use these functions instead of shipping their own copy. Before this PR, only `(en|de)code_rs_char()` were exported.

Now I've noticed that in GNU Radio 3.9.0 `(en|de)code_rs_ccsds()` are not exported. The other functions are exported fine:

```
$ objdump -T libgnuradio-fec.so.3.9.0 | grep code_rs
000000000009cb70 g    DF .text	0000000000000119  Base        encode_rs_char
000000000009bcb0 g    DF .text	0000000000000eb8  Base        decode_rs_char
000000000009bba0 g    DF .text	0000000000000109  Base        encode_rs_8
000000000009af00 g    DF .text	0000000000000c97  Base        decode_rs_8
```

I think the problem has something to do with #3821, since apparently my PR gave build problems in Windows. What I don't understand is why the `_ccsds()` functions are the only not being exported, given that `FEC_API` was removed from all of them.

We should probably think of a good way to export these in a cross-platform way, since @ryanvolz was still getting some build warnings.

Yamakaja, 2021-08-02:
Mhm, with objdump you're listing all objects in that file, not necessarily the ones which are being exported. Though obviously symbols which aren't exported might end up being inlined or mangled in other ways by the compiler. So the fact that the symbols aren't there is certainly interesting ...

To me it looks like what we have here is a declaration without a definition. `{de,en}code_rs_ccsds` is defined in the `reed-solomon` subdirectory, which is built into an independent library that is never linked into gr-fec, and maybe (probably*) not even used in `gr-fec` at all? From a quick glance there appear to be no references to that code except for the cmake subdirectory declaration and those bad function declarations.

Yamakaja, 2021-08-03:
Okay, so ignore my message from tonight, apparently it was a little too late :D

It looks like the actual issue here is just that the declaration of those functions is not actually included into the source file that defines them. As such the symbols are marked local by default. I see two options:

* Declare [`decode_rs_ccsds`](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/lib/reed-solomon/decode_rs_ccsds.c#L11) and [`encode_rs_ccsds`](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/lib/reed-solomon/encode_rs_ccsds.c#L11) as `FEC_API` like in [fixed.h](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/lib/reed-solomon/fixed.h#L37-L41)
* Include [rs.h](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/include/gnuradio/fec/rs.h) in [encode_rs_ccsds.c](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/lib/reed-solomon/encode_rs_ccsds.c) or similar, to make sure the declarations are available.

daniestevez, 2021-08-03:
Hi,

The thing is that when I submitted the PR #3657, I marked `decode_rs_ccsds` and `encode_rs_ccsds` as `FEC_API`. However, this gave problems with building on Windows (take a look at 00fd051dff4a73081c40f4ccfc3b4ac8c62d74f1). How this causes conflicts with Windows is the part that I don't understand, so I don't know of a proper way to fix it.

dkozel, 2021-08-03:
@ryanvolz Can you shed any light on how we can do this and not break Windows builds?

ryanvolz, 2021-08-03:
I don't understand the issue that MSVC has with marking the ccsds functions as `FEC_API`, so I'd recommend basing a branch off of #4410 (better yet, LGTM!) and trying out the approaches mentioned upthread. That way everyone will be able to see the Windows build output and know whether something works.

Yamakaja, 2021-08-04:
> Hi,
> 
> The thing is that when I submitted the PR #3657, I marked `decode_rs_ccsds` and `encode_rs_ccsds` as `FEC_API`. However, this gave problems with building on Windows (take a look at [00fd051](https://github.com/gnuradio/gnuradio/commit/00fd051dff4a73081c40f4ccfc3b4ac8c62d74f1)). How this causes conflicts with Windows is the part that I don't understand, so I don't know of a proper way to fix it.

Oh, i get it now .... third time's the charm :D Not sure what to do about it though ...

daniestevez, 2024-01-13:
This has been solved by #6984.

closed 2024-01-13 by daniestevez