gnuradio/gnuradio #4464: Help Request - upgrade gr-satellites to latest commits

XE3ADC, 2021-03-26:
Hello,
The Gr-satellites gnuradio OOT module was installed on my system with GNUradio v3.8 under Ubuntu 18.04lt, I used the following commands:
**git clone --recursive -b maint-3.8 –single-branch https://github.com/gnuradio/gnuradio.git 
cd gnuradio/
mkdir build
cd build
cmake ../
make -j4
make test 
sudo make install
sudo ldconfig**

The installation was successful and gr-satellites worked fine.

During testing GR-satellites showed issues decoding the Aztechsat-1 telemetry beacon and the developer modified the code to solve the issues. The code changes were committed to the maint-3.8 and master branches of gr-satellites.

Now, I need to apply those commits on my system to have the issues solved. The developer recommended to pull the commits to my system and build from source.
I virtually have no experience using git and have not found a concise procedure online that allows me to upgrade my system.
I was thinking to do a re-install from scratch, but as the system is stable and working fine I do not want to mess with it too much.
What would be the appropriate and most efficient procedure to achieve my purpose?, would you please provide a set of concise instructions?  
Thanks in advance and best regards
Abraham Diaz de Leon
Villahermosa - Tabasco - Mexico

willcode, 2021-03-26:
The same sequence of commands will work to install gr-satellites from source (substituting the git clone command below). I'm assuming you installed gr-satellites from source the first time. If it was installed via a package manager, remove it first.

`git clone --recursive -b maint-3.8 --single-branch https://github.com/daniestevez/gr-satellites`

XE3ADC, 2021-03-26:
Thanks Willcode, In fact gr-satellites was installed with the following commands, not as indicated in my previous note:
```
git clone -b maint-3.8 --single-branch https://github.com/daniestevez/gr-satellites.git
cd gr-satellites
mkdir build
cd build
cmake ../
make -j5
sudo make install
sudo ldconfig
```
Therefore it was installed from source.
I tried to do as suggested but I get  the following:

```
abesr@abraham-5520:~$ git clone --recursive -b maint-3.8 --single-branch https://github.com/daniestevez/gr-satellites
fatal: destination path 'gr-satellites' already exists and is not an empty directory.
abesr@abraham-5520:~$ cd gr-satellites
abesr@abraham-5520:~/gr-satellites$ ls
apps             CMakeLists.txt  lib          readthedocs.yml       test.sh
build            docs            LICENSE      satellite-recordings
CCSDS_README.md  examples        MANIFEST.md  satellite_teams.md
CHANGELOG.md     grc             python       swig
cmake            include         README.md    tests

```
Sorry, for placing this request on the wrong place, I will close the issue follow up as advised,
Thanks and regards
Abraham Diaz de Leon
Villahermosa -Tabasco - Mexico

closed 2021-03-26 by XE3ADC