gnuradio/gnuradio #4469: TypeError: primitive_connect()

adigeak, 2021-03-28:
when executing the code having problem.
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "a.py", line 209, in <module>
    main()
  File "a.py", line 187, in main
    tb = top_block_cls()
  File "a.py", line 144, in __init__
    self.connect((self.rtlsdr_source_0, 0), (self.qtgui_freq_sink_x_0, 0))
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.top_block_pb object at 0x7fe58fbaf630>, <Swig Object of type 'gr::basic_block_sptr *' at 0x7fe58bd46540>, 0, <gnuradio.gr.gr_python.basic_block object at 0x7fe58e3915b0>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.
```

mormj, 2021-03-29:
@adigeak - can you please provide more information about how your `a.py` block is connected, preferably a minimal example where the issue can be replicated?

adigeak, 2021-03-29:
here it's want me to send the grc file also
https://paste.debian.net/1191571/

adigeak, 2021-04-01:
@mormj you are able to replicate it?