gnuradio/gnuradio #4475: Fresh install, AttributeError: type object 'audio' has no attribute 'sink'

PenelopeFudd, 2021-04-01:
I'm a new GnuRadio user, and I'm having a problem:

  *  Fedora 32 system
  *  Installed Gnuradio
  *  Tried to hook a signal source to an audio sink and run it
  *  Got the error "AttributeError: type object 'audio' has no attribute 'sink'"

Here are the commands I used to install it:
```
    sudo dnf copr enable gnuradio/gnuradio-master  
    sudo dnf install gnuradio  
    gnuradio-companion  
```
 
Here's how I triggered the error:

* I dragged a Signal Source and an Audio Sink onto the field, and hooked them together
* It gave a flowgraph error (red arrow) because the signal source was set to 'Complex', while the audio sink was set to 'Integer'.
* When I set the signal source to integer, the error went away (black arrow), and the Run>Generate and Run>Execute menu options enabled themselves.

It gave the following error, both in and out of the GUI:
```
    Executing: /usr/bin/python3 -u /home/user/work/gr/audio.py   
      
    Warning: failed to XInitThreads()   
    QSocketNotifier: Can only be used with threads started with QThread   
    Traceback (most recent call last):   
      File "/home/user/work/gr/audio.py", line 135, in <module>   
        main()   
      File "/home/user/work/gr/audio.py", line 111, in main   
        tb = top_block_cls()   
      File "/home/user/work/gr/audio.py", line 77, in __init__   
        self.audio_sink_0 = audio.sink(samp_rate, '', True)   
    AttributeError: type object 'audio' has no attribute 'sink'   
      
    >>> Done (return code 1)  
```

Here is the python program it generated:
```
    #!/usr/bin/env python3  
    # -*- coding: utf-8 -*-  
      
    #  
    # SPDX-License-Identifier: GPL-3.0  
    #  
    # GNU Radio Python Flow Graph  
    # Title: Not titled yet  
    # GNU Radio version: 3.8.2.0  
      
    from distutils.version import StrictVersion  
      
    if __name__ == '__main__':  
        import ctypes  
        import sys  
        if sys.platform.startswith('linux'):  
            try:  
                x11 = ctypes.cdll.LoadLibrary('libX11.so')  
                x11.XInitThreads()  
            except:  
                print("Warning: failed to XInitThreads()")  
      
    from gnuradio import analog  
    from gnuradio import audio  
    from gnuradio import gr  
    from gnuradio.filter import firdes  
    import sys  
    import signal  
    from PyQt5 import Qt  
    from argparse import ArgumentParser  
    from gnuradio.eng_arg import eng_float, intx  
    from gnuradio import eng_notation  
      
    from gnuradio import qtgui  
      
    class audio(gr.top_block, Qt.QWidget):  
      
        def __init__(self):  
            gr.top_block.__init__(self, "Not titled yet")  
            Qt.QWidget.__init__(self)  
            self.setWindowTitle("Not titled yet")  
            qtgui.util.check_set_qss()  
            try:  
                self.setWindowIcon(Qt.QIcon.fromTheme('gnuradio-grc'))  
            except:  
                pass  
            self.top_scroll_layout = Qt.QVBoxLayout()  
            self.setLayout(self.top_scroll_layout)  
            self.top_scroll = Qt.QScrollArea()  
            self.top_scroll.setFrameStyle(Qt.QFrame.NoFrame)  
            self.top_scroll_layout.addWidget(self.top_scroll)  
            self.top_scroll.setWidgetResizable(True)  
            self.top_widget = Qt.QWidget()  
            self.top_scroll.setWidget(self.top_widget)  
            self.top_layout = Qt.QVBoxLayout(self.top_widget)  
            self.top_grid_layout = Qt.QGridLayout()  
            self.top_layout.addLayout(self.top_grid_layout)  
      
            self.settings = Qt.QSettings("GNU Radio", "audio")  
      
            try:  
                if StrictVersion(Qt.qVersion()) < StrictVersion("5.0.0"):  
                    self.restoreGeometry(self.settings.value("geometry").toByteArray())  
                else:  
                    self.restoreGeometry(self.settings.value("geometry"))  
            except:  
                pass  
      
            ##################################################  
            # Variables  
            ##################################################  
            self.samp_rate = samp_rate = 32000  
      
            ##################################################  
            # Blocks  
            ##################################################  
            self.audio_sink_0 = audio.sink(samp_rate, '', True)  
            self.analog_sig_source_x_0 = analog.sig_source_i(samp_rate, analog.GR_COS_WAVE, 1000, 1, 0, 0)  
      
      
      
            ##################################################  
            # Connections  
            ##################################################  
            self.connect((self.analog_sig_source_x_0, 0), (self.audio_sink_0, 0))  
      
      
        def closeEvent(self, event):  
            self.settings = Qt.QSettings("GNU Radio", "audio")  
            self.settings.setValue("geometry", self.saveGeometry())  
            event.accept()  
      
        def get_samp_rate(self):  
            return self.samp_rate  
      
        def set_samp_rate(self, samp_rate):  
            self.samp_rate = samp_rate  
            self.analog_sig_source_x_0.set_sampling_freq(self.samp_rate)  
      
      
      
      
      
    def main(top_block_cls=audio, options=None):  
      
        if StrictVersion("4.5.0") <= StrictVersion(Qt.qVersion()) < StrictVersion("5.0.0"):  
            style = gr.prefs().get_string('qtgui', 'style', 'raster')  
            Qt.QApplication.setGraphicsSystem(style)  
        qapp = Qt.QApplication(sys.argv)  
      
        tb = top_block_cls()  
      
        tb.start()  
      
        tb.show()  
      
        def sig_handler(sig=None, frame=None):  
            Qt.QApplication.quit()  
      
        signal.signal(signal.SIGINT, sig_handler)  
        signal.signal(signal.SIGTERM, sig_handler)  
      
        timer = Qt.QTimer()  
        timer.start(500)  
        timer.timeout.connect(lambda: None)  
      
        def quitting():  
            tb.stop()  
            tb.wait()  
      
        qapp.aboutToQuit.connect(quitting)  
        qapp.exec_()  
      
    if __name__ == '__main__':  
        main()  
```

How do I make it work?

Thanks

willcode, 2021-04-01:
Audio Sink has a float input. GRC should have refused to generate this program, but indeed it did when I tried it. Use a float Signal Source. For me, this resulted in no audio, but not the error message you see.

The next problem is that the program is getting confused about what `audio` is. It is both the class that was imported from the gr library and the name of your top block. Under Options, rename it to something else. Again, we with GRC could pick up naming conflicts like this.

Your output starts off with X11 problems - are you running this remotely? In the Options block, select "No GUI" if so.

willcode, 2021-04-01:
Tagged this as a GRC bug for allowing the int->float connection. Were the id conflicts ironed out in a later version?

willcode, 2021-06-02:
Type checking added. Closing.

closed 2021-06-02 by willcode