gnuradio/gnuradio #4478: Issues with running first tutorial on scotts

shinobirain, 2021-04-02:
Hi Everyone! 

  I recently started using the hackrf one on my windows 10 pc and have having issues with the first tutorial from great scotts in that I cannot seem to receive/see any of the FM signals I see in SDR#. [Here](https://youtu.be/zNUCiGVJQo0?t=862) is the link to the tutorial I am using with the time stamped right has hes about to run the project.
  
Below are some pictures that I have attached to show what I have going on right now in the project file. 

[image]
[image]
[image]
[image]

**A few background things**:
1. Currently running the 2021.03.1 Hackrf One Firmware and GNUradio 3.8

2. The command prompt does indicate warnings that say they are over written by something else that is very similarly titled (shown in the picture below). There are many more above; however, I'm not sure if that's effecting the output...?
[image]

3. I have also used SDR# with my Hackrf One and have been able to receive FM radio and listen to them as well. 
[image]

Thank you for your help and suggestions in advance!

KorgBoy, 2021-04-06:
For GNU Radio Companion, did you try using the hackrf_info.exe utility contained in GNU Radio Companion's 'bin' directory? When the HackRF One is connected up to the computer ---- the utility will probably provide details such as "hackrf=0". 

So that's when we have to put something like "hackrf=0" into the "Device Arguments" box in the osmocomm source settings - which is the settings directly under the 'output type' settings in your screenshot. So your "Device Arguments" box is currently blank ------- so the software most likely isn't patched through to the HackRF One at the moment.

shinobirain, 2021-04-07:
> For GNU Radio Companion, did you try using the hackrf_info.exe utility contained in GNU Radio Companion's 'bin' directory? When the HackRF One is connected up to the computer ---- the utility will probably provide details such as "hackrf=0".
  
I tried running the hackrf_info.exe and it results in opening a window for like 0.25 of a second and I cant view anything in it lol. 

I also tried changing the text in "Device Arguments" to both "hackrf=0" & "hackrf=1":

- hackrf=0 results in the same as the graph shown in picture 4
- hackrf=1 results in "Runtimerror: Failed to use "1" as HackRF device index: not enough devices"

I apologize for my late reply, I didn't receive a notice; however, I do have it enabled now so I should be able to get the replies now! Thank you again.

KorgBoy, 2021-04-07:
> I tried running the hackrf_info.exe and it results in opening a window for like 0.25 of a second and I cant view anything in it lol.

I think that's only due to usage of MS Windows maybe. Do a google search for opening a 'command window' in MS Windows - and then within that command window - need to change directory to gnu radio companion's 'bin' directory. And then need to execute the hackrf_info.exe from there ...... such as to type hackrf_info.exe (followed by hitting the Enter key).

shinobirain, 2021-04-07:
> I think that's only due to usage of MS Windows maybe. Do a google search for opening a 'command window' in MS Windows - and then within that command window - need to change directory to gnu radio companion's 'bin' directory. And then need to execute the hackrf_info.exe from there ...... such as to type hackrf_info.exe (followed by hitting the Enter key).
> 
> The nice thing is that the 'hackrf=0' setting is working.
> 
> Could also try this grc that works form me for GRC 3.7 -------- that I got from a tutorial.

I was able to run the hackrf_info and found the following:
[image]

I also imported for your lesson file; however, I needed to delete "missing" blocks and edited it to become this:
[image]

The results are still the same as picture 4. No audio and no observable FM stations

shinobirain, 2021-04-07:
Also to add on, I currently have set up on my GNURadio project:

[test1.zip](https://github.com/gnuradio/gnuradio/files/6274456/test1.zip)

KorgBoy, 2021-04-08:
Thanks shino. I will later use your project file to test on my HACKRF One, and will get back to you on the results/outcome.

KorgBoy, 2021-04-08:
Shino ------ I had to roll it back to GRC 3.7 ---- as my Win 8.1 machine doesn't handle GRC 3.8. But the setup that you used works just fine. I will load your test1.grc file into GRC 3.8 tomorrow on another machine running Win 10. I notice that you left the frequency of 99.9 MHz for your screenshot. Your frequency needs to be set to 100.6 MHz right?

Anyway - I'll attach the screenshot I got just moments ago ....... it's receiving an FM station - listening to the music from it. Also getting other stations nearby too.

[image]

shinobirain, 2021-04-08:
Wow that's looks way better than my current results lol. I look forward to seeing the results on 3.8. 

On a side note, I noticed that a few thing were added on to the file: such as the additional variable frequencies and the change from the GT to WX GUI. Would the change from GT to WX have anything  to do with the results I am getting or could it be something else like drivers?

KorgBoy, 2021-04-09:
Shino - I just executed the file you provided test1.grc with no changes at all to anything - running GRC v3.8.2.0-57-gd71cd177 on a Win 10 machine, and the HackRF One has an antenna that came with the system that covers the FM radio station band.

What sort of antenna are you using?

[image]

shinobirain, 2021-04-09:
Hi Korg, 

 Thank you for sticking with me through the process. 
 
 In terms of the antenna, I'm not quite sure about the antenna I am using. My university just the hackrf around with an antenna already with it, so maybe getting one more specific to fm radio would be good, though that would still make me question how I could get the signals in SDR# and not in GNURadio. 
 
 With regards to the error at the bottom of the link you provided, I haven't run into that issue so I'm sure if it's the cause either.

KorgBoy, 2021-04-10:
shino ----- could also check the "run_gr.bat" file in the 'bin' directory. 

There will be lines in that batch file - such as :

`REM --- Set GRC environment ---
REM set GRC_BLOCKS_PATH=%~dp0..\share\gnuradio\grc\blocks
set GRC_BLOCKS_PATH=`

The second line (ie. set GRC_BLOCKS_PATH=%~dp0..\share\gnuradio\grc\blocks) tends to double- up. So I put a 'REM' at the front of it - to stop that command from executing. That might not necessarily be the cause of the activity seen in your MS Windows Command Window, but could just check that batch file - just in case.

shinobirain, 2021-04-16:
Hi Korg, 

  I apologize for not being able to get back to you sooner! With regards to what you posted above, I cant seem to edit the file. I have tried changing the security GNURadio so that I can edit it; however, it doesn't seem to work.
  
In terms of the GNURadio, I still have not been able to figure out the issue with the program. I am still coming out with the same results.

KorgBoy, 2021-04-19:
Hi shino! No problem at all. I will test your code on two other HackRF One modules that I ordered separately from my existing 2 ----- and I'll see if your code works on the two that only just arrived. The FM receiver code is working properly with my first 2 modules. Will report back soon about how it goes with these other two (and will also include pic later showing all four modules).

KorgBoy, 2021-04-21:
Hi again shino. The two new HackRF One modules (that arrived) are also working with the exact same file you provided --- test1.grc

Running on Windows 10, GNU Radio Companion v3.8.2.0-57-gd71cd177 (Python 3.9.0)

I used the hackrf_info.exe program to get the information for these two modules - pasted below.

C:\Program Files\GNURadio-3.8\bin>hackrf_info
hackrf_info version: git-cc7f599
libhackrf version: git-cc7f599 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc2a5c231b
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00584f5c

C:\Program Files\GNURadio-3.8\bin>hackrf_info
hackrf_info version: git-cc7f599
libhackrf version: git-cc7f599 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc244b741b
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0066435d

KorgBoy, 2021-04-23:
Shino ----- I forgot to mention - does that relative gain versus frequency window come up when you run that test1.grc code? Anyway - at lease one nice thing is that your HackRF One actually works fine with another software.

shinobirain, 2021-04-27:
Hi Korg,

   Yes, the relative gain vs frequency window does come up when I run the test1 file.